clash-lang/clash-compiler #1591: singletons-generated code hangs clash

uzytkownik, 2020-11-15:
Following code loads in `clashi` but seems to hang when running `:verilog` or `:systemverilog`
```
$(singletons [d|
    countStates :: Nat -> Nat -> Nat
    countStates a b = 1 + aux a where
      aux :: Nat -> Nat
      aux s
        | s == 0       = 0
        | s >= max a b = aux (s - max a b)
        | otherwise    = 1 + aux (s + a)
  |])
data NextLineMask this that =
  NextLineMask
  { _nextIndex    :: Index (CountStates this that)
  } deriving (Generic)
makeLenses ''NextLineMask

nextLineMask :: forall this that. (KnownNat this, KnownNat that) => Index (CountStates this that) -> NextLineMask this that
nextLineMask =
  errorX ""

topEntity :: Signal System (Index (CountStates 3 4))
          -> Signal System (NextLineMask 3 4)
topEntity = fmap nextLineMask
```

christiaanb, 2020-11-15:
Confirmed on HEAD of master. It even already hangs before printing out its debug output that you could normally get by compiling with `-fclash-debug Debug(Final/Name/Applied)`. Interesting..

christiaanb, 2020-11-15:
I've tracked it down to at least something in https://github.com/clash-lang/clash-compiler/blob/12f8e23b87cc5346f3e2a04b2f4492e526f3f1d2/clash-lib/src/Clash/Core/Type.hs#L497 going into a loop.

Not surprising, since that function, together with https://github.com/clash-lang/clash-compiler/blob/12f8e23b87cc5346f3e2a04b2f4492e526f3f1d2/clash-lib/src/Clash/Core/Type.hs#L405

implements type family calculations

alex-mckenna, 2020-11-15:
Could it be caused by 1dd8790?

christiaanb, 2020-11-15:
No, `normalizeType` isn't used by `reduceTypeFamily`

commit 4d3dcfa6 (2020-11-17): Reduce type family arguments eagerly

Fixed #1591
commit 4ce864d0 (2020-11-17): Reduce type family arguments eagerly

Fixed #1591
commit ef73a452 (2020-11-17): Reduce type family arguments eagerly

Fixed #1591
PR #1593 (2020-11-17): Reduce type family arguments eagerly
Fixed #1591
closed 2020-11-17 by christiaanb