clash-lang/clash-compiler #1592: Add `SaturatingEnum`?

martijnbastiaan, 2020-11-15:
https://old.reddit.com/r/haskell/comments/jtjgej/where_lions_roam_haskell_hardware_on_the_veldt/gceunep/?context=3

gergoerdi, 2020-11-16:
Does this need a typeclass? I have this [in RetroClash](https://github.com/gergoerdi/retroclash-lib/blob/master/src/RetroClash/Utils.hs) and have had a ton of uses for them: 

```
nextIdx :: (Eq a, Enum a, Bounded a) => a -> a
nextIdx = fromMaybe minBound . succIdx

prevIdx :: (Eq a, Enum a, Bounded a) => a -> a
prevIdx = fromMaybe maxBound . predIdx

moreIdx :: (Eq a, Enum a, Bounded a) => a -> a
moreIdx = fromMaybe maxBound . succIdx

lessIdx :: (Eq a, Enum a, Bounded a) => a -> a
lessIdx = fromMaybe minBound . predIdx

succIdx :: (Eq a, Enum a, Bounded a) => a -> Maybe a
succIdx x | x == maxBound = Nothing
          | otherwise = Just $ succ x

predIdx :: (Eq a, Enum a, Bounded a) => a -> Maybe a
predIdx x | x == minBound = Nothing
          | otherwise = Just $ pred x
```

martijnbastiaan, 2020-11-16:
I was mostly thinking of `satSucc` and `satPred` which are currently defined as members of `SaturatingNum`: http://hackage.haskell.org/package/clash-prelude-1.2.4/docs/Clash-Class-Num.html#v:satSucc. We could pull those out into a separate type class `SaturatingEnum`. We'd remain backwards compatible and allow types that can't implement `Num` to have `satSucc` still. Its default implementation could definitely be based on Eq/Enum/Bounded though!

paddytheplaster, 2020-11-26:
> Does this need a typeclass? I have this [in RetroClash](https://github.com/gergoerdi/retroclash-lib/blob/master/src/RetroClash/Utils.hs) and have had a ton of uses for them:
>
> ```
> nextIdx :: (Eq a, [...]

A class defines an API. You can use functions like yours as a default implementation and programmers may override the functions in the API. If you define functions at the top level, users can no longer override them.

Regards,

Paddy

gergoerdi, 2020-11-26:
Yeah I'm well aware what typeclasses are for... The below definitions are
also using typeclasses, but instead of new degrees of freedom, they build
on `Enum`, `Bounded` and `Eq`.

So the question here is whether there are useful `SaturatingEnum` instances
that would differ from what you get based on the above three typeclasses.

gergoerdi, 2020-11-26:
For example, when is `satSucc SatBound x` equal to `x`? I would argue it
should be `maxBound`. If you have a `SatEnum` typeclass, then this can be
by convention at most. But if the definition of `satSucc` is based on the
`Bound` instance, then it will hold by definition.