clash-lang/clash-compiler #1601: Use singletons instead of own SNat

uzytkownik, 2020-11-28:
singletons library is already in dependencies of clash. However clash uses own `SNat` implementation. That means that code using singletons needs to convert back and forth between them.

gergoerdi, 2020-11-28:
Can you extract a `KnownNat` witness from a `singletons`-style `Nat` singleton? I would expect that to be the biggest hurdle.

uzytkownik, 2020-11-28:
@gergoerdi Yes. The definitions are almost exactly the same:

[Singletons](https://www.stackage.org/haddock/lts-16.23/singletons-2.6/src/Data.Singletons.TypeLits.Internal.html#SNat):
```haskell
data SNat (n :: Nat) = KnownNat n => SNat
type instance Sing = SNat
```

[Clash](https://www.stackage.org/haddock/lts-16.23/clash-prelude-1.2.5/src/Clash.Promoted.Nat.html#SNat):
```haskell
data SNat (n :: Nat) where
  SNat :: KnownNat n => SNat n
```

christiaanb, 2020-11-28:
It doesn’t have a Lift instance though

martijnbastiaan, 2020-11-28:
`singletons` already depends on `template-haskell`, so perhaps the maintainers would accept a PR?