clash-lang/clash-compiler #1602: Unexpected behaviours when using derived BitPack

uzytkownik, 2020-11-28:
Derived bitpacks uses cascade of if's approach. This seems to have unintended side-effects.

### Generic derive

```haskell

data Foo = F0 | F1 | F2 deriving (Generic, BitPack)

f :: Foo -> Unsigned 2
f F0 = 0
f F1 = 1
f F2 = 2

g :: Foo -> Unsigned 2
g F2 = 2
g F1 = 1
g F0 = 0

f' :: BitVector 2 -> Unsigned 2
f' = f . pack

g' :: BitVector 2 -> Unsigned 2
g' = g . pack
```

- In simulation both behave identically and for `0b11` return 2
- In HDL f' behaves like simulation
- In HDL g' returns 0 for `0b11`
- IMHO the principle of least surprise would indicate that returned value should be `errorX`. This allows synthesis to produce more efficient implementation and matches behaviour of partial function in Haskell returning ⊥.

### Catch-all clause

I'd note that sometimes the simulation-like behaviour is intended:

```haskell

data MMIOReg = Reg0 | Reg1 | ... RegN | UnknownReg deriving (Generic)

{-# ANN module (DataReprAnn
                  $(liftQ [t|MMIO|])
                  32
                  [ ConstrRepr 'Reg0 0xFFF 0x000 []
                  , ConstrRepr 'Reg1 0xFFF 0x004 []
                  ...
                  , ConstrRepr 'RegN 0xFFF 0x??? []
                  , ConstrRepr 'UnknownReg 0x000 0xFFC []
                  ]) #-}

-- This works as intended
readMMIO :: MMIOState -> MMIOReg -> Unsigned 8
readMMIO st Reg0 = ...
readMMIO st Reg1 = ...
readMMIO st ... = ...
readMMIO _ UnknownReg = 0 -- For unknown values return 0

-- But this always return 0 in HDL (but not simulation)
readMMIO' :: MMIOState -> MMIOReg -> Unsigned 8
readMMIO' _ UnknownReg = 0 -- For unknown values return 0
readMMIO' st Reg0 = ...
readMMIO' st Reg1 = ...
readMMIO' st ... = ...
```

### Proposed behaviour

Since both behaviors seems to have use-cases probably it would be best:

 - Always include `errorX` as last case (user cannot do it without having dead code warning from GHC)
 - On pattern match don't check for bit representation only but check for all previous states. In case of `readMMIO' _ UnknownReg = 0` it would lower to `readMMIO' _ r | not (reg0Pattern r || reg1Pattern r || ...) = 0; | otherwise = ...`

christiaanb, 2020-11-29:
I'm trying to reproduce your first example with:
```haskell
module Test where

import Clash.Prelude

data Foo = F0 | F1 | F2 deriving (Generic, BitPack)

f :: Foo -> Unsigned 2
f F0 = 0
f F1 = 1
f F2 = 2
{-# NOINLINE f #-}

g :: Foo -> Unsigned 2
g F2 = 2
g F1 = 1
g F0 = 0
{-# NOINLINE g #-}

f' :: BitVector 2 -> Unsigned 2
f' = f . unpack
{-# NOINLINE f' #-}
{-# ANN f' (defSyn "fp") #-}

g' :: BitVector 2 -> Unsigned 2
g' = g . unpack
{-# NOINLINE g' #-}
{-# ANN g' (defSyn "gp") #-}
```
And then run:
```
clash -- --vhdl Test.hs
```
But I don't see any differences except in the types package, and the name of the entity:
```
$ diff vhdl/Test/fp/fp.vhdl vhdl/Test/gp/gp.vhdl 
8c8
< use work.fp_types.all;
---
> use work.gp_types.all;
10c10
< entity fp is
---
> entity gp is
15c15
< architecture structural of fp is
---
> architecture structural of gp is
18c18
<   signal \c$case_alt\             : fp_types.clash_internal;
---
>   signal \c$case_alt\             : gp_types.clash_internal;
21c21
<   signal \c$case_alt_0\           : fp_types.clash_internal_0;
---
>   signal \c$case_alt_0\           : gp_types.clash_internal_0;
23,24c23,24
<   signal \c$case_alt_1\           : fp_types.foo;
<   signal result_0                 : fp_types.foo;
---
>   signal \c$case_alt_1\           : gp_types.foo;
>   signal result_0                 : gp_types.foo;
29c29
<   signal \c$case_alt_1_selection\ : fp_types.clash_internal_0;
---
>   signal \c$case_alt_1_selection\ : gp_types.clash_internal_0;
68c68
<   f_0_result : entity f_0
---
>   g_0_result : entity g_0
```
```
$ diff vhdl/Test/fp/f_0.vhdl vhdl/Test/gp/g_0.vhdl 
8c8
< use work.fp_types.all;
---
> use work.gp_types.all;
10,11c10,11
< entity f_0 is
<   port(ds     : in fp_types.foo;
---
> entity g_0 is
>   port(ds     : in gp_types.foo;
15c15
< architecture structural of f_0 is
---
> architecture structural of g_0 is
```
What version of GHC and Clash are you using? I'm using ghc 8.10.1 and clash 1.2.5

uzytkownik, 2020-11-29:
@christiaanb You're right - I got confused reading the output most likely.

christiaanb, 2020-11-30:
I'll leave the issue open as "enhancement", as an explicit "fall-through" to a call to `errorX` might lead the downstream logic synthesis to a better circuit as you mentioned.

uzytkownik, 2020-11-30:
@christiaanb Sounds good. Though I'd note that depending on use-case one might want to have catch-all field to catch unparsed input.