clash-lang/clash-compiler #1606: Error with Vectors of Clock Reset and Enable

jacquescomeaux, 2020-12-01:
I encountered a runtime error in Clash when trying to synthesize VHDL for a larger project and managed to recreate it with this example. Trying to synthesize `functionA` causes the list-index operator (`Prelude.!!`) to be called with an index that is too large. The offending `!!` is in  the `ArgGen` case of `Clash.Netlist.BlackBox.Util.renderTag` in `clash-lib`. Meanwhile `functionB` can be synthesized, and `functionC` fails with a slightly more informative error.

Is it a normal usage of Clash to pass around Clock, Reset, and Enable signals explicitly like this as the results of functions?

```haskell
{-# LANGUAGE NoImplicitPrelude     #-}
module Test where

import Clash.Prelude

data CRE = CRE (Clock System) (Reset System) (Enable System)

{-# ANN functionA (defSyn "FunctionA") #-}
{-# ANN functionB (defSyn "FunctionB") #-}
{-# ANN functionC (defSyn "FunctionC") #-}

functionA
    :: Clock System
    -> Reset System
    -> Enable System
    -> Vec 2 (Signal System Bool, CRE)
functionA = \ c r e ->
  let
    term   = CRE c r e
    vec    = term :> term :> Nil
    result = fmap (\a -> (pure True, a)) vec
  in
    result 

functionB
    :: Clock System
    -> Reset System
    -> Enable System
    -> Vec 2 CRE
functionB = \ c r e ->
  let
    term   = CRE c r e
    result = term :> term :> Nil
  in
    result

functionC
    :: Clock System
    -> Reset System
    -> Enable System
    -> Vec 2 CRE
functionC = \ c r e ->
  let
    term   = CRE c r e
    vec    = term :> term :> Nil
    result = fmap id vec
  in
    result
```

jacquescomeaux, 2020-12-01:
Actually, I didn't think to try isolating further by testing out just clocks, resets, or enables by themselves in a vector

christiaanb, 2020-12-01:
Ah, interesting. There are a couple of things going on here which result in this bug:

1. Certain higher-order functions over `Vec`, such as `map`, have specialized code-paths to turn them into generate-for loops in VHDL, instead of having to having to unroll/inline their recursive definitions: https://github.com/clash-lang/clash-compiler/blob/e6738d181690c22e931779f47ad058eb50c266f4/clash-lib/prims/vhdl/Clash_Sized_Vector.json#L102-L128
2. Clash, in general, translates Haskell product types (like your `CRE` type) to VHDL records. This mostly works out fine, there is however one exception: certain synthesis tools, and some HDL simulation tools (like `verilator`), do not like it when the clock (and certain other global control signals) is contained in a record type; they want them to be separate inputs to the `entity`/`module`. And Clash actually does some transformations to try to ensure that values of type `Clock` do not end up in a VHDL record type.

The problem is that the transformations in 2. never took into account the specialized code-paths in 1. We/I will have to think about what the best way forward is.

christiaanb, 2020-12-01:
Also, looking at the code for the transformations of 2., I see that it is actually missing a check for "recursive" product types such as `Vec`.

commit 95da304e (2021-03-28): Unroll primitives for splittable types

Fixes #1606
commit 11b35fdc (2021-03-28): Unroll vector primitives for splittable types

Fixes #1606
commit 07b6cf72 (2021-03-28): Unroll vector primitives for splittable types

Fixes #1606
commit b1baec9d (2021-03-30): Unroll vector primitives for splittable types

Fixes #1606
closed 2021-03-31 by christiaanb