clash-lang/clash-compiler #1609: Hit Specialization Limit (Again)

paddytheplaster, 2020-12-07:
Hello again,

I am getting the following error, which I'm getting because I'm trying to synthesise my design for slightly larger `Nat` sizes than before.

    Hit specialisation limit 100 on function `cPaddy.paddy[1297153]'.

    The function `cpaddy[1297153] :: Clash.Sized.Vector.Vec[8214565720323789728]
                                                256
                                                GHC.Types.Bool[3674937295934324744]
                                          -> forall (f[8286623314361998616] :: GHC.Types.Nat[3674937295934325064])
                                          (g[8286623314361998617] :: GHC.Types.Nat[3674937295934325064]).
                                          GHC.Types.~[3674937295934324816]
    [40 lines omitted]

    Body of `Paddy.paddy[1297153] :: Clash.Sized.Vector.Vec[8214565720323789728]
                                                256
    [357 lines omitted]

    Argument (in position: 1) that triggered termination:
    52
    Run with '-fclash-spec-limit=N' to increase the specialisation limit to N.

    The source location of the error is not exact, only indicative, as it is acquired 
    after optimizations. The actual location of the error can be in a function that is 
    inlined. To prevent inlining of those functions, annotate them with a NOINLINE pragma.
    Regards,

I'm already running with `-fclash-spec-limit=100`.

I've had this error _numerous_ times and I am pretty sure I've reported this error at least once. Unfortunately, I can't find the report.

This error shows up after two `clash` runs: one to generate the top entity, and one `--vhdl` session. Together the two runs take more than one minute and it's _difficult_ to simplify the source to a small example in a reasonable amount of time. (I've spent days doing this before. Usually I get different errors for smaller examples, and it soon becomes impossible to keep track of what's going on.)

Any advice on how to proceed?

TIA/Regards,

Paddy

christiaanb, 2020-12-07:
The specialization limit is there to prevent the compiler to go into an infinite specialization loop for certain recursive functions. The current check only looks at the number of times a function is specialised on a certain argument, but not whether the function is recursive (or part of a recursive group of functions). The original assumption was that only recursive functions are specialised so many times on different arguments, but that's probably a wrong assumption to make.

I guess we could:
1. Extend the check so that the termination measure only triggers on recursive functions
2. Increase the default limit

It's always safe to increase the specialisation limit, insofar that it won't affect the correctness of the translation to HDL.  You can e.g. check if it compiles if you change the limit to 4000.

paddytheplaster, 2020-12-07:
> The specialization limit is there to prevent the compiler to go into an infinite specialization loop for certain recursive functions. The current check only looks at the number of times a function is specialised on a certain argument, but not whether the function is recursive (or part of a recursive group of functions). The original assumption was that only recursive functions are specialised so many times on different arguments, but that's probably a wrong assumption to make.
> 
> I guess we could:
> 
> 1. Extend the check so that the termination measure only triggers on recursive functions
> 2. Increase the default limit
> 
> It's always safe to increase the specialisation limit, insofar that it woHn't affect the correctness of the translation to HDL. You can e.g. check if it compiles if you change the limit to 4000.

Thanks. I'll try to increase the limit.

I can scrutinise my code for recursive functions but I am pretty much 100% certain I am not using recursive functions in my top entity. Having said that, I do have some recursive functions which I use in phase 1: generation of initial state of top entity. In the next phase (synthesis with `--vhdl`) I disable all recursive function definitions using conditional compilation. However, I may have overlooked some of these definitions. Should I still make sure _any_ recursive function definition is hidden to `clash` even if it isn't used?

Regards,

Paddy

paddytheplaster, 2020-12-07:
> The specialization limit is there to prevent the compiler to go into an infinite specialization loop for certain recursive functions. The current check only looks at the number of times a function [...]

Not sure if this is possible, but IMO it'd be useful if `clash` had a flag which would trigger warnings about recursive user-defined functions. Understand this may throw some false positives. However it may also tell them about imported "standard" modules which provide recursively defined functions.

Regards,

Paddy

paddytheplaster, 2020-12-07:
> The specialization limit is there to prevent the compiler to go into an infinite specialization loop for certain recursive functions. The current check only looks at the number of times a function [...]

Relief. The error disappears if I define `-fclash-spec-limit=400`. I'll set it to `4000` from now on.

Thanks again.

Regards,

Paddy

christiaanb, 2020-12-07:
You can define and use recursive functions, it's just that Clash must be able to terminate on their use. Sadly, checking whether a function belongs to that set of functions is inpossible to check, and also hard to define.

Basically, Clash has to be able to fully unroll the recursive definition at compile-time in order for it to translate it to HDL. e.g. by itself the function:
```
foo [] = 0
foo (x:xs) = x + foo xs
```
cannot be fully unrolled at compile-time; but `foo` specialized on a particular argument, `foo [x,y,z]`, can be fully unrolled. It's not easy to formalize this, but hopefully one builds up an intuition for it.

paddytheplaster, 2020-12-08:
> You can define and use recursive functions, it's just that Clash must be able to terminate on their use. Sadly, checking whether a function belongs to that set of functions is inpossible to check, [...]

Thanks for explaining. I didn't know about the unrolling specialisations.

It's still not clear what causes the problem with the specialisation limit for my source files. I checked all sources for recursive function definitions but I couldn't find any.

Using conditional compilation I've now disabled most, if not all, irrelevant function definitions for the different phases in the compilation process. E.g. functions which are only needed for testing and evaluation are now completely disabled for the synthesis phase (clash `--vhdl`).

Compilation is now faster and synthesis now _succeeds_ with a _lower_ `-fclash-spec-limit`, whereas before it failed with a higher limit.

Could dead code optimisation be possible explanation of the problem?

Regards,

Paddy

paddytheplaster, 2020-12-08:
> You can define and use recursive functions, it's just that Clash must be able to terminate on their use. Sadly, checking whether a function belongs to that set of functions is inpossible to check, [...]

  * How does one decide which external libraries are fine to use in combination with Clash? E.g. some of them may define recursive functions for which Clash doesn't have patterns which allows it to unroll the recursion at compile time. Is there a list of trusted libraries, which can be used safely in combination with Clash? If not, wouldn't it be a good idea to create such a list?
  * Where can I find out more about how one defines patterns which allow Clash to unroll recursive functions at compile-time, please?

Regards,

Paddy

alex-mckenna, 2020-12-08:
It's not a case of _this recursive function is okay to use_ per se, it's more a case of _how_ it's used within a design. If we take a function like `foo` above, you cannot have just that as your design because there is no way to know how many times it has to be unrolled (and thus how much logic needs to be generated). We can still write / use recursive functions, but ultimately you have to be able to determine how many times to unroll it. e.g.

```haskell
topEntity = foo [1..100]
```

would render fine, because we know how many times `foo` is to be rendered. Types in Clash prelude can also get around this by having the size in the type (like `map f (xs :: Vec 5 Int)` would be okay to render, because we can determine how many times to unroll `map` from the type). A good way to think of it is "could I inline every recursive function in my topentity without needing to inline something infinitely many times", which as @christiaanb said is something you build an intuition for.

(Hopefully this is all correct, I'm sure Christiaan will correct any mistakes).

paddytheplaster, 2020-12-09:
> It's not a case of _this recursive function is okay to use_ per se, it's more a case of _how_ it's used within a design. If we take a function like `foo` above, you cannot have just that as your [...]

Thanks Alex. This makes sense.

Regards,

Paddy