clash-lang/clash-compiler #1610: Curious Warning about -fworker-wrapper

paddytheplaster, 2020-12-07:
Hi again,

I am running `clash --vhdl` with optimitsation (`-O`) and `-fno-worker-wrapper` enabled. Howere I still am getting a warning that it's _enabled._ Am I doing something wrong?

The following is the nub of how I run `clash --vhdl`.

    nice -n 19 clash -i/home/paddy/Altera/Clash/src \
                     # other -i flags omitted
                     -odir /home/paddy/Altera/Clash/obj \
                     -tmpdir /home/paddy/Altera/Clash/ghc-tmp/ghc \
                     -hidir /run/user/1000/hi-dir \
                     -j4 \
                     -fclash-no-cache \
                     -fclash-spec-limit=200 \
                     -fclash-hdldir /home/paddy/Altera/Clash/vhdl/DE0_NANO \
                     -fclash-hdlsyn Quartus \
                     -fno-worker-wrapper \
                     -O \
                     --vhdl /home/paddy/Altera/Clash/tops/TopEntity.lhs

    WARNING:
    `-fworker-wrapper` option is globally enabled, this can result in incorrect code.
    Are you compiling with `-O` or `-O2`? Consider adding `-fno-worker-wrapper`.
    `-fworker-wrapper` can be use in a diligent manner on a file-by-file basis
    by using a `{-# OPTIONS_GHC -fworker-wrapper` #-} pragma.

TIA/Regards,

Paddy

christiaanb, 2020-12-07:
The order of the flags matters. Do you get the warning if you reverse the position of `-O` and `-fno-worker-wrapper`?

paddytheplaster, 2020-12-07:
> The order of the flags matters. Do you get the warning if you reverse the position of `-O` and `-fno-worker-wrapper`?

Thanks. Should have thought of that. Using `-O` as the first flag solves the problem.

Thanks again,

Paddy

closed 2020-12-07 by martijnbastiaan