clash-lang/clash-compiler #1611: Clash inliner just can't even with this "benignly recursive" code

gergoerdi, 2020-12-08:
While being blocked on #1536, I started working on a new design for my memory mapper, [based on the idea of tracking backpane signals separately](https://stackoverflow.com/q/65179983/477476). The new version is... [somewhat complicated in its API](https://github.com/gergoerdi/retroclash-lib/blob/memory-map-backpane/src/RetroClash/Memory2.hs) since it uses a graded monad to track the backpane connections. But, it has **no knot-tying-like shenanigans** going on -- the only recursion is structurally into the `Addressing` tree, so it should be very nicely bounded for small memory layout descriptions.

However, [when I try using it](https://github.com/gergoerdi/clash-tinybasic/blob/memory-map-backpane/src/Hardware/TinyBASIC/Intel8080.hs), it still seems that Clash isn't able to synthesize it: 

```
Clash.Normalize.Transformations(432): InlineNonRep: RetroClash.Memory2.$fMonoidFanIn_z[8214565720323925287] already inlined 40000 times in:Hardware.TinyBASIC.Intel8080.logicBoard[4876]
Type of the subject is: Clash.Signal.Internal.Signal[8214565720323789676]
  "Native"
  (Data.Monoid.First[8214565720323790477]
     a[8214565720324055698])
Function Hardware.TinyBASIC.Intel8080.logicBoard[4876] will not reach a normal form, and compilation might fail.
Run with '-fclash-inline-limit=N' to increase the inlining limit to N.
```

Full code is at https://github.com/gergoerdi/clash-tinybasic/tree/memory-map-backpane.

christiaanb, 2020-12-08:
Thanks for the report. At the moment we don't have the resources to minimize the example, and this bug is unlikely to get fixed until there is a minimal(ized) example.

gergoerdi, 2020-12-08:
Sure, I'm already working on minimizing it.

gergoerdi, 2020-12-08:
Oh ho ho, turns out the minimized test for https://github.com/clash-lang/clash-compiler/issues/1536#issuecomment-740544378 actually is this in disguise: running with `-fclash-debug DebugSilent` and `-fclash-inline-limit=1000`, we can see that the normalizer chokes on this input, even though, conceptually, there isn't really that deep a recursion going on (since the `runAddressing` block is fairly small):

```
Clash.Normalize.Transformations(432): InlineNonRep: RetroClash.Memory.firstIn[8214565720324031151] already inlined 1000 times in:Board.logicBoard[398]
Type of the subject is: Clash.Signal.Internal.Signal[8214565720323789676]
  "System"
  (GHC.Maybe.Maybe[3674937295934324792]
     a[6989586621679260264])
Function Board.logicBoard[398] will not reach a normal form, and compilation might fail.
Run with '-fclash-inline-limit=N' to increase the inlining limit to N.
```

Full code is at https://github.com/gergoerdi/clash-issue-1611/tree/clash-issue-1611

gergoerdi, 2020-12-08:
Here's a fully self-contained way of reproducing the inliner choking. This version seems to run forever with an inliner limit of 1000 -- I killed it after 10 minutes, considering how little circuitry remains:

```
{-# LANGUAGE DataKinds, GADTs, PolyKinds, KindSignatures, TypeOperators #-}
{-# LANGUAGE RebindableSyntax, LambdaCase #-}
{-# LANGUAGE GeneralizedNewtypeDeriving, DerivingStrategies #-}
module Board where

import Clash.Prelude
import Clash.Annotations.TH

import Data.Kind
import Data.Singletons.Prelude.List (type (++))
import Control.Monad.RWS hiding (Product)

import Data.Maybe
import Data.Kind
import Data.Dependent.Map as DMap
import Data.Dependent.Sum as DSum
import Data.GADT.Compare
import Type.Reflection

data Component (addr :: Type) = Component (TypeRep addr) Int

instance GEq Component where
    geq (Component a _) (Component b _) = geq a b

instance GCompare Component where
    gcompare (Component a _) (Component b _) = gcompare a b

newtype FanIn dom a = FanIn{ getFanIn :: Signal dom `Ap` First a }
    deriving newtype (Semigroup, Monoid)

newtype AddrMap dom = AddrMap{ addrMap :: DMap Component (FanIn dom) }
    deriving newtype (Monoid)

instance Semigroup (AddrMap dom) where
    AddrMap map1 <> AddrMap map2 = AddrMap $ unionWithKey (const mappend) map1 map2

newtype Susp0 dom dat addr = Susp0{ unSusp0 :: Signal dom (Maybe addr) -> Signal dom (Maybe dat) }
newtype Susp1 w dom dat addr = Susp1{ unSusp1 :: Signal dom (Maybe addr) -> (Signal dom (Maybe dat), w) }

type Addressing0 dom addr dat = RWS
    (Signal dom (Maybe dat))
    (AddrMap dom)
    Int

data Addressing dom addr dat (ts :: [Type]) a where
    Return :: a -> Addressing dom addr dat '[] a
    Bind :: Addressing dom addr dat ts a -> (a -> Addressing dom addr dat us b) -> Addressing dom addr dat (ts ++ us) b

    Fresh :: (Typeable addr') => Addressing dom addr dat '[] (Component addr')
    WR :: Addressing dom addr dat '[] (Signal dom (Maybe dat))
    Match :: (addr -> Maybe addr') -> Addressing dom addr' dat ts a -> Addressing dom addr dat ts a
    Connect :: Component addr -> Addressing dom addr dat '[] ()
    Tell0 :: DSum Component (Susp0 dom dat) -> Addressing dom addr dat '[] ()
    Tell1 :: DSum Component (Susp1 w dom dat) -> Addressing dom addr dat '[w] ()

data Susps dom dat (ts :: [Type]) where
    NilS :: Susps dom dat '[]
    Cons0 :: DSum Component (Susp0 dom dat) -> Susps dom dat ts -> Susps dom dat ts
    Cons1 :: DSum Component (Susp1 t dom dat) -> Susps dom dat ts -> Susps dom dat (t:ts)

data Results (ts :: [Type]) where
    NilR :: Results '[]
    ConsR :: a -> Results ts -> Results (a : ts)

concatS :: Susps dom dat ts -> Susps dom dat us -> Susps dom dat (ts ++ us)
concatS = \case
    NilS -> id
[... 143 lines cut ...]
board
    :: (HiddenClockResetEnable dom)
    => Signal dom (Maybe (Unsigned 16))
    -> Signal dom (Maybe (Unsigned 8))
board addr = dataIn
  where
    wr = Just <$> cnt
      where
        cnt = register (0 :: Unsigned 8) $ cnt + 1

    (dataIn, (), NilR) = runAddressing addr wr $ do
        rom <- romFromFile (SNat @0x0800) "image.bin"
        ram <- ram0 (SNat @0x1800)

        from 0x0000 $ connect rom
        from 0x0800 $ connect ram
      where
        return = Return
        (>>=) = Bind
        (=<<) = flip (>>=)
        m >> n = m >>= \_ -> n

makeTopEntity 'topEntity
```
```

gergoerdi, 2020-12-08:
With the whole graded monad thing removed: 

```
{-# LANGUAGE DataKinds, GADTs, PolyKinds, KindSignatures, TypeOperators #-}
{-# LANGUAGE LambdaCase #-}
{-# LANGUAGE GeneralizedNewtypeDeriving, DerivingStrategies #-}
module Board where

import Clash.Prelude
import Clash.Annotations.TH

import Data.Kind
import Control.Monad.RWS hiding (Product)

import Data.Maybe
import Data.Kind
import Data.Dependent.Map as DMap
import Data.Dependent.Sum as DSum
import Data.GADT.Compare
import Type.Reflection

data Component (addr :: Type) = Component (TypeRep addr) Int

instance GEq Component where
    geq (Component a _) (Component b _) = geq a b

instance GCompare Component where
    gcompare (Component a _) (Component b _) = gcompare a b

newtype FanIn dom a = FanIn{ getFanIn :: Signal dom `Ap` First a }
    deriving newtype (Semigroup, Monoid)

newtype AddrMap dom = AddrMap{ addrMap :: DMap Component (FanIn dom) }
    deriving newtype (Monoid)

instance Semigroup (AddrMap dom) where
    AddrMap map1 <> AddrMap map2 = AddrMap $ unionWithKey (const mappend) map1 map2

newtype Susp0 dom dat addr = Susp0{ unSusp0 :: Signal dom (Maybe addr) -> Signal dom (Maybe dat) }
newtype Susp1 w dom dat addr = Susp1{ unSusp1 :: Signal dom (Maybe addr) -> (Signal dom (Maybe dat), w) }

type Addressing0 dom addr dat = RWS
    (Signal dom (Maybe dat))
    (AddrMap dom)
    Int

data Addressing dom addr dat a where
    Return :: a -> Addressing dom addr dat a
    Bind :: Addressing dom addr dat a -> (a -> Addressing dom addr dat b) -> Addressing dom addr dat b

    Fresh :: (Typeable addr') => Addressing dom addr dat (Component addr')
    WR :: Addressing dom addr dat (Signal dom (Maybe dat))
    Match :: (addr -> Maybe addr') -> Addressing dom addr' dat a -> Addressing dom addr dat a
    Connect :: Component addr -> Addressing dom addr dat ()
    Tell0 :: DSum Component (Susp0 dom dat) -> Addressing dom addr dat ()

instance Functor (Addressing dom addr dat) where
    fmap f m = m >>= \x -> return (f x)

instance Applicative (Addressing dom addr dat) where
    pure = Return
    ff <*> fx = ff >>= \f -> fmap f fx

instance Monad (Addressing dom addr dat) where
    return = Return
    (>>=) = Bind

type Susps dom dat = [DSum Component (Susp0 dom dat)]

runAddressing1
[... 116 lines cut ...]
    :: "CLK"   ::: Clock System
    -> "RESET" ::: Reset System
    -> "ADDR"  ::: Signal System (Maybe (Unsigned 16))
    -> "OUT"   ::: Signal System (Maybe (Unsigned 8))
topEntity clk rst = withClockResetEnable clk rst enableGen board

board
    :: (HiddenClockResetEnable dom)
    => Signal dom (Maybe (Unsigned 16))
    -> Signal dom (Maybe (Unsigned 8))
board addr = dataIn
  where
    wr = Just <$> cnt
      where
        cnt = register (0 :: Unsigned 8) $ cnt + 1

    (dataIn, ()) = runAddressing addr wr $ do
        rom <- romFromFile (SNat @0x0800) "image.bin"
        ram <- ram0 (SNat @0x1800)

        from 0x0000 $ connect rom
        from 0x0800 $ connect ram

makeTopEntity 'topEntity
```

gergoerdi, 2020-12-08:
So to summarize, the above code, which defines `dataIn` via `runAddressing` which is recursive over a pure, finite data structure, causes Clash to take a comparatively very long time until it actually decides that it can't inline anymore. @christiaanb is this a small enough way of reproducing the problem?

gergoerdi, 2020-12-08:
If I comment out `rom` from the definition of `dataIn` in `board`, I get to the inliner limit break message MUCH faster!

christiaanb, 2020-12-08:
Thanks, I'll check it out.

christiaanb, 2020-12-08:
Got my first interesting hint while compiling with `-fclash-debug DebugSilent`:

```
Irreducible constant as case subject: GHC.Types.I# 0
Can be reduced to: GHC.Types.I# 0
```
Where I'm actually forgetting why the `GHC.Types.I#` constructor is tagged as a primitive.

alex-mckenna, 2020-12-08:
`GHC.Types` is in `ghc-prim`, is it because it's originally a prim in the GHC core and we just convert it to a Clash core prim directly?

christiaanb, 2020-12-08:
What's also weird is that there is an evaluator rule, but it's apparently not triggered. I changed it to:
```
  "GHC.Types.I#"
    | isSubj
    , [Lit (IntLiteral i)] <- args
    ->  let (_,tyView -> TyConApp intTcNm []) = splitFunForallTy ty
            (Just intTc) = lookupUniqMap intTcNm tcm
            [intDc] = tyConDataCons intTc
        in  reduce (mkApps (Data intDc) [Left (Literal (IntLiteral i))])
    | otherwise
    -> error (show args)
```
But it never throws the error.

alex-mckenna, 2020-12-08:
That is strange, is the primitive evaluator being called at all?

christiaanb, 2020-12-08:
Perhaps it would be if my `cabal.project.local` wouldn't have `+experimental-evaluator` in some places...

alex-mckenna, 2020-12-08:
Oh yes, that would certainly ruin your day. At least you can delete those flags for good now given they don't exist in the PE branch

christiaanb, 2020-12-08:
Alright, I've added some addition instrumentation so that Clash reports whether it's compile-time evaluator is missing a rewrite-rule for "primitives". The ones that are reported are:

```
No reduction rule for: "Clash.Promoted.Symbol.SSymbol" :: forall (s[6989586621679129497] :: GHC.Types.Symbol[3674937295934325066]).
GHC.TypeLits.KnownSymbol[3602879701896396843] s[6989586621679129497]
-> Clash.Promoted.Symbol.SSymbol[8214565720323787975] s[6989586621679129497]

No reduction rule for: "Data.Typeable.Internal.$wmkTrCon" :: forall (k[6989586621679127546] :: GHC.Prim.TYPE[3674937295934324912]
                                    GHC.Types.LiftedRep[3891110078048108766])
(a[6989586621679127547] :: k[6989586621679127546]).
GHC.Prim.Word#[3674937295934324854]
-> GHC.Prim.Word#[3674937295934324854]
-> GHC.Types.Module[3891110078048108670]
-> GHC.Types.TrName[3891110078048108676]
-> GHC.Prim.Int#[3674937295934324764]
-> GHC.Types.KindRep[3891110078048108688]
-> GHC.Types.[][3674937295934324788]
     Data.Typeable.Internal.SomeTypeRep[3674937295934325112]
-> GHC.Prim.(#,,,,#)[3819052484010180618]
     GHC.Types.WordRep[3891110078048108787]
     GHC.Types.WordRep[3891110078048108787]
     GHC.Types.LiftedRep[3891110078048108766]
     GHC.Types.LiftedRep[3891110078048108766]
     GHC.Types.LiftedRep[3891110078048108766]
     GHC.Prim.Word#[3674937295934324854]
     GHC.Prim.Word#[3674937295934324854]
     GHC.Types.TyCon[3891110078048108664]
     (GHC.Types.[][3674937295934324788]
        Data.Typeable.Internal.SomeTypeRep[3674937295934325112])
     (Data.Typeable.Internal.TypeRep[3674937295934325110]
        (GHC.Prim.TYPE[3674937295934324912] GHC.Types.LiftedRep[3891110078048108766])
        k[6989586621679127546])
```
The first one is an oversight on our end. 

The second one is a bit more worrying, as that looks like some internal Typeable function to build up a TypeRep. We'll have to figure out how easy it is to recreate that function in our Core AST.

alex-mckenna, 2020-12-08:
> The first one is an oversight on our end.

https://github.com/clash-lang/clash-compiler/blob/a54c6b6d4f5aaf4e616468fa5f729f3d9e28fa6d/clash-ghc/src-ghc/Clash/GHC/PartialEval/Primitive/Promoted.hs#L36

Somehow I managed to implement this anyway, I must've seen it in my "missing primitive" warning message for PE. As for the second, I've also been seeing it in PE (and ignoring it) to seemingly no negative effect. Are you sure this is the problem?

christiaanb, 2020-12-08:
No, I’m not sure it is the problem. My analysis is 100% speculation.

christiaanb, 2020-12-09:
Okay, I made a somewhat smaller `topEntity`:
```haskell
topEntity
  :: Clock System
     -> Reset System
     -> Signal System (Maybe (Unsigned 8))
     -> Signal System (Maybe (Unsigned 16))
     -> (Signal System (Maybe (Unsigned 16)), ())
topEntity clk rst addr wr =
    withClockResetEnable clk rst enableGen $ runAddressing addr wr $ do
        () <- return ()
        return ()
```
that also hits the inlining limit because there is no Core unfolding for `GHC.Base.++`:
````
No reduction rule for: "GHC.Base.++" :: forall (a[6989586621679125296] :: GHC.Prim.TYPE[3674937295934324912]
                                    GHC.Types.LiftedRep[3891110078048108766]).
GHC.Types.[][3674937295934324788]
  a[6989586621679125296]
-> GHC.Types.[][3674937295934324788]
     a[6989586621679125296]
-> GHC.Types.[][3674937295934324788]
     a[6989586621679125296]
Irreducible constant as case subject: GHC.Base.++
  @(Data.Dependent.Sum.DSum[8214565720323789109]
      (GHC.Prim.TYPE[3674937295934324912]
         GHC.Types.LiftedRep[3891110078048108766])
      Test.Component[8214565720323886481]
      (Test.Susp0[8214565720323886472]
         "System"
         (Clash.Sized.Internal.Unsigned.Unsigned[8214565720323788081]
            16)))
  <prefixName>"runAddressing_ds"
  <prefixName>"runAddressing1"
  (GHC.Types.[][3891110078048108577]
     @(Data.Dependent.Sum.DSum[8214565720323789109]
         (GHC.Prim.TYPE[3674937295934324912]
            GHC.Types.LiftedRep[3891110078048108766])
         Test.Component[8214565720323886481]
         (Test.Susp0[8214565720323886472]
            "System"
            (Clash.Sized.Internal.Unsigned.Unsigned[8214565720323788081]
               16))))
  <prefixName>"runAddressing_ds2"
  <prefixName>"runAddressing1"
  (GHC.Types.[][3891110078048108577]
     @(Data.Dependent.Sum.DSum[8214565720323789109]
         (GHC.Prim.TYPE[3674937295934324912]
            GHC.Types.LiftedRep[3891110078048108766])
         Test.Component[8214565720323886481]
         (Test.Susp0[8214565720323886472]
            "System"
            (Clash.Sized.Internal.Unsigned.Unsigned[8214565720323788081]
               16))))
Can be reduced to: GHC.Base.++
  @(Data.Dependent.Sum.DSum[8214565720323789109]
      (GHC.Prim.TYPE[3674937295934324912]
         GHC.Types.LiftedRep[3891110078048108766])
      Test.Component[8214565720323886481]
      (Test.Susp0[8214565720323886472]
         "System"
         (Clash.Sized.Internal.Unsigned.Unsigned[8214565720323788081]
            16)))
  (GHC.Types.[][3891110078048108577]
     @(Data.Dependent.Sum.DSum[8214565720323789109]
         (GHC.Prim.TYPE[3674937295934324912]
            GHC.Types.LiftedRep[3891110078048108766])
         Test.Component[8214565720323886481]
         (Test.Susp0[8214565720323886472]
            "System"
            (Clash.Sized.Internal.Unsigned.Unsigned[8214565720323788081]
               16))))
  (GHC.Types.[][3891110078048108577]
     @(Data.Dependent.Sum.DSum[8214565720323789109]
         (GHC.Prim.TYPE[3674937295934324912]
            GHC.Types.LiftedRep[3891110078048108766])
         Test.Component[8214565720323886481]
         (Test.Susp0[8214565720323886472]
            "System"
            (Clash.Sized.Internal.Unsigned.Unsigned[8214565720323788081]
               16))))
```

christiaanb, 2020-12-09:
Adding:
```
appendSusps :: Susps dom dat -> Susps dom dat -> Susps dom dat
appendSusps []     ys = ys
appendSusps (x:xs) ys = x : appendSusps xs ys
```
and changing:
```
runAddressing1 addr = \case
    Return x -> nilS $ return x
    Bind m n -> do
        (rd1, x) <- runAddressing1 addr m
        (rd2, y) <- runAddressing1 addr $ n x
        return (appendSusps rd1 rd2, y)
```
makes
```haskell
topEntity
  :: Clock System
     -> Reset System
     -> Signal System (Maybe (Unsigned 8))
     -> Signal System (Maybe (Unsigned 16))
     -> (Signal System (Maybe (Unsigned 16)), ())
topEntity clk rst addr wr =
    withClockResetEnable clk rst enableGen $ runAddressing addr wr $ do
        () <- return ()
        return ()
```
succeed

christiaanb, 2020-12-09:
It does seem that we would eventually need `$wmkTrCon`, because when I change `runAddressing` to:
```haskell
runAddressing addr wr body = (join <$> firstIn rd, x)
  where
    ((susps, x), conns) = evalRWS (runAddressing1 addr body) wr 0
    rd = toRead [(Component (typeRep @ Int) 0) :=> Susp0 (const (pure Nothing))] 
                (DMap.singleton (Component (typeRep @ Int) 0) (fanIn (pure 0)))
```
It finishes normalization, but I get the following netlist level error:
```
Test.hs:238:1: error:
    
    Clash.Netlist.BlackBox(291): No blackbox found for: Data.Typeable.Internal.$wmkTrCon. Did you forget to include directories containing primitives? You can use '-i/my/prim/dir' to achieve this.
    
    The source location of the error is not exact, only indicative, as it is acquired 
    after optimizations. The actual location of the error can be in a function that is 
    inlined. To prevent inlining of those functions, annotate them with a NOINLINE pragma.
    |
238 | topEntity clk rst addr wr =
    | ^^^^^^^^^
```

gergoerdi, 2020-12-09:
I am compiling `dependent-map` with `-fno-worker-wrapper`, but of course `Typeable` itself comes from `base` so that's not helping...

I wonder if the correct solution wouldn't be to have Clash always compile everything with `-fno-worker-wrapper` and `-fexpose-all-unfoldings` **and also** present itself as a standalone Haskell compiler, i.e.  not sharing any compiled objects/`.hi` files/`.so`s with GHC-compiled libraries. So as far as Cabal, Stack, etc. would be concerned, Clash would be just as independent of GHC as, e.g., Hugs or AJHC.

gergoerdi, 2020-12-11:
> Alright, I've added some additional instrumentation so that Clash reports whether it's compile-time evaluator is missing a rewrite-rule for certain "primitives".

Can you push that to a branch? I'd like to try it to the code in https://github.com/clash-lang/clash-compiler/issues/1536#issuecomment-741687988

christiaanb, 2020-12-11:
It's https://github.com/clash-lang/clash-compiler/tree/prim_instrumentation, commit https://github.com/clash-lang/clash-compiler/commit/e7e161170899681a01316f00e892733e20fe62a9