clash-lang/clash-compiler #1612: Clash.Annotations.TH.makeTopEntity doesn't play nice with implicit parameters

JvWesterveld, 2020-12-09:
From the documentation of [`Clash.Annotations.TH.makeTopEntity`](http://hackage.haskell.org/package/clash-prelude-1.2.5/docs/Clash-Annotations-TH.html#v:makeTopEntity):
_The function arguments and return values of the function specified by the given `Name` must be annotated with `(:::)`. This annotation provides the given name of the port._

This doesn't seem to work as one might expect with implicit parameters:
```
import Clash.Prelude
import Clash.Annotations.TH
topEntity
  :: ( "clk" ::: HiddenClock System
     , "rst" ::: HiddenReset System
     , "en"  ::: HiddenEnable System
     )
  => "input" ::: Signal System Bit
  -> "output" ::: Signal System Bit
topEntity = register 0
makeTopEntity 'topEntity
```
compiles to VHDL with entity
```
entity topEntity is
  port(-- clock
       input      : in topentity_types.clk_system;
       -- reset
       \$d(%,%)1\ : in topentity_types.rst_system;
       \$d(%,%)2\ : in boolean;
       s          : in std_logic;
       output     : out std_logic);
end;
```
Tested on 1.2.5 and today's git master.

martijnbastiaan, 2020-12-09:
Hmm, a quick look at https://hackage.haskell.org/package/template-haskell-2.16.0.0/docs/Language-Haskell-TH.html#t:Type seems to suggest we couldn't even support it if we wanted to with TH as constraints are missing. We should just implement this in the compiler itself.

The documentation of `makeTopEntity`does indicate that this is explicitly not supported, so I'm marking this as `enhancement` instead of `bug`.

[related issue](https://github.com/clash-lang/clash-compiler/issues/1208).

martijnbastiaan, 2020-12-09:
On second glance, `ImplicitParamT` is in there, so it might work as long as all the `Hidden` stuff is transparant?

JvWesterveld, 2020-12-09:
> The documentation of `makeTopEntity`does indicate that this is explicitly not supported, so I'm marking this as `enhancement` instead of `bug`.

You're right, I should have looked better!