clash-lang/clash-compiler #1617: clashi/ghci experience "the impossible" while using TemplateHaskell

danielsmw, 2020-12-18:
Hi,

I'm trying to just write a simple CORDIC in clash, but I wanted the number of iterations and bitwidth to both be flexible. Here, I'm trying to generate the many (constant) values of arctan(1/2^j) needed for that algorithm. I thought that would be easier in clash than it actually seems, but please let me know if I'm missing something obvious.

I couldn't figure out how to make bitwidth or iterations truly flexible, so I decided to generate a large data structure holding a reasonably large number of reasonably high bitwidth values. Here's what I tried:

```
alphaVector64 :: Vec 16 (UFixed 1 63)
alphaVector64 = $(listToVecTH
                     [ ($$(fLit $ atan $ 1.0 / (2.0 ** fromIntegral j)) :: UFixed 1 63)
                     | j <- [0..15]
                     ] )
```

This code actually works fine if I write the lines out manually, but adding the list comprehension landed me with the following error emitted from clashi:

```
[1 of 1] Compiling Main             ( qsclash.hs, interpreted )
clashi: panic! (the 'impossible' happened)
  (GHC version 8.6.5 for x86_64-unknown-linux):
	nameModule
  internal j_ahO6
  Call stack:
      CallStack (from HasCallStack):
        callStackDoc, called at compiler/utils/Outputable.hs:1160:37 in ghc:Outputable
        pprPanic, called at compiler/basicTypes/Name.hs:240:3 in ghc:Name

Please report this as a GHC bug:  http://www.haskell.org/ghc/reportabug
```

I see, of course, that this claims to be a GHC bug, but for obvious reasons I thought it was likely worth reporting here first. If this is simply disallowed behavior, let me know, but it seems like someone (either clash or base GHC) should be handling it more gracefully.

christiaanb, 2020-12-18:
It does seem like a GHC bug. This already fails loading it in `ghci` (not `clashi`):
```haskell
{-# LANGUAGE TemplateHaskell, DataKinds #-}

module Test2 where

import Clash.Prelude

alphaVector64 :: Vec 1 (UFixed 1 63)
alphaVector64 = $(listToVecTH
                     [ $$(fLit j) :: UFixed 1 63
                     | j <- [0.0 :: Double]
                     ] )
```
giving:
```
[1 of 1] Compiling Test2            ( Test2.hs, interpreted )
ghc: panic! (the 'impossible' happened)
  (GHC version 8.10.2:
        nameModule
  internal j_ad7e
  Call stack:
      CallStack (from HasCallStack):
        callStackDoc, called at compiler/utils/Outputable.hs:1179:37 in ghc:Outputable
        pprPanic, called at compiler/basicTypes/Name.hs:251:3 in ghc:Name

Please report this as a GHC bug:  https://www.haskell.org/ghc/reportabug

*Test2> 
```

danielsmw, 2021-01-05:
Ah, I see. Okay then, I guess we can close this and I'll report it to GHC sometime soon. Thanks!

closed 2021-01-05 by danielsmw