clash-lang/clash-compiler #1619: Tutorial No longer works?

mayl, 2020-12-19:
I've been trying to go through the tutorial [here](https://github.com/clash-lang/clash-compiler/blob/master/docs/getting-started/first-circuit.rst#L41), but the synchronous MAC section from that tutorial (below) does not compile for me:
```
mac :: (Num a) => a -> (a, a) -> (a, a)
mac acc (x, y) = (acc + x * y, acc)

macS :: (HiddenClockResetEnable dom, Num a) => Signal dom (a, a) -> Signal dom a
macS = mealy mac 0
```

When I run this I get the following error:
<pre>[1 of 1] Compiling MAC              ( MAC.hs, interpreted )

<b>MAC.hs:10:8: </b><font color="#FF5555"><b>error:</b></font>
<b>    * Could not deduce (NFDataX a) arising from a use of `mealy&apos;</b>
<b>      from the context: (HiddenClockResetEnable dom, Num a)</b>
<b>        bound by the type signature for:</b>
<b>                   macS :: forall (dom :: Symbol) a.</b>
<b>                           (HiddenClockResetEnable dom, Num a) =&gt;</b>
<b>                           Signal dom (a, a) -&gt; Signal dom a</b>
<b>        at MAC.hs:9:1-79</b>
<b>      Possible fix:</b>
<b>        add (NFDataX a) to the context of</b>
<b>          the type signature for:</b>
<b>            macS :: forall (dom :: Symbol) a.</b>
<b>                    (HiddenClockResetEnable dom, Num a) =&gt;</b>
<b>                    Signal dom (a, a) -&gt; Signal dom a</b>
<b>    * In the expression: mealy mac 0</b>
<b>      In an equation for `macS&apos;: macS = mealy mac 0</b>
<font color="#5555FF"><b>   |</b></font>
<font color="#5555FF"><b>10 |</b></font> macS = <font color="#FF5555"><b>mealy mac 0</b></font>
<font color="#5555FF"><b>   |</b></font><font color="#FF5555"><b>        ^^^^^^^^^^^</b></font>
Failed, no modules loaded.
</pre>

I am a *complete* haskell/hardware novice so hopefully this is a a real issue and not just noise, my apologies if not :)

christiaanb, 2020-12-19:
Thanks for the report, it seems we missed that. That second definition should be:
```haskell
macS :: (HiddenClockResetEnable dom, Num a, NFDataX a) => Signal dom (a, a) -> Signal dom a
macS = mealy mac 0
```

mayl, 2020-12-19:
Thanks, your suggested solution works for me!

christiaanb, 2020-12-19:
We'll close it once the tutorial is updated

martijnbastiaan, 2021-01-02:
We might want to use something like [markdown-unlit](https://github.com/sol/markdown-unlit) to prevent future regressions.

PR #1627 (2021-01-05): Fix broken example code
The one line fix for the example code is done, but there should
really be a way to regression test the code in the readthedocs so
we can tell when this happens again.

Fixes #1619.
closed 2021-01-05 by alex-mckenna