clash-lang/clash-compiler #1620: Problem eliminating Redundant Variable

paddytheplaster, 2020-12-25:
> {-# OPTIONS_GHC -fplugin GHC.TypeLits.Extra.Solver #-}
    > {-# OPTIONS -fconstraint-solver-iterations=0 #-}

Compiling the following fails to terminate. However, if you uncomment the commented lines, compiling will terminate. This is a bit odd as the variable `l` is only used to define itself.

I have a similar, more complex, case where I'd like to eliminate a redundant variable such as `l`...

Regards,

Paddy

    > module Example where
    > import Clash.Prelude
    > problem
    >   :: forall (d :: Nat)
    >             -- (l :: Nat)
    >   .  ( KnownNat d
    >      , 1 <= d
    >      -- , KnownNat l
    >      -- , l ~ (CLog 2 d)
    >      , CLog 2 d ~ (CLog 2 (2 ^ (CLog 2 d)))
    >      )
    >   => Vec d () -> ()
    > problem _ = ()
    >     where t2 = repeat @(2^ (CLog 2 d) - 1) Nothing

alex-mckenna, 2020-12-25:
Hi Paddy,

I've quickly tried this locally off `master`, and can't seem to reproduce. Some things:

  * You haven't provided a `topEntity`, so I tried using `topEntity = problem @2` which compiled without problems. With no topEntity it also passes through `clash` without problem for me
  * You didn't specify the command you were using to compile, so I tried `clash --vhdl Issue1620.hs` and `clash --verilog Issue1620.hs`

Can you try locally with (1) a more recent version of `clash` if you are using an older version and (2) the `-ddump-tc-trace` flag to see if the loop is caused by the typechecker plugins or something else

Alex

paddytheplaster, 2020-12-25:
Hi Alex,

Thanks for looking at it.

That's odd. I'm using the most recently available clash from GitHub (Clash, version 1.3.0 (using clash-lib, version: 1.3.0).
I compile with `clash -fno-code <SOURCE>` or with `clash <SOURCE>`. If I remove the `OPTIONS` I get an error but if I keep them, it doesn't terminate (I just tried again before hitting the _comment_ button).

alex-mckenna, 2020-12-25:
Hi Paddy,

I've had another look and managed to reproduce it. Turns out I didn't read your first message closely enough and thought it only looped when the lines were _not_ commented.

Running with `-ddump-tc-traces` confirms my suspicions though, this appears to be a loop happening somewhere in the typechecker plugins. I'll mark this as a bug and hopefully someone can take a look a look at it after the holidays.

christiaanb, 2020-12-25:
Unless this is an example derived from a larger example, I'm inclined to label it as a "wontfix", given that the equality `CLog 2 d ~ (CLog 2 (2 ^ (CLog 2 d)))` results in the infinite rewrite `d => 2 ^ (CLog 2 d)`, which unsurprisingly throws the plugins in a loop. That's why GHC has the constraint solver limit.

paddytheplaster, 2020-12-25:
> Unless this is an example derived from a larger example, I'm inclined to label it as a "wontfix", given that the equality `CLog 2 d ~ (CLog 2 (2 ^ (CLog 2 d)))` results in the infinite rewrite `d => [...]

It's part of a much larger example. Originally I only had `l ~ (CLog 2 d)` and used `l` for all `CLog 2 d` expressions. Then I tried to eliminate the `l` because it really only depended on `d`.