clash-lang/clash-compiler #1621: Type Inference Problem with `2^(CLog 2 d)` Constraint

paddytheplaster, 2020-12-25:
The code below forces the type checker to (wrongly) infer that `2 ^ CLog 2 d` must be equal to `d`: I get:

     * Could not deduce: (2 ^ CLog 2 d) ~ d
          from the context: (KnownNat d, 1 <= d)
            bound by the type signature for:
                       bug :: forall (d :: Nat). (KnownNat d, 1 <= d) => Index d -> ()
            at Example.lhs:7:3-67
          `d' is a rigid type variable bound by
            the type signature for:
              bug :: forall (d :: Nat). (KnownNat d, 1 <= d) => Index d -> ()
            at Example.lhs:7:3-67
          Expected type: Vec (2 ^ CLog 2 d) ()
            Actual type: Vec d ()

Interestingly enough, it compiles fine if you un-comment the commented line and comment out the last three lines.

    > module Example where
    > import Clash.Prelude

    > bug :: forall (d :: Nat) . ( KnownNat d, 1 <= d) => Index d -> ()
    > bug _ = ()
    >     where b :: Vec (2^(CLog 2 d)) ()
    >           -- b = map (const ()) indicesI
    >           b = map f indicesI
    >             where f :: Index d -> ()
    >                   f _ = ()

christiaanb, 2020-12-25:
The type of `indicesI` is:
```haskell
indicesI :: KnownNat n => Vec n (Index n)
```
since you're using scoped type variables it the type checker will try to unify `Index n` with `Index d` given that:
```haskell
f :: Index d -> ()
map :: (a -> b) -> Vec n a -> Vec n b
```
That means it will try to unify `n` with `d`. But then the type signature for `b` also says that `n ~ 2 ^ (CLog 2 d)` must hold.
So now we have two unifications:
```
n ~ d
n ~ 2 ^ (CLog 2 d)
```
But that only holds when:
```
d ~ 2 ^ (CLog 2 d)
```
which I can easily falsify by picking any non-power of 2 for `d`, e.g. `3`.

paddytheplaster, 2020-12-25:
> The type of `indicesI` is:
> 
> ```haskell
> indicesI :: KnownNat n => Vec n (Index n)
> ```
> 
> since you're using scoped type variables the type checker will try to unify `Index n` with `Index d` given that:
> 
> ```haskell
> f :: Index d -> ()
> map :: (a -> b) -> Vec n a -> Vec n b
> ```
> 
> That means it will try to unify `n` with `d`. But then the type signature for `b` will aslo result in a unification such that `n ~ 2 ^ (CLog 2 d)` must hold.
> So now we have two unifications:
> 
> ```
> n ~ d
> n ~ 2 ^ (CLog 2 d)
> ```
> 
> But that only holds when:
> 
> ```
> d ~ 2 ^ (CLog 2 d)
> ```
> 
> which I can easily falsify by picking any non-power of 2 for `d`, e.g. `3`.

Thanks. I overlooked that. That's very useful.

christiaanb, 2020-12-25:
It compiles if you use the exact integer logarithm (`Log`) for the type signature of `b`:
```haskell
b :: Vec (2^(Log 2 d)) ()
```

paddytheplaster, 2020-12-25:
I should have mentioned in my previous reply. It's not immediately clear why the `map (const ())` _does_ work.