clash-lang/clash-compiler #1622: doctests:  executable not found

juhp, 2021-01-01:
For the Stackage Nightly build:

```
Test suite failure for package clash-prelude-1.2.5
    doctests:  executable not found
```

The rest of the tests all seem to pass.

ezzieyguywuf, 2021-01-01:
I've run into this issue before with other packages - see [this issue](https://github.com/haskell/cabal/issues/7213) with the cabal project for more info.

christiaanb, 2021-01-01:
So it seems our options are:
1. Stop using `cabal-doctest` and a custom `Setup.hs`, which is rather annoying, since it actually fixed some issues we had with our doctests
2. Wait for the upstream `Cabal`-the-library fix

Both are quite unsatisfactory... we'll have to think about how to solve this.

martijnbastiaan, 2021-01-01:
As a workaround, perhaps we could patch `Setup.hs` to always run `build` after `haddock`?

ezzieyguywuf, 2021-01-01:
I can confirm that running build after haddock works as a temporary
workaround, it's what we're doing in gentoo

martijnbastiaan, 2021-01-01:
My suggestion doesn't really seem feasible: the various commands passed to `Setup.hs` might have wildly different arguments, hence making it brittle at best. Perhaps Stackage could work around this by running tests first and only then run Haddock?

christiaanb, 2021-01-01:
@martijnbastiaan I think your suggestion isn’t that fragile: just intercept all the arguments and pass them along unchanged to `defaultMainWithDoctests` (or `defaultMain`), but when the arguments contain the string “haddock” we simply call `defaultMainWithDoctests` again, only this time with a single “build” argument.

martijnbastiaan, 2021-01-02:
Yeah, the reason I'm saying it might be brittle is that I ran it and saw (something like):

```
command --builddir dist_newstyle/foo --another-flag
```

If we don't pass `--builddir` we might regenerate files in the wrong place. If we do pass the flags we risk passing haddock only flags to `build`. But we might as well try :)

commit 3ebc0999 (2021-11-28): Use `doctest-parallel` for Prelude's doctests

From the project's README, this has a couple of advantages:

 * It runs tests in parallel
 * It runs tests against compiled code, instead of reinterpreting
   your whole project
 * It isolates examples in modules, ensuring your tests don't
   accidentally rely on each other
 * It parses cabal files to discover modules, no need for custom setup
   anymore!

The project lists two disadvantges, of which only one is relevant for
clash-compiler:

 * Use of conditionals in a cabal file as well as CPP flags will be
   ignored

This patch speeds up doctes [... cut]
commit 0a356de2 (2021-11-28): Use `doctest-parallel` for Prelude's doctests

From the project's README, this has a couple of advantages:

 * It runs tests in parallel
 * It runs tests against compiled code, instead of reinterpreting
   your whole project
 * It isolates examples in modules, ensuring your tests don't
   accidentally rely on each other
 * It parses cabal files to discover modules, no need for custom setup
   anymore!

The project lists two disadvantges, of which only one is relevant for
clash-compiler:

 * Use of conditionals in a cabal file as well as CPP flags will be
   ignored

This patch speeds up doctes [... cut]
commit c9e655a3 (2021-11-28): Use `doctest-parallel` for Prelude's doctests

From the project's README, this has a couple of advantages:

 * It runs tests in parallel
 * It runs tests against compiled code, instead of reinterpreting
   your whole project
 * It isolates examples in modules, ensuring your tests don't
   accidentally rely on each other
 * It parses cabal files to discover modules, no need for custom setup
   anymore!

The project lists two disadvantges, of which only one is relevant for
clash-compiler:

 * Use of conditionals in a cabal file as well as CPP flags will be
   ignored

This patch speeds up doctes [... cut]
commit 2655fe7a (2021-11-28): Use `doctest-parallel` for Prelude's doctests

From the project's README, this has a couple of advantages:

 * It runs tests in parallel
 * It runs tests against compiled code, instead of reinterpreting
   your whole project
 * It isolates examples in modules, ensuring your tests don't
   accidentally rely on each other
 * It parses cabal files to discover modules, no need for custom setup
   anymore!

The project lists two disadvantges, of which only one is relevant for
clash-compiler:

 * Use of conditionals in a cabal file as well as CPP flags will be
   ignored

This patch speeds up doctes [... cut]
closed 2021-12-13 by martijnbastiaan