clash-lang/clash-compiler #1632: AutoReg can't be generated with constant Nat in type

alex-mckenna, 2021-01-18:
Happens on `master` but not on `1.2.5` from hackage. Can be shown trying to build [christiaanb/contranomy](https://github.com/christiaanb/contranomy).

```
src/Contranomy/Instruction.hs:1:1: error:    Exception when trying to run compile-time code:
      Error while deriveAutoReg: don't know how to handle: GHC.TypeNats.KnownNat (4)CallStack (from HasCallStack):
  error, called at src/Clash/Class/AutoReg/Internal.hs:386:14 in clash-prelude-1.3.0-inplace:Clash.Class.AutoReg.Internal    Code: deriveAutoReg ''MCause
  |1 | {-# LANGUAGE GeneralizedNewtypeDeriving #-}
  | ^cabal: Failed to build contranomy-0.1 (which is required by exe:clash from
contranomy-0.1).
```

alex-mckenna, 2021-01-18:
The failing declaration:

```haskell
data MCause
  = MCause (BitVector 4)
  deriving (Generic, NFDataX)
deriveAutoReg ''MCause
```

From @leonschoorl:

> The 1.2 branch doesn't have this issue.
Neither do Signed and Unsigned.

leonschoorl, 2021-01-18:
It's "caused" by this commit: https://github.com/clash-lang/clash-compiler/commit/0c78f7d3294f0511177ce1e418547f05005686aa

leonschoorl, 2021-01-18:
But the real cause is not handling LitT here: https://github.com/clash-lang/clash-compiler/blob/master/clash-prelude/src/Clash/Class/AutoReg/Internal.hs#L380-L388

martijnbastiaan, 2021-01-18:
> But the real cause is not handling LitT here: https://github.com/clash-lang/clash-compiler/blob/master/clash-prelude/src/Clash/Class/AutoReg/Internal.hs#L380-L388

Just slap a `False` in there I suppose?

leonschoorl, 2021-01-18:
Yes
I'm just adding a test now

PR #1635 (2021-01-18): Teach deriveAutoReg to handle LitT's in constraints
This allows it to drop constraints like `KnownNat 4`.

Fixes #1632
commit 399bf5dc (2021-01-18): Teach deriveAutoReg to handle LitT's in constraints

This allows it to drop constraints like `KnownNat 4`.

Fixes #1632
commit 16d9baef (2021-01-18): Teach deriveAutoReg to handle LitT's in constraints

This allows it to drop constraints like `KnownNat 4`.

Fixes #1632

(cherry picked from commit 399bf5dc660f9af497672336317a66442b913b5a)
closed 2021-01-18 by leonschoorl