clash-lang/clash-compiler #1633: How to handle multiple `topEntity` definitions in multiple modules?

gergoerdi, 2021-01-18:
Let's say I have two modules, let's call them `A` and `B`, where both define a `topEntity` via `makeTopEntity`, and `B` imports `A` hiding `topEntity`. Normally, if this was `main` instead of `topEntity`, there would be no issue: compiling `A` would make `A.main` into "the" main function, and compiling `B` would make `B.main` (and of course `B.main` is free to use other exported definitions from `A`). But with `topEntity`, this doesn't seem to work.

First try: `clash --verilog B.hs`. This produces weird output showing that first, `B` is compiled into Verilog, then `A` is, overwriting the first compilation's result:

```
$ stack exec -- clash -outputdir _build/tmp --verilog -isrc src/Hardware/SpaceInvaders/SimBoard.hs 
GHC: Parsing and optimising modules took: 9.273s
GHC: Loading external modules from interface files took: 0.020s
GHC: Parsing annotations took: 0.027s
Clash: Parsing and compiling primitives took 0.152s
GHC+Clash: Loading modules cumulatively took 11.526s
Clash: Compiling Hardware.SpaceInvaders.SimBoard.topEntity
Clash: Normalisation took 18.951s
Clash: Netlist generation took 0.449s
Clash: Compiling Hardware.SpaceInvaders.topEntity
Clash: Normalisation took 20.598s
Clash: Netlist generation took 0.657s
Clash: Total compilation took 52.571s
```

(here, `Hardware.SpaceInvaders` stands for `A` and `Hardware.SpaceInvaders.SimBoard` stands for `B`; complete code is at https://github.com/gergoerdi/clash-spaceinvaders/tree/verilator-board but I don't think the details matter)

And if I look at the `topEntity.v` file generated, indeed it contains Verilog generated from `A.topEntity`!

Second try: using `-main-is`. Figuring out the details of how this is supposed to work is tricky.

```
$ stack exec -- clash -outputdir _build/tmp --verilog -isrc src/Hardware/SpaceInvaders/SimBoard.hs -main-is Hardware.SpaceInvaders.SimBoard.topEntity

src/Hardware/SpaceInvaders/SimBoard.hs:1:1: error:
    The IO action ‘main’ is not defined in module ‘Hardware.SpaceInvaders.SimBoard’
  |
1 | module Hardware.SpaceInvaders.SimBoard where
  | ^

```

OK let's try just `topEntity`. This results in an eerie error message:

```$ stack exec -- clash -outputdir _build/tmp --verilog -isrc src/Hardware/SpaceInvaders/SimBoard.hs -main-is topEntity

GHC: Parsing and optimising modules took: 9.204s
GHC: Loading external modules from interface files took: 0.020s
GHC: Parsing annotations took: 0.026s
Clash: Parsing and compiling primitives took 0.148s

<no location info>: error:
    Clash error call:
    Internal error: multiple top entities called topEntity ([Id {varName = Name {nameSort = User, nameOcc = "Hardware.SpaceInvaders.topEntity", nameUniq = 8214565720324087478, nameLoc = RealSrcSpan SrcSpanOneLine "src/Hardware/SpaceInvaders.hs" 36 1 10}, varUniq = 8214565720324087478, varType = AppTy (AppTy (ConstTy Arrow) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Clock", nameUniq = 8214565720323789631, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "Dom25")))) (AppTy (AppTy (ConstTy Arrow) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Reset", nameUniq = 8214565720323789655, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "Dom25")))) (AppTy (AppTy (ConstTy Arrow) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Signal", nameUniq = 8214565720323789676, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "Dom25"))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Sized.Internal.BitVector.BitVector", nameUniq = 8214565720323789708, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (NumTy 8))))) (AppTy (AppTy (ConstTy Arrow) (AppTy (AppTy (AppTy (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Tuple.(,,,,)", nameUniq = 3746994889972252682, nameLoc = UnhelpfulSpan "<no location info>"}))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Signal", nameUniq = 8214565720323789676, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "Dom25"))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.Utils.Active", nameUniq = 8214565720323789837, nameLoc = UnhelpfulSpan "<no location info>"}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.Utils.High", nameUniq = 8214565720323789842, nameLoc = UnhelpfulSpan "<no location info>"})))))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Signal", nameUniq = 8214565720323789676, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "Dom25"))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.Utils.Active", nameUniq = 8214565720323789837, nameLoc = UnhelpfulSpan "<no location info>"}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.Utils.High", nameUniq = 8214565720323789842, nameLoc = UnhelpfulSpan "<no location info>"})))))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Signal", nameUniq = 8214565720323789676, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "Dom25"))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.Utils.Active", nameUniq = 8214565720323789837, nameLoc = UnhelpfulSpan "<no location info>"}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.Utils.High", nameUniq = 8214565720323789842, nameLoc = UnhelpfulSpan "<no location info>"})))))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Signal", nameUniq = 8214565720323789676, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "Dom25"))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.Utils.Active", nameUniq = 8214565720323789837, nameLoc = UnhelpfulSpan "<no location info>"}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.Utils.High", nameUniq = 8214565720323789842, nameLoc = UnhelpfulSpan "<no location info>"})))))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Signal", nameUniq = 8214565720323789676, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "Dom25"))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.Utils.Active", nameUniq = 8214565720323789837, nameLoc = UnhelpfulSpan "<no location info>"}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.Utils.High", nameUniq = 8214565720323789842, nameLoc = UnhelpfulSpan "<no location info>"}))))))) (AppTy (AppTy (ConstTy Arrow) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.PS2.PS2", nameUniq = 8214565720324080545, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "Dom25")))) (AppTy (AppTy (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "RetroClash.VGA.VGAOut", nameUniq = 8214565720324058041, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "Dom25"))) (LitTy (NumTy 8))) (LitTy (NumTy 8))) (LitTy (NumTy 8))))))), idScope = GlobalId},Id {varName = Name {nameSort = User, nameOcc = "Hardware.SpaceInvaders.SimBoard.topEntity", nameUniq = 8214565720324103336, nameLoc = RealSrcSpan SrcSpanOneLine "src/Hardware/SpaceInvaders/SimBoard.hs" 24 1 10}, varUniq = 8214565720324103336, varType = AppTy (AppTy (ConstTy Arrow) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Clock", nameUniq = 8214565720323789631, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "System")))) (AppTy (AppTy (ConstTy Arrow) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Reset", nameUniq = 8214565720323789655, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "System")))) (AppTy (AppTy (ConstTy Arrow) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Signal", nameUniq = 8214565720323789676, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "System"))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Sized.Internal.BitVector.BitVector", nameUniq = 8214565720323789708, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (NumTy 8))))) (AppTy (AppTy (ConstTy Arrow) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Signal", nameUniq = 8214565720323789676, nameLoc = UnhelpfulSpan "<no location info>"}))) (LitTy (SymTy "System"))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Types.Bool", name [... cut ...]
GHC+Clash: Loading modules cumulatively took 11.126s
Clash: Compiling Hardware.SpaceInvaders.SimBoard.topEntity
Clash: Normalisation took 18.746s
Clash: Netlist generation took 0.483s
Clash: Total compilation took 30.582s
```

Looks promising! However, if I look at the resulting `topEntity.v` file, it is not picking up the signal names from the type annotations.

I am out of ideas. Is there any way to make this work, short of refactoring `A` into some `A1` that contains everything but `topEntity`, and importing `A1` into both `A` and `B`?

christiaanb, 2021-01-18:
Right... this is broken, multiple top entities (which should actually be called "synthesis entry points") where:
1. The fully qualified names share the same root, e.g. `Hardware.SpaceInvaders` and `Hardware.SpaceInvaders.SimBoard` both share the root `Hardware`
2. The fully qualified names share the same final suffix, e.g  `Hardware.SpaceInvaders.topEntity` and `Hardware.SpaceInvaders.SimBoard.topEntity` both share the final suffix `topEntity`

As a work-around, I just wouldn't call them both `topEntity`, but e.g. call them `Hardware.SpaceInvaders.game` and `Hardware.SpaceInvaders.board`. The `makeTopEntity` will ensure that Clash knows they are synthesis entry points, and will start to synthesize them.

gergoerdi, 2021-01-19:
> 1. The fully qualified names share the same root, e.g. `Hardware.SpaceInvaders` and `Hardware.SpaceInvaders.SimBoard` both share the root `Hardware`

So is there some compatibility reason that [this isn't fixed](https://github.com/clash-lang/clash-compiler/issues/1314), or is this just a question of priorities?

> As a work-around, I just wouldn't call them both `topEntity`, but e.g. call them `Hardware.SpaceInvaders.game` and `Hardware.SpaceInvaders.SimBoard.board`. The `makeTopEntity` will ensure that Clash knows they are synthesis entry points, and will start to synthesize them.

Thanks, in fact this even allowed me to collapse everything into just one module, using `Hardware.Spaceinvaders.topEntity` and `Hardware.SpaceInvaders.boardEntity` as the two entity names. I also took the opportunity to teach [Clashilator](https://github.com/gergoerdi/clashilator) to pick up the entity name from a new Cabal field `x-clashilator-entity`.

closed 2021-03-09 by martijnbastiaan