clash-lang/clash-compiler #1634: deriveAutoReg fails for data types containing BitVector

leonschoorl, 2021-01-18:
```haskell
data MCause
  = MCause (BitVector 4)
  deriving (Generic, NFDataX)
deriveAutoReg ''MCause
```
fails on current master with 
```
    Exception when trying to run compile-time code:
      Error while deriveAutoReg: don't know how to handle: GHC.TypeNats.KnownNat (4)
CallStack (from HasCallStack):
  error, called at src/Clash/Class/AutoReg/Internal.hs:386:14 in clash-prelude-1.3.0-inplace:Clash.Class.AutoReg.Internal
    Code: deriveAutoReg ''MCause
```

The 1.2 branch doesn't have this issue.
Neither do `Signed` and `Unsigned`.

alex-mckenna, 2021-01-18:
_cough_ [hi there](https://github.com/clash-lang/clash-compiler/issues/1632) _cough_

leonschoorl, 2021-01-18:
my bad, duplicate of #1632

closed 2021-01-18 by leonschoorl