clash-lang/clash-compiler #1637: Option to ignore json files in current directory

acairncross, 2021-01-20:
Is there a way to have Clash not look at json files in the current directory? Right now, if I have a json file in the current directory that's not a valid primitive file Clash returns an error. It would be convenient if this could be opted out of, or maybe even opted in to.

christiaanb, 2021-01-20:
We should probably stop using/looking for the `.json` file extension for the primitive definition file(s) and use something like `.clashprimitive` instead; of course the format will remain json.

So to give a direct answer: there’s currently no opt-out.

martijnbastiaan, 2021-03-18:
This has been more or less solved by Clash 1.4: primitive definitions are now stored n `.primitives` files (as the format isn't really JSON). I hope this solves your direct issue. Please re-open if it doesn't.

closed 2021-03-18 by martijnbastiaan