clash-lang/clash-compiler #1644: Space leak when simulating ClearOnReset RAM

gergoerdi, 2021-02-03:
In 1993, a gaming magazine reported on [Microsoft's upcoming astronautical game](https://en.wikipedia.org/wiki/Microsoft_Space_Simulator); that was a **Space Simulator leak**. This ticket, instead, is about a **simulator space leak**. 

In a situation eerily reminiscent of #1256, I have discovered that the following program keeps using more and more memory:

```
import Clash.Prelude
import Control.Arrow.Transformer.Automaton

main :: IO ()
main = do
    let go (Automaton step) = do
            let (out, next) = step ()
            printX out
            go next
    go $ signalAutomaton @System board1

board1 :: (HiddenClockResetEnable dom) => Signal dom () -> Signal dom (Unsigned 1)
board1 inp = blockRam1 @_ @_ @(Unsigned 1) ClearOnReset (SNat @1) 0 (pure 0) (pure Nothing)
```

while changing the `blockRam1`'s `ResetStrategy` to `NoClearOnReset` results in a constant memory usage. This can easily be tested by running with the flag `+RTS -M2M`, since 2M of heap is enough for the non-leaking one, while the leaking one quickly runs out of it after ~5000 simulated periods.

martijnbastiaan, 2021-02-03:
My guess is that a strict mux should be used here:

https://github.com/clash-lang/clash-compiler/blob/e6ee4fb929af4fbe2b8a0695b1f33aa07bf91ef2/clash-prelude/src/Clash/Explicit/BlockRam.hs#L931-L934

I'll try it out.

martijnbastiaan, 2021-02-03:
Self-contained reproducer:

```haskell
import Clash.Prelude
import Control.Exception
import Control.Monad

main :: IO ()
main = do
  let samples = sampleN @System 10000 (board1 (pure ()))
  forM_ samples (evaluate . rnfX)

board1 :: (HiddenClockResetEnable dom) => Signal dom () -> Signal dom (Unsigned 1)
board1 inp = blockRam1 @_ @_ @(Unsigned 1) ClearOnReset (SNat @1) 0 (pure 0) (pure Nothing)
```

martijnbastiaan, 2021-02-03:
Frustratingly, I haven't been able to get any closer to a solution.

* Cabal throws linker errors when I enable profiling
* Stack works, but the binary it produces then crashes with `getTopHandlerThread: neither a WEAK nor a DEAD_WEAK` when I try to profile the memory
* Using a strict `mux` doesn't plug the space leak
* Using `-fstrict-mapSignal` doesn't plug the space leak
* Inside `blockRam1#`: Using `seqX` on the signal elements instead of signal doesn't plug the space leak

martijnbastiaan, 2021-02-03:
If I replace `waCounter` inside `blockRam1` with:

```diff
-  waCounter = register clk rstInv en 0 (satSucc SatBound <$> waCounter)
+  waCounter = pure 0 -- register clk rstInv en 0 (satSucc SatBound <$> waCounter)
```

Bizarrely, when I revert that patch and instead force it in `blockRam#`:

```diff
-    in  o `seqX` o :- (ret `seq` rt `seq` et `seq` wt `seq` dt `seq` go ram' o' res rs en wr din)
+    in  o `seqX` o :- (deepseqX w (ret `seq` rt `seq` et `seq` wt `seq` dt `seq` go ram' o' res rs en wr din))
```

The leak is still there. We don't even need the feedback behavior to trigger the space leak. This;

```
-  waCounter = register clk rstInv en 0 (satSucc SatBound <$> waCounter)
+  waCounter = register clk rstInv en 0 (pure 1)
```

triggers it too!

martijnbastiaan, 2021-02-03:
@gergoerdi https://github.com/clash-lang/clash-compiler/pull/1645 :)

commit 60f11e1f (2021-02-03): Force accumulating argument 'o' to WHNF in 'asyncRegister#'

Fixes memory leak triggered on asserted resets. This went under the
radar for a long time, as reset periods are typically very short.

Fixes #1644
commit 4c8960fa (2021-02-03): Force accumulating argument 'o' to WHNF in 'asyncRegister#'

Fixes memory leak triggered on asserted resets. This went under the
radar for a long time, as reset periods are typically very short.

Fixes #1644
commit 380c06ab (2021-02-03): Force accumulating argument 'o' to WHNF in 'asyncRegister#'

Fixes memory leak triggered on asserted resets. This went under the
radar for a long time, as reset periods are typically very short.

Fixes #1644
commit 1d1d5e6d (2021-02-03): Force accumulating argument 'o' to WHNF in 'asyncRegister#'

Fixes memory leak triggered on asserted resets. This went under the
radar for a long time, as reset periods are typically very short.

Fixes #1644
closed 2021-02-04 by martijnbastiaan