clash-lang/clash-compiler #1667: specializing a function on a non work-free cast

DigitalBrains1, 2021-02-14:
The following code:
```haskell
module MultiBlinker where

import Clash.Prelude

import qualified Clash.Explicit.Prelude as CEP
import Clash.Intel.ClockGen

createDomain vSystem{vName="DomInput", vPeriod=hzToPeriod 50e6}
createDomain vSystem{vName="Dom15", vPeriod=hzToPeriod 15e6}
createDomain vSystem{vName="Dom10", vPeriod=hzToPeriod 10e6}

topEntity
  :: Clock DomInput
  -> Signal DomInput Bool
  -> Signal Dom15 Bit
  -> ( Signal Dom15 (BitVector 4)
     , Signal Dom10 (BitVector 4)
     )

topEntity clk rst key15 = (leds15, leds10)
 where
  leds15 = exposeClockResetEnable blinker pll15Out rst15Sync enableGen key15
  leds10 = exposeClockResetEnable blinker pll10Out rst10Sync enableGen key10
  key10 = CEP.dualFlipFlopSynchronizer pll15Out pll10Out rst10Sync enableGen 1
            key15
  (pll15Out,pll15Stable) =
    altpll @Dom15 (SSymbol @"altpll15") clk (unsafeFromLowPolarity rst)
  rst15Sync =
    resetSynchronizer pll15Out (unsafeFromLowPolarity pll15Stable) enableGen
  (pll10Out,pll10Stable) =
    altpll @Dom10 (SSymbol @"altpll10") clk (unsafeFromLowPolarity rst)
  rst10Sync =
    resetSynchronizer pll10Out (unsafeFromLowPolarity pll10Stable) enableGen
{-# ANN topEntity
  (Synthesize
    { t_name   = "blinker"
    , t_inputs = [PortName "CLOCK_50", PortName "KEY0", PortName "KEY1"]
    , t_output = PortProduct ""
                   [ PortName "LEDH"
                   , PortName "LEDL"
                   ]
    }) #-}

blinker
  :: ( HiddenClockResetEnable dom
     , KnownNat n)
  => Signal dom Bit
  -> Signal dom (BitVector n)
blinker =
    mealy
      -- 15 MHz: update period 333 ms
      -- 10 MHz: update period 500 ms
      (blinkerT @5000000)
      (1, False, 0)
  . isRising 1

blinkerT
  :: forall d n
   . ( KnownNat d
     , KnownNat n
     , 1 <= d)
  => (BitVector n, Bool, Index d)
  -> Bool
  -> ((BitVector n, Bool, Index d), BitVector n)
blinkerT (leds,mode,cntr) key1R = ((leds',mode',cntr'),leds)
  where
    cntr' = satSucc SatWrap cntr

    mode' | key1R     = not mode
          | otherwise = mode

    leds' | cntr == 0 = if mode then complement leds
                                else rotateL leds 1
          | otherwise = leds
```

gives warnings:
```
WARNING: Clash.Normalize.Transformations(1019):  specializing a function on a non work-free cast. Generated HDL implementation might contain duplicate work. Please report this as a bug.

Expression where this occured:

Clash.Sized.Internal.Index.$fSaturatingNumIndex_$csatAdd[8214565720323859044][GlobalId]
  @5000000
  5000000
  (GHC.Types.Eq#[3891110078048108574]
     @GHC.Types.Bool[3674937295934324744]
     @GHC.Types.True[3891110078048108589]
     @GHC.Types.True[3891110078048108589]
     (_CO_
        @(GHC.Prim.~#[3674937295934324842]
            GHC.Types.Bool[3674937295934324744]
            GHC.Types.Bool[3674937295934324744]
            GHC.Types.True[3891110078048108589]
            GHC.Types.True[3891110078048108589])))
  Clash.Class.Num.SatWrap[8214565720323789578]
  ((cntr[6989586621679221188][LocalId] ▷
    Clash.Sized.Internal.Index.Index[8214565720323789710]
      5000000
    ~
    GHC.Integer.Type.Integer[3674937295934324784]) ▷
    GHC.Integer.Type.Integer[3674937295934324784]
    ~
    Clash.Sized.Internal.Index.Index[8214565720323789710]
      5000000)
WARNING: Clash.Normalize.Transformations(1019):  specializing a function on a non work-free cast. Generated HDL implementation might contain duplicate work. Please report this as a bug.

Expression where this occured:

Clash.Sized.Internal.Index.plus# @5000000
  @5000000
  ((eta1[8214565720324009380][LocalId] ▷
    Clash.Sized.Internal.Index.Index[8214565720323789710]
      5000000
    ~
    GHC.Integer.Type.Integer[3674937295934324784]) ▷
    GHC.Integer.Type.Integer[3674937295934324784]
    ~
    Clash.Sized.Internal.Index.Index[8214565720323789710]
      5000000)
  (((Clash.Sized.Internal.Index.fromInteger#
       @5000000
       5000000
       1) ▷
    Clash.Sized.Internal.Index.Index[8214565720323789710]
      5000000
    ~
    GHC.Integer.Type.Integer[3674937295934324784]) ▷
    GHC.Integer.Type.Integer[3674937295934324784]
    ~
    Clash.Sized.Internal.Index.Index[8214565720323789710]
      5000000)
WARNING: Clash.Normalize.Transformations(1019):  specializing a function on a non work-free cast. Generated HDL implementation might contain duplicate work. Please report this as a bug.

Expression where this occured:

Clash.Sized.Internal.Index.plus# @5000000
  @5000000
  ((eta1[8214565720324009380][LocalId] ▷
    Clash.Sized.Internal.Index.Index[8214565720323789710]
      5000000
    ~
    GHC.Integer.Type.Integer[3674937295934324784]) ▷
    GHC.Integer.Type.Integer[3674937295934324784]
    ~
    Clash.Sized.Internal.Index.Index[8214565720323789710]
      5000000)
```

I didn't try reducing it to a MWE yet.

DigitalBrains1, 2021-02-14:
```
$ clash --numeric-version
8.8.4
```

DigitalBrains1, 2021-02-15:
A smaller working example:
```
module I1667 where

import Clash.Explicit.Prelude

topEntity
  :: Clock System
  -> Reset System
  -> Enable System
  -> Signal System (BitVector 4)
topEntity clk rst en =
  let (s,o) = unbundle $ blinker @50000 <$> register clk rst en 0 s
  in o

blinker
  :: forall d n
   . ( KnownNat d
     , KnownNat n
     , 1 <= d)
  => Index d
  -> (Index d, BitVector n)
blinker s = let s' = satSucc SatWrap s
            in (s', resize $ pack s')
```

gives:
```
Clash: Compiling I1667.topEntity
WARNING: Clash.Normalize.Transformations(1019):  specializing a function on a non work-free cast. Generated HDL implementation might contain duplicate work. Please report this as a bug.

Expression where this occured:

Clash.Sized.Internal.Index.plus# @50000 @50000
  ((eta1[3013][LocalId] ▷
    Clash.Sized.Internal.Index.Index[8214565720323871107]
      50000
    ~
    GHC.Integer.Type.Integer[3674937295934324784]) ▷
    GHC.Integer.Type.Integer[3674937295934324784]
    ~
    Clash.Sized.Internal.Index.Index[8214565720323871107]
      50000)
  (((Clash.Sized.Internal.Index.fromInteger#
       @50000
       50000
       1) ▷
    Clash.Sized.Internal.Index.Index[8214565720323871107]
      50000
    ~
    GHC.Integer.Type.Integer[3674937295934324784]) ▷
    GHC.Integer.Type.Integer[3674937295934324784]
    ~
    Clash.Sized.Internal.Index.Index[8214565720323871107]
      50000)
WARNING: Clash.Normalize.Transformations(1019):  specializing a function on a non work-free cast. Generated HDL implementation might contain duplicate work. Please report this as a bug.

Expression where this occured:

Clash.Sized.Internal.Index.plus# @50000 @50000
  ((eta1[3013][LocalId] ▷
    Clash.Sized.Internal.Index.Index[8214565720323871107]
      50000
    ~
    GHC.Integer.Type.Integer[3674937295934324784]) ▷
    GHC.Integer.Type.Integer[3674937295934324784]
    ~
    Clash.Sized.Internal.Index.Index[8214565720323871107]
      50000)
```

leonschoorl, 2021-02-15:
Can be further simplified to:
```haskell
topEntity
  :: Index 5
  -> Index 5
topEntity = foo

foo
  :: ( KnownNat d
     , 1 <= d)
  => Index d
  -> Index d
foo s = satSucc SatWrap s
```

DigitalBrains1, 2021-02-15:
That's interesting, a minor variation on that did not trigger for me.

DigitalBrains1, 2021-02-15:
This one does not trigger warnings. Only difference is point free style in `foo`.
```haskell
topEntity
  :: Index 5
  -> Index 5
topEntity = foo

foo
  :: ( KnownNat d
     , 1 <= d)
  => Index d
  -> Index d
foo = satSucc SatWrap
```

alex-mckenna, 2021-02-15:
> Only difference is point free style in foo.

Perhaps something to do with eta-expansion then?

leonschoorl, 2021-02-18:
The warning was triggered by the recent [changes to `isWorkFree`](https://github.com/clash-lang/clash-compiler/pull/1655).
But clash should never have been specializing prims like `Index.plus#`, that only creates more work for itself.

commit 2402bf06 (2021-02-22): argCastSpec: Don't specialise prims on casted arguments

We specialise functions on casted arguments to push the casts down into
the body of the function.
But prims don't have bodies to push the casts into.

This also fixes #1667
commit 5a6b5602 (2021-02-22): argCastSpec: Don't specialise prims on casted arguments

We specialise functions on casted arguments to push the casts down into 
the body of the function.
But prims don't have bodies to push the casts into.

This also fixes #1667
commit 6a591882 (2021-02-22): isWorkFree: Look through Cast and Tick

This fixes the spurious warnings about non work-free casts seen in

Fixes #1667
PR #1672 (2021-02-22): argCastSpec: Don't specialise prims on casted arguments 
Also make `isWorkFree` look through Casts.

Both fix #1667
closed 2021-02-22 by martijnbastiaan