clash-lang/clash-compiler #1674: `createDomain` creates definitions without documentation

martijnbastiaan, 2021-02-22:
This is an issue when trying to achieve 100% documentation coverage. E.g.:

```
  Missing documentation for:
    Core (src/Config.hs:11)
    vCore (src/Config.hs:11)
```

alex-mckenna, 2021-02-25:
Not much we can do for this right? Haddock's _floating documentation_ doesn't let you attach docs to a specific entity I don't think, and we can't generate docs with TH currently. Unless you have some idea?

martijnbastiaan, 2021-02-25:
> and we can't generate docs with TH currently. 

Right, I forgot that. That's pretty annoying.

Haddock should probably get an option to locally disable specific warnings (and it should get a `-Werror` pff). Marking this `external bug` (external feature?) for now.

christiaanb, 2021-02-25:
GHC 9.2 will solve this, I think: https://gitlab.haskell.org/ghc/ghc/-/merge_requests/3330

alex-mckenna, 2021-02-25:
> external bug (external feature?)

You say potato, I say :potato:

commit 3aed58fd (2021-02-25): Create domain alias/helper when they don't exit

fixes #1674

This makes it possible to explicitly define the type alias (or the
helper). This can be useful when writing documentation for both.

For example:

```haskell
-- | MySystem is a custom domain ...
type MySystem = ("MySystem" :: Domain)

-- | vMySystem is is the domain configuration associated with `MySystem`
createDomain vSystem{vName="MySystem"}
```
PR #1680 (2021-02-25): Create domain alias/helper only when they don't exist
fixes #1674

This makes it possible to explicitly define the type alias (or the
helper). This can be useful when writing documentation for both.

For example:

```haskell
-- | MySystem is a custom domain ...
type MySystem = ("MySystem" :: Domain)

-- | vMySystem is the domain configuration associated with `MySystem`
createDomain vSystem{vName="MySystem"}
```
closed 2021-02-25 by martijnbastiaan