clash-lang/clash-compiler #1686: Modules loaded by Clash.Main.defaultMain somehow linger on?

gergoerdi, 2021-02-28:
This is a bit of a shot in the dark, because I haven't set up a way to test this (and it's not that simple), but I think I am seeing signs of bad interference between multiple calls to `Clash.Main.defaultMain`  in the same process.

I have a Cabal file with multiple executables using Clashilator, and when I try to build them, the first one goes through, but the second invocation of `Clash.Main.defaultMain` fails with a linker conflict, and the error message references a module that should have been loaded by the previous invocation only:

```
Clash.defaultMain --verilog -outputdir .stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-cpu-screen-verilator/_clashilator/clash-syn -main-is simEntity Hardware.Compucolor2 -isrc -isim -package-db /home/cactus/prog/clash/compucolor-book/.stack-work/install/x86_64-linux-tinfo6/d1338d6a42d0d7cf6bd4d96f732d015d963ef02ddb7a2dbfbfe1f741a2225c2a/8.10.3/pkgdb -package-db /home/cactus/sdk/stack/snapshots/x86_64-linux-tinfo6/d1338d6a42d0d7cf6bd4d96f732d015d963ef02ddb7a2dbfbfe1f741a2225c2a/8.10.3/pkgdb
GHC: Parsing and optimising modules took: 18.117s
GHC: Loading external modules from interface files took: 0.048s
GHC: Parsing annotations took: 0.059s
Clash: Parsing and compiling primitives took 0.149s
GHC+Clash: Loading modules cumulatively took 21.515s
Clash: Compiling Hardware.Compucolor2.simEntity
Clash: Normalisation took 26.144s
Clash: Netlist generation took 0.592s
Clash: Total compilation took 48.764s
make -f .stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-cpu-screen-verilator/_clashilator/verilator/Makefile
/usr/bin/pkg-config --cflags verilator
Clash.defaultMain --verilog -outputdir .stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-cpu-textscreen-verilator/_clashilator/clash-syn -main-is simEntity Hardware.Compucolor2 -isrc -isim -package-db /home/cactus/prog/clash/compucolor-book/.stack-work/install/x86_64-linux-tinfo6/d1338d6a42d0d7cf6bd4d96f732d015d963ef02ddb7a2dbfbfe1f741a2225c2a/8.10.3/pkgdb -package-db /home/cactus/sdk/stack/snapshots/x86_64-linux-tinfo6/d1338d6a42d0d7cf6bd4d96f732d015d963ef02ddb7a2dbfbfe1f741a2225c2a/8.10.3/pkgdb
GHC runtime linker: fatal error: I found a duplicate definition for symbol
   HardwareziCompucolor2ziVideoziPlot_plots2_closure
whilst processing object file
   /home/cactus/prog/clash/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-cpu-textscreen-verilator/_clashilator/clash-syn/Hardware/Compucolor2/Video/Plot.o
The symbol was previously defined in
   /home/cactus/prog/clash/compucolor-book/.stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/compucolor2-cpu-screen-verilator/_clashilator/clash-syn/Hardware/Compucolor2/Video/Plot.o
This could be caused by: 
   * Loading two different object files which export the same symbol
   * Specifying the same object file twice on the GHCi command line
   * An incorrect `package.conf' entry, causing some object to be
     loaded twice.       
```

The two calls to `defaultMain` are almost the same, except for using different directories.  

Is `defaultMain` changing some internal state (perhaps of the RTS itself?) that is causing this?

gergoerdi, 2021-02-28:
I should also add that this problem seems to have started when I moved from `lts-16.21` (GHC 8.8.4) with 5a5a121204579992d4c15934060ebc92c6f024b5 to `lts-17.2` (GHC 8.10.3) with fb903e0b227045e387d5df6ed83009948c403d33.

martijnbastiaan, 2021-02-28:
Perhaps we're not allowed to do stuff like this twice in one thread:

https://github.com/clash-lang/clash-compiler/blob/a397d9bb4a484bdc8822c2d4b797195c9ecd53e2/clash-ghc/src-bin-8.10/Clash/Main.hs#L119-L125

I'm fairly certain Clash itself doesn't keep use a hidden/IO state.

gergoerdi, 2021-02-28:
I'll try to allocate some time tomorrow into making a way of reproducing this problem.

gergoerdi, 2021-02-28:
Oooh that's funny, found this in the GHC 8.10 changelog:

> GHC’s runtime linker no longer uses global state. This allows programs that use the GHC API to safely use multiple GHC sessions in a single process, as long as there are no native dependencies that rely on global state.

gergoerdi, 2021-03-01:
I have put a standalone repro (no external dependencies) to https://github.com/gergoerdi/clash-issue-1686

Just clone and `stack build`, and you should see:

```
GHC runtime linker: fatal error: I found a duplicate definition for symbol
   Lib2_plots2_closure
whilst processing object file
   .stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/exe2/_clashilator/clash-syn/Lib2.o
The symbol was previously defined in
   .stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/exe1/_clashilator/clash-syn/Lib2.o
This could be caused by:
   * Loading two different object files which export the same symbol
   * Specifying the same object file twice on the GHCi command line
   * An incorrect `package.conf' entry, causing some object to be
     loaded twice.
```

gergoerdi, 2021-03-01:
I have added two Stack resolvers to the repo. The only difference is the Stack LTS version, and thus the GHC version, that is underneath Clash:

* `stack build --resolver=clash-ghc-8.8.yaml` works
* `stack build --resolver=clash-ghc-8.10.yaml` fails

christiaanb, 2021-03-05:
I tried to use both `forkIO` and `forkOS` around the calls to `clashMain` as a work-around, but I still got the error. I guess as a solution we need to unload all of the loaded symbols. Sadly, there's nothing like that in the Linker API https://gitlab.haskell.org/ghc/ghc/-/blob/master/compiler/GHC/Linker/Loader.hs#L11-31, all we have is https://gitlab.haskell.org/ghc/ghc/-/blob/master/compiler/GHC/Linker/Loader.hs#L1085-1108

gergoerdi, 2021-03-06:
I've asked around on the GHC mailing list, and got some [interesting](https://mail.haskell.org/pipermail/ghc-devs/2021-March/019588.html) [replies](https://mail.haskell.org/pipermail/ghc-devs/2021-March/019589.html). Unfortunately, when I tried to use GHC `0dc7985663efa1739aafb480759e2e2e7fca2a36^` to build Clash and then my repo, it turned into a shit-show of mismatched Stackage versions: there's 100+ packages that would all need a version that works both with each other and with whatever quirky state GHC's built-in libraries were in in that given commit...

christiaanb, 2021-03-06:
Ah, I misunderstood, http://hackage.haskell.org/package/ghc-8.10.2/docs/Linker.html#v:unload is what we want. The list of objects that you give as an argument are the things you want to *keep*, where we do not want to keep anything.

Perhaps calling that `unload` just before the return on this line https://github.com/clash-lang/clash-compiler/blob/a397d9bb4a484bdc8822c2d4b797195c9ecd53e2/clash-ghc/src-ghc/Clash/GHC/LoadModules.hs#L538 will do “the right thing”. It will however need the *updated* HscEnv, so not this one: https://github.com/clash-lang/clash-compiler/blob/a397d9bb4a484bdc8822c2d4b797195c9ecd53e2/clash-ghc/src-ghc/Clash/GHC/LoadModules.hs#L446

bgamari, 2021-03-07:
I don't know the details of what Clash expects but do note that `unloadObj` is [essentially a no-op](https://gitlab.haskell.org/ghc/ghc/-/blob/49fff41d0385ac1ad80a1a8b2ae33e765e89a088/compiler/ghci/Linker.hs#L1122) in the case that your GHC is dynamically-linked.  However, this has been the case for as long as I can recall (and is due to fundamental difficulties with unloading in the presence of dynamic linking).

martijnbastiaan, 2021-03-07:
@gergoerdi I haven't had the chance to look into this myself, but from a discussion with @christiaanb I think you probably wouldn't see the error if you structure your project as a [Clash Starter Project](https://github.com/clash-lang/clash-starters/tree/main/simple). Roughly what it does is:

- It defines a library
- It defines an executable calling Clash's `defaultMain`, depending on the library.

To compile a design you can call it using `cabal run clash -- FooLib -main-is top`. This makes Clash load definitions directly from interface files, instead of calling GHC's compilation pipeline.

christiaanb, 2021-03-07:
So the exception is thrown somewhere from inside http://hackage.haskell.org/package/ghc-8.10.2/docs/GHC.html#v:typecheckModule because I added several `liftIO (print XYZ)` statements, and the statement after:
https://github.com/clash-lang/clash-compiler/blob/7149a3fe93923c6138ec45f9d5420004004f3ae0/clash-ghc/src-ghc/Clash/GHC/LoadModules.hs#L341 isn't printed when the exception is thrown.

christiaanb, 2021-03-07:
Adding:
```haskell
    hscEnvX <- GHC.getSession
    liftIO (Linker.unload hscEnvX [])
```
just before: https://github.com/clash-lang/clash-compiler/blob/a397d9bb4a484bdc8822c2d4b797195c9ecd53e2/clash-ghc/src-ghc/Clash/GHC/LoadModules.hs#L538
doesn't seem to work, and actually makes it fail silently earlier, during the first call to Clash's `defaultMain`.

gergoerdi, 2021-03-07:
> * It defines a library
> * It defines an executable calling Clash's `defaultMain`, depending on the library.
>
> To compile a design you can call it using `cabal run clash -- FooLib -main-is top`. This [...]

I'm not convinced this will work for Clashilator's Cabal integration. There, I need to run `Clash.defaultMain` during project configuration, since the result of running Clash is needed to run Verilator, whose output is in turn needed to be able to compile the Haskell executable: https://gergo.erdi.hu/blog/2020-05-07-integrating_verilator_and_clash_via_cabal/

martijnbastiaan, 2021-03-09:
That makes sense. Perhaps https://cabal.readthedocs.io/en/3.4/cabal-package.html#sublibs would work. I don't believe Stack/Hpack supports it though..

christiaanb, 2021-03-09:
I'm like 99% sure that https://gitlab.haskell.org/ghc/ghc/-/merge_requests/388 introduced this bug, my reasoning is the following:

In 8.8 and earlier we had:
- RTS C-code contains the ground truth what is linked. The API it provides are set-membership, insert, lookup, and delete. Notably it does not allow you to get the set of linked objects.
- There is a globally shared MVar (using NOINLINE, sharedCaf, unsafePerformIO) to what is basically a log/view of the linked-objects state kept by the RTS C-code.

With MR388, in 8.10 and later we get:
- RTS C-code contains the ground truth what is linked. The API it provides are set-membership, insert, lookup, and delete. Notably it does not allow you to get the set of linked objects.
- A **new** MVar for every call to `runGhc` which is a log/view of the linked-object state kept by the RTS C-code. But that means these MVar get out-of-sync with the ground truth that is the RTS C-code! And since the RTS C-code does not expose an API to get the set of linked objects, there's no way to sync these MVars either!

I'm building a GHC 8.10.2 with MR388 reverted to confirm.

christiaanb, 2021-03-09:
Okay, this is on 8.10.2 with MR388 reverted:
```
clash-issue1686                  > build (exe)
GHC: Parsing and optimising modules took: 1.966s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.001s
Clash: Parsing and compiling primitives took 0.115s
GHC+Clash: Loading modules cumulatively took 2.567s
Clash: Compiling Board.simEntity 
Clash: Normalisation took 0.000s 
Clash: Netlist generation took 0.000s
Clash: Total compilation took 2.575s
GHC: Parsing and optimising modules took: 0.981s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.000s
Clash: Parsing and compiling primitives took 0.113s
GHC+Clash: Loading modules cumulatively took 1.363s
Clash: Compiling Board.simEntity 
Clash: Normalisation took 0.000s 
Clash: Netlist generation took 0.000s
Clash: Total compilation took 1.370s
Preprocessing executable 'exe2' for clash-issue1686-0.0.0..
Building executable 'exe2' for clash-issue1686-0.0.0..
[1 of 5] Compiling Lib2          
[2 of 5] Compiling Lib           
[3 of 5] Compiling Board         
[4 of 5] Compiling Main          
[5 of 5] Compiling Paths_clash_issue1686
Linking .stack-work/dist/x86_64-linux/Cabal-3.2.0.0/build/exe2/exe2 ...
Preprocessing executable 'exe1' for clash-issue1686-0.0.0..
Building executable 'exe1' for clash-issue1686-0.0.0..
[1 of 5] Compiling Lib2          
[2 of 5] Compiling Lib           
[3 of 5] Compiling Board         
[4 of 5] Compiling Main          
[5 of 5] Compiling Paths_clash_issue1686
Linking .stack-work/dist/x86_64-linux/Cabal-3.2.0.0/build/exe1/exe1 ...
clash-issue1686                  > copy/register
Installing executable exe2 in /home/christiaan/devel/clash-issue-1686/.stack-work/install/x86_64-linux/e4dc479d0903f3aa912670878675db59ff6c7b1ecd31a03d3920a6d560e3b61a/8.10.2/bin
Installing executable exe1 in /home/christiaan/devel/clash-issue-1686/.stack-work/install/x86_64-linux/e4dc479d0903f3aa912670878675db59ff6c7b1ecd31a03d3920a6d560e3b61a/8.10.2/bin
Completed 114 action(s). 
```
I'll do a GHC 8.10.3 build with MR388 reverted, only realizing now that stackage lts-17.2 comes with 8.10.3, not 8.10.2

christiaanb, 2021-03-09:
It works on 8.10.3 with MR388 reverted:
```
Configuring clash-issue1686-0.0.0...
clash-issue1686                  > build (exe)
GHC: Parsing and optimising modules took: 2.115s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.000s
Clash: Parsing and compiling primitives took 0.115s
GHC+Clash: Loading modules cumulatively took 2.751s
Clash: Compiling Board.simEntity 
Clash: Normalisation took 0.000s 
Clash: Netlist generation took 0.000s
Clash: Total compilation took 2.759s
GHC: Parsing and optimising modules took: 1.015s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.000s
Clash: Parsing and compiling primitives took 0.114s
GHC+Clash: Loading modules cumulatively took 1.405s
Clash: Compiling Board.simEntity 
Clash: Normalisation took 0.000s 
Clash: Netlist generation took 0.000s
Clash: Total compilation took 1.413s
Preprocessing executable 'exe2' for clash-issue1686-0.0.0..
Building executable 'exe2' for clash-issue1686-0.0.0..
[1 of 5] Compiling Lib2          
[2 of 5] Compiling Lib           
[3 of 5] Compiling Board         
[4 of 5] Compiling Main          
[5 of 5] Compiling Paths_clash_issue1686
Linking .stack-work/dist/x86_64-linux/Cabal-3.2.1.0/build/exe2/exe2 ...
Preprocessing executable 'exe1' for clash-issue1686-0.0.0..
Building executable 'exe1' for clash-issue1686-0.0.0..
[1 of 5] Compiling Lib2          
[2 of 5] Compiling Lib           
[3 of 5] Compiling Board         
[4 of 5] Compiling Main          
[5 of 5] Compiling Paths_clash_issue1686
Linking .stack-work/dist/x86_64-linux/Cabal-3.2.1.0/build/exe1/exe1 ...
clash-issue1686                  > copy/register
Installing executable exe2 in /home/christiaan/devel/clash-issue-1686/.stack-work/install/x86_64-linux/445e94bf0b3d3034acc38b8e62ceb212a309d2612381effdd5be2b0b72a9f57d/8.10.3/bin
Installing executable exe1 in /home/christiaan/devel/clash-issue-1686/.stack-work/install/x86_64-linux/445e94bf0b3d3034acc38b8e62ceb212a309d2612381effdd5be2b0b72a9f57d/8.10.3/bin
Completed 114 action(s).  
```

christiaanb, 2021-03-09:
Alright, Tamar has given the proper "work-around" for our issue: https://mail.haskell.org/pipermail/ghc-devs/2021-March/019605.html

christiaanb, 2021-03-09:
@gergoerdi https://github.com/clash-lang/clash-compiler/pull/1695 contains changes that enables a workaround/fix. https://github.com/gergoerdi/clash-issue-1686/pull/1/files shows the actual work-around/fix. It's all based on Tamar's suggestions given in https://mail.haskell.org/pipermail/ghc-devs/2021-March/019605.html

commit 9af4aa83 (2021-03-09): Allow a configurable starting action in Clash GHC API loop

This allows things like setting a shared DynLinker for all
invocations, enabling a work-around for issue #1686

Fixes #1686
commit 8b3f5a29 (2021-03-09): Allow a configurable starting action in Clash GHC API loop

This allows things like setting a shared DynLinker for all
invocations, enabling a work-around for issue #1686

Fixes #1686
commit 9929e8bb (2021-03-10): Allow a configurable starting action in Clash GHC API loop

This allows things like setting a shared DynLinker for all
invocations, enabling a work-around for issue #1686

Fixes #1686
commit 8fecfba3 (2021-03-10): Allow a configurable starting action in Clash GHC API loop (#1695)

This allows things like setting a shared DynLinker for all
invocations, enabling a work-around for issue #1686

Fixes #1686