clash-lang/clash-compiler #1691: "No blackbox found for: GHC.List.elem"

gergoerdi, 2021-03-06:
The below Clash program fails to synthesize:

```
{-# LANGUAGE ApplicativeDo #-}
module Hardware.Compucolor2 where

import Clash.Prelude
import Clash.Annotations.TH
import Data.Function (fix)

topEntity
    :: "CLK_40MHZ" ::: Clock System
    -> "RESET"     ::: Reset System
    -> "INPUT"     ::: Signal System ()
    -> "VGA"       ::: Signal System Bool
topEntity clk rst = withClockResetEnable clk rst enableGen board
  where
    board _ffs = video (pure Nothing)

video
    :: (HiddenClockResetEnable dom)
    => Signal dom (Maybe (Unsigned 8, Unsigned 8))
    -> Signal dom Bool
video cursor = toSignal rgb
  where
    x1 = fromSignal $ fix $ \r -> register (Just 0) $ fmap (+ 1) <$> r
    y1 = fromSignal $ fix $ \r -> register (Just 0) $ fmap (+ 1) <$> r

    rgb = do
        x1 <- x1
        y1 <- y1
        cursor <- delayI Nothing $ fromSignal cursor

        pure $ case liftA2 (,) x1 y1 of
            Nothing -> False
            Just (x1, y1)
              | isCursor -> True
              | otherwise -> False
              where
                isCursor = cursor == Just (x1', y1') && y1 `elem` [minBound, maxBound]
                x1' = fromIntegral @(Unsigned 6) x1
                y1' = fromIntegral @(Unsigned 5) y1

makeTopEntity 'topEntity
```

with the following error message:

> `Clash.Netlist.BlackBox(307)`: No blackbox found for: `GHC.List.elem`. Did you forget to include directories containing primitives? You can use `-i/my/prim/dir` to achieve this.

Of course, changing the definition of `isCursor` to use `y1 == minBound || y1 == maxBound` works around this, but this is still very strange, and could point to a larger problem. 

The above program is as simple as I could get it, and there are many subtleties in cutting it down further. For example, changing the binding of `cursor` inside `rgb` to `cursor <- fromSignal cursor`, i.e. removing the `delayI Nothing`, also makes the problem go away.

christiaanb, 2021-03-06:
`GHC.List.elem` is defined recursively: http://hackage.haskell.org/package/base-4.14.1.0/docs/src/GHC.List.html#elem and thus there’s no exposed Core unfolding. We might have to define our own `Clash.List` module with our own definitions (not simply reexporting `GHC.List`) for these recursively defined functions, since we compile the clash-prelude with `-fexpose-all-unfolding`.

gergoerdi, 2021-03-06:
I think I've used `elem` before in pure code `fmap`-d on signals, that Clash was able to compile. Is this a false memory?

christiaanb, 2021-03-06:
The fact that Clash complains it’s a primitive without a blackbox means it got auto-tagged as a primitive because no unfolding was available.

A simple reproduced would be:

```
f :: Eq a => a -> [a] -> Bool
f = elem
{-# NOINLINE f #-}

topEntity :: Int -> Bool
topEntity x = f x [1,5,7]
```

(I’m on my phone, so I can’t check it myself)

gergoerdi, 2021-03-06:
Yeah you're right, I got mixed up because it turns out all uses of `elem` I can find in my code has been in simulators only.

gergoerdi, 2021-03-06:
Wait what about the code in https://github.com/clash-lang/clash-compiler/issues/477#issue-403407666? That one used `any`, which should have the same issue as `elem` by this logic, no?

christiaanb, 2021-03-06:
That one is eta-expanded, so the rewrite rule could fire.

gergoerdi, 2021-03-06:
Well I'll be... this compiles:

```
video = (\y -> any (== y) [Just minBound, Just maxBound]) <$> y1
```

What rewrite rule do you mean? If I run Clash with `-ddump-rules`, I don't see any GHC rewrite rules firing for `any`.

gergoerdi, 2021-03-06:
I guess `any` just comes with unfoldings in `base`...?

gergoerdi, 2021-03-06:
Oh, so `any` defined in terms of `foldMap`, which is `INLINE`d; whereas `elem` is a `Foldable` method that happens to have a default implementation in terms of `any`, but is not `INLINE`d or `INLINEABLE` itself.

gergoerdi, 2021-03-06:
Ooooooooh this sucks so much. Is there any light at the end of this tunnel with the "fat" `.hi` files that would contain all unfoldings?

christiaanb, 2021-03-06:
What happens if you eta-expand the use of `elem` in your original minimal reproducer?

gergoerdi, 2021-03-07:
> What happens if you eta-expand the use of `elem` in your original minimal reproducer?

Doesn't change a thing. And why would it -- ```(`f` y)``` is already syntax sugar for `\x -> f x y` since you can't do the partial application on the second argument before you get the first one. 

BTW based on your example I made the following testbed:

```
import Clash.Prelude

f :: Eq a => a -> [a] -> Bool
f x = any (== x)
{-# NOINLINE f #-}

g :: Eq a => a -> [a] -> Bool
g = elem
{-# NOINLINE g #-}

topEntity :: Int -> Bool
topEntity x = g x [1,5,7]
```

Using `f` in `topEntity` works, using `g` doesn't. I think it's because of the `NOINLINE` annotation on `GHC.List.elem`.

gergoerdi, 2021-03-07:
https://mail.haskell.org/pipermail/ghc-devs/2021-March/019592.html

christiaanb, 2021-03-07:
Ah, you're right wrt ```(`f` y)```, my bad.

The NOINLINE certainly does't help, but in my understanding of everything that annotation is pretty much a noop because `GHC.List.elem` is Recursive: I'm capitalizing because you'll argue that `GHC.List.foldr` is also defined recursively. However, in the case of `GHC.List.elem`, `elem` actually occurs on the RHS of the `=`, whereas `foldr` is defined as a call to `go`. The GHC inliner doesn't inline functions that are Recursive, nor are they included in the `.hi` interface files since there's no point since they're never inlined (even if we were to force export to `.hi` files with `-fexpose-all-unfoldings`). i.e. I believe simply dropping the `NOINLINE` is most likely not sufficient. Changing it to `INLINE` will help though, because that does force the export of the unfolding to `.hi` files I believe.

gergoerdi, 2021-03-13:
Interestingly, once we upgrade to GHC 9.0, [this particular issue with `elem` will be solved](https://mail.haskell.org/pipermail/ghc-devs/2021-March/019629.html):

>  Oh, I've found this commit:
> 
> > commit `f10d11fa49fa9a7a506c4fdbdf86521c2a8d3495`
> > Author: Andreas Klebinger `<klebinger.andreas at gmx.at>`
> > Date:   Wed Jan 29 15:25:07 2020 +0100
> > 
> >      Fix "build/elem" RULE.
> > 
> >      An redundant constraint prevented the rule from matching.
> > 
> >      Fixing this allows a call to elem on a known list to be translated
> >      into a series of equality checks, and eventually a simple case
> >      expression.
> > 
> >      [...]
> >