clash-lang/clash-compiler #1697: Complaints about extra files in directory *above* outputdir

gergoerdi, 2021-03-10:
```
Clash.defaultMain -outputdir _build/clash -Wno-partial-type-signatures -isrc --verilog Hardware.Compucolor2
GHC: Parsing and optimising modules took: 20.299s
GHC: Loading external modules from interface files took: 0.089s
GHC: Parsing annotations took: 0.067s
Clash: Parsing and compiling primitives took 0.168s
GHC+Clash: Loading modules cumulatively took 25.545s
Clash: Compiling Hardware.Compucolor2.simEntity

<no location info>: error:
    Clash error call:
    Changes were made to _build/clash/Hardware.Compucolor2.simEntity after last Clash 
    run:
    
      * Unexpected removed file _build/disk.tracks
      * Unexpected removed file _build/v678.rom.bin
      * Unexpected extra file disk.tracks
      * Unexpected extra file v678.rom.bin
```    
   
Note that these "unexpected files" are all **outside `_build/clash`**.

martijnbastiaan, 2021-03-10:
What does your manifest file look like? I don't understand how this could have happened.

gergoerdi, 2021-03-10:
# Directory structure

```
_build/
├── chargen.uf6.bin
├── clash
│   ├── Hardware
│   │   ├── Compucolor2
│   │   │   ├── CRT5027.dyn_hi
│   │   │   ├── CRT5027.dyn_o
│   │   │   ├── CRT5027.hi
│   │   │   ├── CRT5027.o
│   │   │   ├── FloppyDrive.dyn_hi
│   │   │   ├── FloppyDrive.dyn_o
│   │   │   ├── FloppyDrive.hi
│   │   │   ├── FloppyDrive.o
│   │   │   ├── Keyboard.dyn_hi
│   │   │   ├── Keyboard.dyn_o
│   │   │   ├── Keyboard.hi
│   │   │   ├── Keyboard.o
│   │   │   ├── TMS5501
│   │   │   │   ├── Controller.dyn_hi
│   │   │   │   ├── Controller.dyn_o
│   │   │   │   ├── Controller.hi
│   │   │   │   ├── Controller.o
│   │   │   │   ├── UART.dyn_hi
│   │   │   │   ├── UART.dyn_o
│   │   │   │   ├── UART.hi
│   │   │   │   └── UART.o
│   │   │   ├── TMS5501.dyn_hi
│   │   │   ├── TMS5501.dyn_o
│   │   │   ├── TMS5501.hi
│   │   │   ├── TMS5501.o
│   │   │   ├── Video
│   │   │   │   ├── Plot.dyn_hi
│   │   │   │   ├── Plot.dyn_o
│   │   │   │   ├── Plot.hi
│   │   │   │   └── Plot.o
│   │   │   ├── Video.dyn_hi
│   │   │   ├── Video.dyn_o
│   │   │   ├── Video.hi
│   │   │   └── Video.o
│   │   ├── Compucolor2.dyn_hi
│   │   ├── Compucolor2.dyn_o
│   │   ├── Compucolor2.hi
│   │   └── Compucolor2.o
│   ├── Hardware.Compucolor2.simEntity
│   │   ├── disk.tracks
│   │   ├── simEntity.manifest
│   │   ├── simEntity.sdc
│   │   ├── simEntity.v
│   │   └── v678.rom.bin
│   ├── Hardware.Compucolor2.topEntity
│   │   ├── chargen.uf6.bin
│   │   ├── disk.tracks
│   │   ├── topEntity.manifest
│   │   ├── topEntity.sdc
│   │   ├── topEntity.v
│   │   └── v678.rom.bin
│   └── ghc-deps.make
├── disk.tracks
├── nexys-a7-50t
│   └── synth
│       ├── Compucolor2
│       │   └── Compucolor2.runs
│       │       └── impl_1
│       ├── build.tcl
│       └── hs_err_pid45782.log
└── v678.rom.bin

12 directories, 53 files
```

# `simEntity.manifest`

```
Manifest {manifestHash = -4379724263988013168, successFlags = (20,20,False), inPorts = [ManifestPort {mpName = "CLK_40MHZ", mpTypeName = "", mpWidth = 1, mpIsClock = True, mpDomain = Just "Dom40"},ManifestPort {mpName = "VID_READ", mpTypeName = "[8:0]", mpWidth = 9, mpIsClock = False, mpDomain = Nothing}], outPorts = [ManifestPort {mpName = "VID_ADDR", mpTypeName = "[13:0]", mpWidth = 14, mpIsClock = False, mpDomain = Nothing},ManifestPort {mpName = "VID_WRITE", mpTypeName = "[8:0]", mpWidth = 9, mpIsClock = False, mpDomain = Nothing}], componentNames = ["simEntity"], topComponent = "simEntity", fileNames = [("simEntity.sdc","\144\181#-\139j>jy|]\138a\131\SOH\135,R;i\EOT\234'/\165w\199\217B\at\167"),("simEntity.v","\241\207\204|\235\159\130\198\152^\140\ENQ\162\177\227\136\200\158\251\DEL!\246\145S5\a6\180\SYNC%\219"),("_build/v678.rom.bin","\159|z\137?\242\145\SYNG\FS8\228\182\192L\171\SOo\ETX\158\138\224e\185\SOrB\200\219\&9\DLE\235"),("_build/disk.tracks","\228\155P0\236\235\187\ESC\235{~\227\SYN\202;\255\189\165\157\161\"<\198\157\146\131;\165\DEL\228\EM\DC2")], domains = fromList [("Dom40",VDomainConfiguration {vName = "Dom40", vPeriod = 25000, vActiveEdge = Rising, vResetKind = Asynchronous, vInitBehavior = Defined, vResetPolarity = ActiveHigh}),("System",VDomainConfiguration {vName = "System", vPeriod = 10000, vActiveEdge = Rising, vResetKind = Asynchronous, vInitBehavior = Defined, vResetPolarity = ActiveHigh}),("XilinxSystem",VDomainConfiguration {vName = "XilinxSystem", vPeriod = 10000, vActiveEdge = Rising, vResetKind = Synchronous, vInitBehavior = Defined, vResetPolarity = ActiveHigh}),("IntelSystem",VDomainConfiguration {vName = "IntelSystem", vPeriod = 10000, vActiveEdge = Rising, vResetKind = Asynchronous, vInitBehavior = Defined, vResetPolarity = ActiveHigh})]}
```

# `topEntity.manifest`

```
Manifest {manifestHash = -8290110848995791507, successFlags = (20,20,False), inPorts = [ManifestPort {mpName = "CLK_40MHZ", mpTypeName = "", mpWidth = 1, mpIsClock = True, mpDomain = Just "Dom40"},ManifestPort {mpName = "RESET", mpTypeName = "", mpWidth = 1, mpIsClock = False, mpDomain = Just "Dom40"},ManifestPort {mpName = "SWITCHES", mpTypeName = "[7:0]", mpWidth = 8, mpIsClock = False, mpDomain = Nothing},ManifestPort {mpName = "PS2_CLK", mpTypeName = "", mpWidth = 1, mpIsClock = False, mpDomain = Nothing},ManifestPort {mpName = "PS2_DATA", mpTypeName = "", mpWidth = 1, mpIsClock = False, mpDomain = Nothing}], outPorts = [ManifestPort {mpName = "VGA_HSYNC", mpTypeName = "", mpWidth = 1, mpIsClock = False, mpDomain = Nothing},ManifestPort {mpName = "VGA_VSYNC", mpTypeName = "", mpWidth = 1, mpIsClock = False, mpDomain = Nothing},ManifestPort {mpName = "VGA_DE", mpTypeName = "", mpWidth = 1, mpIsClock = False, mpDomain = Nothing},ManifestPort {mpName = "VGA_RED", mpTypeName = "[7:0]", mpWidth = 8, mpIsClock = False, mpDomain = Nothing},ManifestPort {mpName = "VGA_GREEN", mpTypeName = "[7:0]", mpWidth = 8, mpIsClock = False, mpDomain = Nothing},ManifestPort {mpName = "VGA_BLUE", mpTypeName = "[7:0]", mpWidth = 8, mpIsClock = False, mpDomain = Nothing}], componentNames = ["topEntity"], topComponent = "topEntity", fileNames = [("topEntity.sdc","\144\181#-\139j>jy|]\138a\131\SOH\135,R;i\EOT\234'/\165w\199\217B\at\167"),("topEntity.v","9\212.\SO\150:\194\247U\245\138\132\225\189\ETB\a-\165\244e\224\225}\168\210\175\168\167F\224:?"),("_build/chargen.uf6.bin","\211\&3cV\226\144\ae\240U\253\138\197S\191\182`x\191\220\221/h\n\128\181\254\159L\179\SUB\192"),("_build/disk.tracks","\228\155P0\236\235\187\ESC\235{~\227\SYN\202;\255\189\165\157\161\"<\198\157\146\131 [... cut ...] = Defined, vResetPolarity = ActiveHigh}),("XilinxSystem",VDomainConfiguration {vName = "XilinxSystem", vPeriod = 10000, vActiveEdge = Rising, vResetKind = Synchronous, vInitBehavior = Defined, vResetPolarity = ActiveHigh}),("IntelSystem",VDomainConfiguration {vName = "IntelSystem", vPeriod = 10000, vActiveEdge = Rising, vResetKind = Asynchronous, vInitBehavior = Defined, vResetPolarity = ActiveHigh})]}
```

BTW do you maybe want to base64-encode those hashes or print them in hex or something? :)

gergoerdi, 2021-03-10:
Note that `disk.tracks` and `v678.rom.bin` were copied into the manifest directories from `_build` by Clash itself, since they are used in `romFromFile`s.

gergoerdi, 2021-03-10:
Ah, maybe that's the problem? That Clash records these file names in the manifest as `_build/disk.tracks`, whereas once they are copied to the manifest dir, they are just `disk.tracks`?

gergoerdi, 2021-03-10:
Indeed, if I manually edit the manifest file (which, hilariously, is not detected by a hash change) to refer to the files without the `_build/` prefix, then it goes through.

martijnbastiaan, 2021-03-10:
> BTW do you maybe want to base64-encode those hashes or print them in hex or something? :)

This is tackled in https://github.com/clash-lang/clash-compiler/pull/1694 :)

> Ah, maybe that's the problem? That Clash records these file names in the manifest as _build/disk.tracks, whereas once they are copied to the manifest dir, they are just disk.tracks?

Right. This:

https://github.com/clash-lang/clash-compiler/blob/8fecfba36179973004508643f8854a3d8222c97d/clash-lib/src/Clash/Driver.hs#L434

should be `map fst dfiles`.

> Indeed, if I manually edit the manifest file (which, hilariously, is not detected by a hash change) to refer to the files without the _build/ prefix, then it goes through.

I figured, if you're going to edit manifest files you're on your own :P

> I put the files in _build (by various Shake rules)

Is this because the HDL refers to the datafiles as just "disk.tracks" and EDA tools look relative to CWD and not relative to the HDL file?

gergoerdi, 2021-03-11:
> > I put the files in _build (by various Shake rules)
> 
> Is this because the HDL refers to the datafiles as just "disk.tracks" and EDA tools look relative to CWD and not relative to the HDL file?

The workflow is as follows:

* The real source files are binary images: a sequence of bytes where the first byte is the 8 bits to be stored at ROM address 0, the second byte is the 8 bits to be stored at ROM address 1, and so on. Only these files are tracked by source control.

* A Shake rule creates `.bin` files from the above, which are in the format expected by Clash: each line contains 8 characters, each character is `0` or `1`. These files need to live under `_build` because I want a single nukable directory to contain all build artifacts. Everything under `_build` is replacable, and everything not under `_build`  should be version-controlled.

* I refer to these files from Clash as `_build/foo.bin` to make sure that both the Clash compiler, and Clash simulations, when run from the top-level project directory, will correctly find them and copy them next to the generated HDL files. It seems Clash correctly transforms the name to just `foo.bin`, so downstream tools look for it right next to the HDL file which is where they are copied. 

The one weak part in all these is supporting Clashilator: since the generated HDL refers to these files as `foo.bin`, the Verilator output will try to look for them in the current directory. I haven't looked into ways of solving this yet; ideally, I would get either Verilator or Clash to generate absolute paths (pointing into the clashilation directory).

martijnbastiaan, 2021-03-11:
> The one weak part in all these is supporting Clashilator: since the generated HDL refers to these files as foo.bin, the Verilator output will try to look for them in the current directory. I haven't [...]

We had a bit of discussion about this a few days ago. What if Clash would generate `../$fullyqualifiedname/foo.bin` in the HDL? This would work for tools interpreting the filename as relative to CWD as well as those interpreting it as relative to the HDL file. Provided that you'd run the CWD-style simulators in a sibling directory.

commit 47f9c887 (2021-03-10): Use correct filename for data files in manifest

Fixes #1697
PR #1698 (2021-03-10): Use correct filename for data files in manifest
Fixes #1697
commit efcdd30a (2021-03-10): Use correct filename for data files in manifest (#1698)

Fixes #1697