clash-lang/clash-compiler #1742: Annotate attributes for types must be in normal form

thoughtpolice, 2021-04-03:
Consider the following attempt to abstract out an unwieldly use of `Annotate` into an easy-to-use `Pin` type:

```haskell
import           Clash.Prelude
import           Clash.Signal

import           Clash.Annotations.SynthesisAttributes
import           Clash.Annotations.TopEntity
import           Clash.Annotations.TH

createDomain vSystem{vName="Bank2C", vPeriod=20000, vResetKind=Asynchronous, vResetPolarity=ActiveLow}

type Pin t x y = t `Annotate` 'StringAttr "chip_pin" x
                   `Annotate` 'StringAttr "altera_attribute" ("-name IO_STANDARD \"" `AppendSymbol` y `AppendSymbol` "\"")

top :: HiddenClockResetEnable dom => Signal dom (BitVector 4)
top = s where s = register 0 (s + 1)

topEntity
  :: "CLK_50_B2C"  ::: Pin (Clock Bank2C) "PIN_AW38" "1.2 V"
  -> "CPU_RESET_n" ::: Signal Bank2C Bool
  -> "LED"         ::: Signal Bank2C (BitVector 4)
topEntity clk rstnButton = exposeClockResetEnable top clk rst en where
  en = enableGen
  rst = unsafeFromLowPolarity rstnButton
  rstSync = resetSynchronizer clk rst en
makeTopEntityWithName 'topEntity "shell"
```

Using the tip of the 1.4 branch, I get:

```
$ clash --verilog DE10Pro
GHC: Parsing and optimising modules took: 1.584s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.000s
Clash: Parsing and compiling primitives took 0.147s

<no location info>: error:
    Clash error call:
    Could not unpack given type to string: AppendSymbol (AppendSymbol "-name IO_STANDARD \"" "1.2 V") "\""
    CallStack (from HasCallStack):
      error, called at src-ghc/Clash/GHC/GHC2Core.hs:775:19 in clash-ghc-1.4.0-DkSmlpJFpg8Ad9ypRSNVx8:Clash.GHC.GHC2Core
```

I don't see why this can't be supported; it seems that the evaluator merely assumes that the string type is already in normal form, which isn't the case here. My actual goal is to use closed type families to write a more sophisticated version of `Pin`, but the basic idea here is all the same.

christiaanb, 2021-04-03:
Yeah, hopefully something we can easily solve using some calls to: http://hackage.haskell.org/package/ghc-8.10.2/docs/FamInstEnv.html#v:topNormaliseType in the right place

christiaanb, 2021-04-03:
I don’t have access to a PC until Monday though

martijnbastiaan, 2021-04-03:
Should we add support for `AppendSymbol` to `reduceTypeFamily` instead? This would make us handle polymorphic (but monomorphic after propagation ofc) as well.

christiaanb, 2021-04-03:
It won’t help in this case, the error is thrown in a part of the compiler that still deals with GHC Core data types.

christiaanb, 2021-04-03:
Also, it already supports `AppendSymbol`: https://github.com/clash-lang/clash-compiler/blob/3cf6cef6de74883df5ea10ec587ae55cb331fb2c/clash-lib/src/Clash/Core/Type.hs#L570

commit 978b6e7a (2021-04-05): Normalize `Annotate` attributes

Fixes #1742
PR #1745 (2021-04-06): Normalize `Annotate` attributes
Fixes #1742
commit b0cb707a (2021-04-06): Normalize `Annotate` attributes

Fixes #1742
commit 2647e2b8 (2021-04-06): Normalize `Annotate` attributes

Fixes #1742

(cherry picked from commit b0cb707a5c26f3c311069fbb97e3d6f4fc6d02ab)
closed 2021-04-06 by martijnbastiaan