clash-lang/clash-compiler #1743: Derive optimal bit representations automatically

rowanG077, 2021-04-04:
Clash currently is not smart enough to figure out the most optimal way to represent a value of a type as a bitstring. This wastes some space in a lot of cases. To alleviate this user can manually create a bit representation using an [annotation](http://hackage.haskell.org/package/clash-prelude-1.4.0/docs/Clash-Annotations-BitRepresentation.html).

But it would be nice if Clash would always be able to derive the most efficient encoding by itself. There is some prior art here. Rust has an optimization it calls `NicheFilling` that basically does this.

christiaanb, 2021-04-04:
What’s _optimal_? One that uses the least amount of bits to store? http://hackage.haskell.org/package/clash-prelude-1.4.0/docs/Clash-Annotations-BitRepresentation-Deriving.html#v:packedDerivator One that incurs the fewest multiplexers? http://hackage.haskell.org/package/clash-prelude-1.4.0/docs/Clash-Annotations-BitRepresentation-Deriving.html#v:Wide

rowanG077, 2021-04-04:
I was under the impression that the default `Derivator` was the `packedDerivator` but that is not the case. I guess I misrembered reading that in the docs.

closed 2021-04-04 by rowanG077