clash-lang/clash-compiler #1750: TestBench annotation confusion

leonschoorl, 2021-04-06:
There seems to be some confusion about the direction of the TestBench annotation.
Is it:
1. an annotation on a `Synthesize`d binder pointing to its testbench
2. an annotation on a testbench pointing to the thing that it is testing

The [current docs](https://hackage.haskell.org/package/clash-prelude-1.4.0/docs/Clash-Annotations-TopEntity.html#v:TestBench) state:

> Tell what binder is the TestBench for a Synthesize-annotated binder.

Not very clear.
But luckily there is also an example:
> ..., f has a Synthesize annotation, and g is the HDL test bench for f.
> ```haskell
> f = ...
> {-# ANN f (defSyn "f") #-}
> {-# ANN f (TestBench 'g) #-}
> 
> g :: Signal Bool
> g = ...
> ```
This is option 1.
This work, and clash will create HDL output for both `f` and `g`, but it will produce a warning:
```
Clash: Compiling TestNames.g
Clash: Normalization took 0.001s
[WARNING] Dubious primitive instantiation for Clash.Explicit.Testbench.tbClockGen: Clash.Signal.Internal.tbClockGen is not synthesizable! (disable with -fclash-no-prim-warn)
Clash: Netlist generation took 0.000s
Clash: Compiling TestNames.f
Clash: Normalization took 0.001s
Clash: Netlist generation took 0.000s
```

If we try option 2:
```haskell
f = ...
{-# ANN f (defSyn "f") #-}

g :: Signal Bool
g = ...
{-# ANN g (TestBench 'f) #-}
```
This also work, clash will also generate HDL files for both `f` and `g`, but this time the netlist code actually considers g to be a testbench and suppresses the `tbClockGen` warning: 
```
Clash: Compiling TestNames.g
Clash: Normalization took 0.001s
Clash: Netlist generation took 0.000s
Clash: Compiling TestNames.f
Clash: Normalization took 0.001s
Clash: Netlist generation took 0.000s
Clash: Total compilation took 0.957s
```

martijnbastiaan, 2021-04-06:
Good points. I've always interpreted it as (2) and used that for implementations. 

At least in the current implementation there's only one reason to use a test bench annotation: to suppress warnings of dubious blackbox instantiations. Another feature could be (which is currently not implemented) that `-main-is f` would also compile all of `f`'s test benches.

My proposal would be to entirely eliminate the testbench annotation. Instead, we could add a field `Synthesize.isTestbench` which would disable the aforementioned warning. I'm not convinced the second benefit I mentioned is really worth the effort. We could replace it with the ability to do `-main-is=f,g` though.

DigitalBrains1, 2022-04-13:
Currently, a
```haskell
{-# ANN myTestBench (TestBench 'entityBeingTested) #-}
```
annotation in a sense "implies" a `Synthesize` annotation on `entityBeingTested`, so if we adopt the `Synthesize.isTestBench` approach, the semantics slightly change. We need to account for that in a few test benches I wrote in `clash-testsuite`, and users need to be made aware.

closed 2022-04-21 by DigitalBrains1

martijnbastiaan, 2022-08-30:
We've released [v1.6.4](https://github.com/clash-lang/clash-compiler/releases/tag/v1.6.4), which includes a fix for this issue.

commit 5b7a5e3e (2022-04-14): Fix `ANN TestBench` documentation

The documentation had it backwards, the `TestBench` annotation goes on
the test bench, not on the device under test.

Fixes #1750
commit c430fcf4 (2022-04-14): Fix `ANN TestBench` documentation

The documentation had it backwards, the `TestBench` annotation goes on
the test bench, not on the device under test.

Fixes #1750
PR #2171 (2022-04-21): Fix `ANN TestBench` documentation
The documentation had it backwards, the `TestBench` annotation goes on
the test bench, not on the device under test.

Fixes #1750

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files
commit ee87e5b2 (2022-04-21): Fix `ANN TestBench` documentation (#2171)

The documentation had it backwards, the `TestBench` annotation goes on
the test bench, not on the device under test.

Fixes #1750