clash-lang/clash-compiler #1756: *** Exception: Clash.Netlist.BlackBox(201): Couldn't instantiate blackbox

Hendrik-0, 2021-04-08:
VHDL compilation gives the error:
```
*** Exception: Clash.Netlist.BlackBox(201): Couldn't instantiate blackbox for Clash.Signal.Internal.register#. Verification procedure reported:

Argument 5 should be literal, as blackbox used ~CONST[5], but was:

Identifier (UniqueIdentifier {i_baseName = "a_res", i_baseNameCaseFold = "a_res", i_extensionsRev = [], i_idType = Basic, i_hdl = VHDL, i_provenance = []}) Nothing
```
Verilog compiles just fine.

I  tried to reduce the issue to a simple nonsensical example:
```
topEntity :: Clock  System
          -> Reset  System
          -> Signal System (Vec 1 Bool)
topEntity clk rst = exposeClockResetEnable g clk rst enableGen 

g :: HiddenClockResetEnable dom
  => Signal dom (Vec 1 Bool)
g = out
  where
    a  = register (unconcatI $ singleton (1:: Bit)) a'
    a' = a
    b = pure $ replicate d1 $ replicate d1 True
    (_, out) = unbundle $ f <$> a <*> b

f :: (KnownNat n, KnownNat l, KnownNat m, KnownNat k
  , (m * k) ~ (n * l)) 
  => Vec m (Vec k a1)
  -> Vec m (Vec k a2) 
  -> (Vec m (Vec k a1), Vec l a2)
f a b = (a', out)
  where
    a' = (unconcatI . concat) a
    b' = (unconcatI . concat) b
    out = b'!!0
```
Minor changes to the code fixes the exception, but this example reproduces the error (version 1.4.0).

lesjon, 2021-04-08:
For me on Clash 1.2.5 it reports
`
*** Exception: Clash.Netlist.BlackBox(188): Couldn't instantiate blackbox for Clash.Signal.Internal.register#. Verification procedure reported:
Argument 5 should be literal, as blackbox used ~CONST[5], but was:
Identifier "a_res" Nothing`

but also works for verilog.

christiaanb, 2021-04-08:
Yeah, Clash is missing a compile-time unrolling path for  `unconcat xs :: Vec n (Vec m a)` when `m` is anything else than `0`. Shouldn't be too hard to fix.

closed 2021-04-09 by christiaanb

Hendrik-0, 2021-04-09:
I am not sure if this is the same bug, but the code below results in the same type of error message for both verilog and vhdl:
```
fme s i = (s, s)
fmealy index inp = mealy fme index inp
fmes inp = bundle $ imap fmealy (replicate d2 inp)

topEntity 
  :: Clock System
  -> Reset System
  -> Signal System Int
  -> Signal System (Vec 2 (Index 2))
topEntity clk rst inp = exposeClockResetEnable fmes clk rst enableGen inp
```
error: 

```
*** Exception: Clash.Netlist.BlackBox(201): Couldn't instantiate blackbox for Clash.Signal.Internal.register#. Verification procedure reported:

Argument 5 should be literal, as blackbox used ~CONST[5], but was:

Identifier (RawIdentifier "~ARGN[1][0]" Nothing []) Nothing
```

christiaanb, 2021-04-09:
No, that problem/bug is the dual of the error reported here. The problem we saw here in #1756 that Clash didn't unroll a function, for which it has a blackbox, in this case `unconcat`, when it is used as the initial/reset value of a `register`. And the bug you report above is when register uses one of the arguments generated by a function for which Clash has a blackbox, in the above case `imap`, as the initial/reset value of a `register`.

Please report it as a new issue.

commit 548ebd07 (2021-04-09): Unroll `unconcat` in more situations than `m=0`

Fixes #1756
commit 40dcdc4e (2021-04-09): Unroll `unconcat` in more situations than `m=0`

Fixes #1756
PR #1759 (2021-04-09): Unroll `unconcat` in more situations than `m=0`
Fixes #1756
commit 12fa8907 (2021-04-09): Unroll `unconcat` in more situations than `m=0`

Fixes #1756