clash-lang/clash-compiler #1757: Forgetting -fclash-float-support should warn

DigitalBrains1, 2021-04-08:
If you have a project that legitimately uses `Float` and forget to include the command-line parameter `-fclash-float-support`, the generated HDL will silently omit anything that works with `Float`s. It would be much better if Clash at least warned that something seems to go wrong.

christiaanb, 2021-04-09:
Once we have some open-source floating-point operators/wrappers for vendor FP IP we should just deprecate `-fclash-float-support` and handle `Float` as if `-fclash-float-support` is enabled by default.

DigitalBrains1, 2021-10-05:
Commit 8ddb5d131 from 14 July made addition, subtraction, multiplication and division with Xilinx IP available in `clash-cores`. I think it's time we made the change to make float support the default.

alex-mckenna, 2021-10-05:
Sounds good. Is there anywhere we should warn about incomplete support (i.e. if we don't support any FP ops when targeting vivado it might be worth warning unless `-fclash-hdlsyn Vivado` is set)?

alex-mckenna, 2021-10-05:
This isn't something that needs an answer now though, more something to consider if you're making a PR to deprecate the flag :slightly_smiling_face:

DigitalBrains1, 2021-10-05:
I don't really understand what you mean, but I don't think we should in general expect people to use `-fclash-hdlsyn` other than when they run into specific incompatibility issues and want to get rid of them. In other words, I don't ever use that flag when I'm using Vivado.

Back to me not understanding, could you elaborate with an example, hypothetical or not, what people should do or not do to encounter such a warning?

alex-mckenna, 2021-10-05:
Ah wait, I forgot that these primitives are only used explicitly (i.e. not instantiated when using operators like `(+)` in code). In that case nothing to be done

DigitalBrains1, 2021-10-05:
Using the regular operators, you currently get this (with `-fclash-float-support --vhdl`):
```
PlainFloat.hs:9:1: error:
    
    Clash.Netlist.BlackBox(311): No blackbox found for: GHC.Prim.plusFloat#. Did you forget to include directories containing primitives? You can use '-i/my/prim/dir' to achieve this.
    
    CallStack (from HasCallStack):
      extractPrimWarnOrFail, called at src/Clash/Netlist/Util.hs:841:25 in clash-lib-1.5.0-inplace:Clash.Netlist.Util
    
    
    The source location of the error is not exact, only indicative, as it is acquired 
    after optimizations. The actual location of the error can be in a function that is 
    inlined. To prevent inlining of those functions, annotate them with a NOINLINE pragma.
  |
9 | topEntity = (+)
  | ^^^^^^^^^
```

The actual float support for Xilinx is indeed with explicitly used functions.

alex-mckenna, 2021-10-05:
Hmm, I wonder if it would be worth opening a new issue for changing the warnings on these Haskell FP prims. It might be more helpful if they instead warned they were Clash simulation only, and if you want to "synthesize" in Clash you should be using an explicit floating point primitive. Makes it clearer to users that it's deliberately missing

acairncross, 2021-12-08:
I agree that the current `-fclash-float-support` behaviour should be made the default.

The current default of silently removing floating point signals is surprising to me:

```haskell
module Test where

import Clash.Prelude

topEntity :: Signal System Float
topEntity = pure 1
```

produces

```vhdl
-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.test_types.all;

entity topentity is

architecture structural of topentity is

begin

end;
```

Handling floating point operations seems like a separate issue to me.

DigitalBrains1, 2021-12-09:
Thanks for your comment! We've already decided that making float support the default is the right thing to do; it is just waiting for someone to have a moment to work on it!

PR #2048 (2022-01-24): Remove `-fclash-float-support`
Closes #1757

## Still TODO:

  - [X] Write a changelog entry (see changelog/README.md)
  - [x] ~~Check copyright notices are up to date in edited files~~ Such small changes..
commit 632a5355 (2022-01-24): Remove -fclash-float-support (#2048)

Closes #1757
closed 2022-01-24 by alex-mckenna