clash-lang/clash-compiler #1766: Clash generates duplicate singal names

martijnbastiaan, 2021-04-14:
I've got a fairly small example that triggers Clash to generate duplicate signal names:

```haskell
module Knightrider where

import Clash.Class.Counter
import Clash.Explicit.Prelude

modulate :: Unsigned 4 -> Unsigned 4 -> (Unsigned 4, Bool)
modulate counter threshold = (counter + 1, counter < threshold)

modulateVec :: Unsigned 4 -> Vec 8 (Unsigned 4) -> (Unsigned 4, Vec 8 Bool)
modulateVec counter thresholds = (counter + 1, map (counter <) thresholds)

distance :: (Ord a, Num a) => a -> a -> a
distance a b = if a > b then a - b else b - a

brightness :: Unsigned 3 -> Unsigned 3 -> Unsigned 4
brightness epoch led =
  maxBound - satMul SatBound distanceToEpoch stepSize
 where
  distanceToEpoch = distance (extend epoch) (extend led)
  stepSize = 4 -- 2^4 / 4

slope :: Unsigned 3 -> Vec 8 (Unsigned 4)
slope epoch = map (brightness epoch) (iterateI (+1) 0)

knightrider ::
  Clock System ->
  Reset System ->
  Enable System ->
  Signal System (BitVector 8)
knightrider clk rst ena = pack <$> leds
 where
  leds = mealy clk rst ena modulateVec 0 (slope <$> ledPos)

  initial :: (Unsigned 3, Index 10)
  initial = (0, 0)

  (ledPos, delayCount) =
      unbundle
    $ register clk rst ena initial
    $ fmap countSucc
    $ bundle (ledPos, delayCount)

knightriderTop ::
  Clock System ->
  Signal System Bool ->
  Signal System (BitVector 8)
knightriderTop clk rstBool = knightrider clk rst enableGen
 where
  rst = unsafeFromLowPolarity rstBool
```

Compile as:

```
clash -- Test.hs --verilog -main-is knightriderTop
```

Note that

```
clash -- Test.hs --verilog -main-is knightrider
```

doesn't exhibit the behavior. Signal declarations in the Verilog:

```verilog
  // Test.hs:32:1-11
  wire  rst;
  wire [7:0] result;
  wire [7:0] c$app_arg;
  wire [7:0] c$ds_app_arg;
  // Test.hs:12:1-11
  reg [3:0] counter = 4'd0;
  wire [31:0] result;
  // Test.hs:32:1-11
  reg [6:0] ds = {3'd0,   4'd0};
  // Test.hs:32:1-11
  wire [6:0] c$ds_app_arg;
  // Test.hs:32:1-11
  wire [2:0] c$ds_app_arg;
  // Test.hs:32:1-11
  wire [2:0] c$ds_case_alt;
  // Test.hs:32:1-11
  wire  c$ds_case_scrut;
  // Test.hs:32:1-11
  wire [2:0] ledPos;
  wire [23:0] result_res;
```

I've seen this type of behavior before when Clash hit unsynthesizable code, but I'm not sure it's related. I'll try to minimize the example tomorrow - although I've already noticed lots of small changes make the behavior go away.

christiaanb, 2021-04-14:
I looked at the debug log: the issue is a nested let-binding, which shouldn't exist in normalized expressions. There's two solutions:
1. Do correct unique name generation for nested let-bindings in `Nestlist`
2. Have `flattenLet` also flatten `let xs in (let ys in e)` to `let xs ++ ys in e`; be careful to properly deshadow.

alex-mckenna, 2021-04-15:
> knightriderTop

So you don't support the upgrade for KITT [without the top on](https://2.bp.blogspot.com/-2AC07fTbJ8I/TdFEvZ6uhZI/AAAAAAAAA1A/GECf-n7szfU/s1600/Knight_Rider.jpg)?

gergoerdi, 2021-04-15:
> > knightriderTop
> 
> So you don't support the upgrade for KITT [without the top on](https://2.bp.blogspot.com/-2AC07fTbJ8I/TdFEvZ6uhZI/AAAAAAAAA1A/GECf-n7szfU/s1600/Knight_Rider.jpg)?

You don't want to accidentally end up with [another kind of topless Knight Rider](https://2.bp.blogspot.com/-tcQLazXv3JI/XHSsV65zV1I/AAAAAAADcoQ/QiJx-2gkTE4FCqXqrVbT9XL4XLETAChewCLcBGAs/s1600/young-david-hasselhoff-7.jpg).

commit 3ae619f6 (2021-06-02): flatten nested letrec (fixes #1766)

The flattenLet transformation now also flattens nested letrec
expressions into a single letrec, before applying the previous
behaviour of flattening a non-nested letrec.
PR #1837 (2021-06-02): flattenLet nested letrec; error on unexpected change
The flattenLet transformation now also flattens nested letrec
expressions, by deshadowing and merging the bindings into a single
expression.

The error for when setChanged is not called was never being called,
this is now fixed so that transformations that change the core
without declaring throw an error at runtime.

Fixes: #1766 

## Still TODO:

  - [ ] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files
commit fce72d5d (2021-06-02): flatten nested letrec (fixes #1766)

The flattenLet transformation now also flattens nested letrec
expressions into a single letrec, before applying the previous
behaviour of flattening a non-nested letrec.
closed 2021-06-02 by alex-mckenna