clash-lang/clash-compiler #1770: Allow users to derive tuple instances themselves

martijnbastiaan, 2021-04-16:
We currently disable `-flarge-tuples` on releases. If a user needs a bit more tuple instances, they have no choice but to either enable `-flarge-tuples` and generate them all, or go through internal functions / manually write instances. We could:

* Create a new module exporting TH functions generating tuple instances (one for `NFDataX`, `Counter`, `BitPack`, etc)
* Export an extra function that simply creates instances for _all_ classes 
* Make the functions check whether there already exists an instance for certain tuple sizes, which would make the code safe to use anywhere - even though it's generating orphan instances.

alex-mckenna, 2021-04-16:
Assuming people only need a specific few extra instances, I would lean towards exposing the template Haskell and letting people make orphan instances. That way they aren't burdened with having instances for all sizes / classes generated.

To play devil's advocate: what's wrong with just making people use `-flarge-tuples`?

basile-henry, 2021-04-16:
> To play devil's advocate: what's wrong with just making people use `-flarge-tuples`?

I am guessing 62/64 for the tuple size is likely too many for most users, I think we could give more fine grained control by using configure flags. Since `clash-prelude` is already using a `Custom` build type, I don't think that would be too involved.

We could imaging having the user specify: `--configure-option="max-tuple-size-instance=27"` or something along those lines.

In terms of implementation I think something using [`defaultMainWithHooks`](https://hackage.haskell.org/package/Cabal-3.4.0.0/docs/Distribution-Simple.html#v:defaultMainWithHooks), and one of the hooks to control `cppOptions` in [`buildInfo`](https://hackage.haskell.org/package/Cabal-3.4.0.0/docs/Distribution-Types-BuildInfo.html#t:BuildInfo) should do the trick :smile:

martijnbastiaan, 2021-04-17:
> Assuming people only need a specific few extra instances, I would lean towards exposing the template Haskell and letting people make orphan instances. That way they aren't burdened with having instances for all sizes / classes generated.

Right, my proposal was meant to be and/and. So I'd imagine exporting functions to generate instances for specific types, but then also a convenience function that generates them for all types.

> To play devil's advocate: what's wrong with just making people use -flarge-tuples?

It makes Clash unusable on low-memory (our interns can confirm :-)). On Windows it triggers a TH bug that stops compilation.

Thanks for the suggestions @basile-henry! I'm biased against introducing flags like those, as it's usually a struggle to find out how to pass them in the context of the various build systems used in the Haskell ecosystem (Stack/Nix/Bazel/Shake, maybe even custom scripts). Perhaps it could just be another option in the toolbox though :)

martijnbastiaan, 2021-07-03:
As @christiaanb noted in a private channel, allowing users to derive tuple instances causes problems when depending on two projects generating conflicting (as far as GHC can see) tuple instances. I'm therefore closing this issue. I'll reopen a separate, related discussion issue soon.

closed 2021-07-03 by martijnbastiaan