clash-lang/clash-compiler #1772: `bLit` should handle `0x`, `0b`, and `0o` literals

martijnbastiaan, 2021-04-19:
With `BinaryLiterals` and friends we can write:

```haskell
0xFF
0o345
0b0012
```

but [`bLit`](https://hackage.haskell.org/package/clash-prelude-1.4.1/docs/Clash-Sized-BitVector.html#v:bLit) only supports binary literals. Ideally it would support all of them.

basile-henry, 2021-04-19:
What about have similar `hexLit`/`hLit` and `octLit`/`oLit` functions instead?

martijnbastiaan, 2021-04-19:
Good idea. I like the more explicit prefixes too!

leonschoorl, 2021-04-20:
Slightly related, I've been wondering if it would be more convenient if `bLit` took its length information from the provided string. 

Right now `bLit` needs to get the length information from its context, and it will zero extend or truncate the provided value as needed:
```
clashi> $$(bLit "10")

<interactive>:9:4: error:
    • No instance for (KnownNat n0) arising from a use of ‘bLit’
    • In the expression: bLit "10"
      In the Template Haskell splice $$(bLit "10")
      In the expression: $$(bLit "10")

clashi> $$(bLit "10") :: BitVector 2
10
clashi> $$(bLit "10") :: BitVector 4
0010
clashi> $$(bLit "10") :: BitVector 1
0
```

Instead we make the resulting type depend on the amount of bits provide:
```
clashi> :set +t
clashi> -- ^ this make it print the result type

clashi> $(bLit "10")
10
it :: BitVector 2

clashi> $(bLit "10") :: BitVector 4
  error:
    • Couldn't match type ‘2’ with ‘4’
      Expected: BitVector 4
        Actual: BitVector 2

clashi> $$(bLit "0010")
0010
it :: BitVector 4
```

basile-henry, 2021-04-20:
I like inferring the type from the String length very much! :smile: It's more concise and probably what a user would expect. We can easily recover the old behavior by wrapping with an explicit `resize` if needed.

martijnbastiaan, 2021-04-20:
We should absolutely do this! I was very surprised it didn't do this while writing a tutorial the other day.

leonschoorl, 2021-04-20:
We even have both.
But in that case I'd like to follow `Clash.Sized.Vector`'s pattern of having  `foo` and `fooI`.
Where `fooI` is `foo` except it takes its size information implicitly from context.

christiaanb, 2021-04-20:
Yes, it would have to be an untyped splice, instead of the typed splice it is now. because of things like:
```
>>> $(bLit (List.replicate 4 '0'))
0000
```

PR #2505 (2023-08-12): Add hex and octal BitVector parsing per issue 1772
Automatically switching between hex, octal and binary based on the prefix, akin to how BinaryLiterals does it, may be a better option. The API functions added are hLit and oLit, by analogy with bLit.

Fixes: #1772 

## Still TODO:
  - [ ] Write a changelog entry (see changelog/README.md)
  - [ ] Check copyright notices are up to date in edited files
commit 43145e7d (2023-08-12): Add hex and octal BitVector parsing (#2505)

Fixes #1772
closed 2023-08-12 by martijnbastiaan