clash-lang/clash-compiler #1773: Show instance of `BitVector` produces misleading results and sometimes invalid Haskell

martijnbastiaan, 2021-04-19:
Example of misleading results:

```haskell
>>> 0xFF :: BitVector 8
1111_1111
>>> 1111_1111 :: BitVector 8
1100_0111
```

This could be fixed by pretending `0b`. Hypothetical new situation:

```haskell
>>> 0xFF :: BitVector 8
0b1111_1111
>>> 0b1111_1111 :: BitVector 8
0b1111_1111
```

As a second issue, `Show` instances produces invalid Haskell in presence of undefined bits:

```haskell
>>> pack (Nothing :: Maybe (BitVector 7))
0..._....
>>> 0..._....
<error>
>>> 0b0..._....
<error>
```

Hypothetical new situation:

```haskell
>>> pack (Nothing :: Maybe (BitVector 7))
$$(bLit "0..._....")
```

or perhaps:

```haskell
>>> pack (Nothing :: Maybe (BitVector 7))
fromInteger# 0b0111_1111 0b0000_0000
```

paddytheplaster, 2021-04-20:
Hi Martijn,

Thanks for the suggestion.

A related problem is that `Show` instances of `Vec` also output invalid Haskell. For me this is a real issue, which I reported in a previous thread.

Regards,

Paddy

leonschoorl, 2021-04-20:
That first example is horrible.

Although when I try it I get:
```
$ clash.clashi 
Clashi, version 1.4.1 (using clash-lib, version 1.4.1):
https://clash-lang.org/  :? for help
Clash.Prelude> 0xFF :: BitVector 8
1111_1111
Clash.Prelude> 1111_1111 :: BitVector 8

<interactive>:2:1: error:
    Use NumericUnderscores to allow underscores in integer literals
Clash.Prelude> 
```

But if you have NumericUnderscores enabled it could be a horrible source of bugs when you copy-paste some pre-computed constants from clashi into your code.

leonschoorl, 2021-04-20:
NumericUnderscores doesn't "save" you with smaller numbers:
```
Clash.Prelude> 0xf :: BitVector 4
1111
Clash.Prelude> 1111 :: BitVector 4
0111
```

martijnbastiaan, 2021-04-21:
> A related problem is that Show instances of Vec also output invalid Haskell. For me this is a real issue, which I reported in a previous thread.

@paddytheplaster I've merged PRs https://github.com/clash-lang/clash-compiler/pull/1778, https://github.com/clash-lang/clash-compiler/pull/1779, and https://github.com/clash-lang/clash-compiler/pull/1783. Together they make sure `show` and `showX` on a `Vec` produces valid (copyable) Haskell. This is not yet true when the `Vec` contains a `BitVector` as can be seen by this issue still being open. There's still some discussion on what to do here; I'll keep you updated.

paddytheplaster, 2021-04-29:
> > A related problem is that Show instances of Vec also output invalid Haskell. For me this is a real issue, which I reported in a previous thread.
> 
> @paddytheplaster I've merged PRs #1778, #1779, and #1783. Together they make sure `show` and `showX` on a `Vec` produces valid (copyable) Haskell. This is not yet true when the `Vec` contains a `BitVector` as can be seen by this issue still being open. There's still some discussion on what to do here; I'll keep you updated.
> 
> Edit: These changes won't be backported to 1.4 as they're an API change.

Thanks Martijn. I just installed the most recent development version. Show now works for `Vec`s, which should save a few KB of code, because I won't have to override `Show` for lots of classes any more.

I have one suggestion for improvement. With the current implementation, we get " :> " as a delimiter. With ":>" as a delimiter, we can put more showed elements on a line/screen. To me ":>" is more in spirit with the previous implementation, which used ",". 

- Personally, I'd prefer ":>".
- Maybe it's possible to specify/specialise the delimiter at compilation time of the Clash compiler itself?
- Then again, for compatibility reasons (bug reports etc) among different compilers it may be better to disallow delimiter specialisation.

Just a suggestion.

Regards,

Paddy

martijnbastiaan, 2021-04-29:
We've followed how most languages format lists/vectors (Python, JavaScript, Java, Rust). IMO these languages got it right - a little bit of whitespace makes it easier to read. 

I'm happy to hear the change is otherwise working out for you!