clash-lang/clash-compiler #1776: Show instance of `Vec` produces invalid Haskell

martijnbastiaan, 2021-04-20:
As reported by @paddytheplaster in https://github.com/clash-lang/clash-compiler/issues/1773:

```haskell
>>> 1 :> 2 :> 3 :> Nil
<1,2,3>
>>> <1,2,3>

<interactive>:2:1: error: parse error on input ‘<’
```

christiaanb, 2021-04-20:
The only real alternative is
```
>>> 1 :> 2 :> 3 :> Nil
1 :> 2 :> 3 :> Nil
```
because, while
```
>>> 1 :> 2 :> 3 :> Nil
$(listToVecTH [1,2,3])
>>> $(listToVecTH [1,2,3])
$(listToVecTH [1,2,3])
```
would work in the interpreter, the moment you do:
```haskell
module Test11 where

import Clash.Prelude

v = $(listToVecTH [1,2,3])
```
you get the following error when you load that file:
```
[1 of 1] Compiling Test11           ( Test11.hs, interpreted )

Test11.hs:5:7: error:
    • Ambiguous type variable ‘a0’ arising from a use of ‘listToVecTH’
      prevents the constraint ‘(Lift a0)’ from being solved.
      Probable fix: use a type annotation to specify what ‘a0’ should be.
      These potential instances exist:
        32 instances involving out-of-scope types
        (use -fprint-potential-instances to see them all)
    • In the expression: listToVecTH [1, 2, 3]
      In the untyped splice: $(listToVecTH [1, 2, 3])
  |
5 | v = $(listToVecTH [1,2,3])
  |       ^^^^^^^^^^^^^^^^^^^

Test11.hs:5:20: error:
    • Ambiguous type variable ‘a0’ arising from the literal ‘1’
      prevents the constraint ‘(Num a0)’ from being solved.
      Probable fix: use a type annotation to specify what ‘a0’ should be.
      These potential instances exist:
        instance Num Integer -- Defined in ‘GHC.Num’
        instance Num Double -- Defined in ‘GHC.Float’
        instance Num Float -- Defined in ‘GHC.Float’
        ...plus two others
        ...plus 13 instances involving out-of-scope types
        (use -fprint-potential-instances to see them all)
    • In the expression: 1
      In the first argument of ‘listToVecTH’, namely ‘[1, 2, 3]’
      In the expression: listToVecTH [1, 2, 3]
  |
5 | v = $(listToVecTH [1,2,3])
  |     
```

martijnbastiaan, 2021-04-20:
Agreed :). I've got a patch.

martijnbastiaan, 2021-04-20:
Changed to `enhancement` as `Show` is a lawless class.

commit 4a802c3b (2021-04-20): Make `Show` instance of `Vec` produce valid Haskell

Fixes #1776
commit 9e1f8f64 (2021-04-20): Make `Show` instance of `Vec` produce valid Haskell

Fixes #1776
commit 39281339 (2021-04-20): Make `Show` instance of `Vec` produce valid Haskell

Fixes #1776
commit 58f3b97c (2021-04-20): Make `Show` instance of `Vec` produce valid Haskell

Fixes #1776
closed 2021-04-21 by martijnbastiaan