clash-lang/clash-compiler #1777: `showX` misbehaves if spine of Vector is `XException`

martijnbastiaan, 2021-04-20:
For example:

```haskell
>>> showX (1 :> errorX "abc" :> 3 :> Nil)
"<1,X,3>"
>>> showX (1 :> 2 :> errorX "def")
"<1,*** Exception: X: def
CallStack (from HasCallStack):
  errorX, called at <interactive>:3:29 in interactive:Ghci3
```

alex-mckenna, 2021-04-20:
Isn't that expected? i.e. if you want the spine to be defined shouldn't you use `deepErrorX`?

```haskell
>>> let v1 = 1 :> 2 :> errorX "BOOM" :: Vec 5 Int
>>> let v2 = 1 :> 2 :> deepErrorX "BOOM" :: Vec 5 Int
>>> showX v1
"<1,*** Exception: X: BOOM
CallStack (from HasCallStack):
  errorX, called at <interactive>:11:20 in interactive:Ghci7"
>>> showX v2
"<1,2,X,X,X>"
```

martijnbastiaan, 2021-04-20:
> Isn't that expected? i.e. if you want the spine to be defined shouldn't you use deepErrorX?

The documentation of `ShowX` says:

```haskell
-- | Like the 'Show' class, but values that normally throw an 'XException' are
-- converted to \"X\", instead of error'ing out with an exception.
```

So at least to me, that indicates that it should not error out if there's an `XException` in there. It's highly surprising to me that there's a caveat that it will still error out if the `XException` is part of the spine.

> @martijnbastiaan what would the correct behavior be?

I think the most sensible thing would be to (implicitly) do a shallow `ensureSpine`. E.g.:

```haskell
>>> showX (1 :> 2 :> errorX "def" :: Vec 3 Int)
<1,2,X>
>>> showX (1 :> 2 :> errorX "def" :: Vec 4 Int)
<1,2,X,X>
```

An alternative would be

```haskell
>>> showX (1 :> 2 :> errorX "def" :: Vec 3 Int)
<1,2,X
>>> showX (1 :> 2 :> errorX "def" :: Vec 4 Int)
<1,2,X
```

if we want to signal that the rest of the spine is undefined.

alex-mckenna, 2021-04-20:
> I think the most sensible thing would be to (implicitly) do a shallow `ensureSpine`.

I'm not sure. Say I'm debugging an X in a design and I `showX` a Vector to see what's going on. If I see a defined spine with undefined elements but in reality I don't even have a defined spine that could be confusing. The current behaviour may not be completely desirable, but at least I can tell if the spine is there or not.

`<1,2,X` is clearer regarding what you actually have IMO, but would really mess with #1776 too. Maybe it's a case of picking what's more important?

christiaanb, 2021-04-20:
But what if you don't type annotate? Surely the only consistent option is:
```
<1,2,x
```
or
```
1 :> 2 :> X
```

christiaanb, 2021-04-20:
@alex-mckenna are you saying `ShowX` should produce valid Haskell expression as well? e.g.
```
>>> 1 :> 2 :> errorX "qq"
1 :> 2 :> errorX "qq"
```

alex-mckenna, 2021-04-20:
No, but I'm saying if you want it to then it maybe becomes a bit more difficult. I don't really have a strong opinion on whether `Show` should show valid Haskell or not

martijnbastiaan, 2021-04-20:
> But what if you don't type annotate?

I'd say we'd add a `KnownNat n` constraint to `ShowX (Vec n a)` if we want that notation.

>  `<1,2,X` is clearer regarding what you actually have IMO, but would really mess with #1776 too. Maybe it's a case of picking what's more important?

I agree. 

For `ShowX` validity has never been important as we print a bare `X` anyway. This is different for `Show`: all automatically derived instances are valid Haskell, so it's widely expected to be valid (even for custom instances). Having said that, it would be incredibly helpful if the output of `ShowX` _would_ be copy+pastable.

martijnbastiaan, 2021-04-20:
Also note that the current behavior causes:

```
>>> showX ((1 :> 2 :> errorX "def") :> (3 :> 4 :> 5 :> Nil) :> Nil)
"<<1,*** Exception: X: def
CallStack (from HasCallStack):
  errorX, called at <interactive>:2:19 in interactive:Ghci2
```

martijnbastiaan, 2021-04-20:
We've done a small internal poll, resulting in: 

[image]

For this particular issue I'm just going to implement :apple: . It solves the immediate issue and should be backportable to Clash 1.4. Subsequent pull requests should cover:

* Printing `:>` instead of `<>`, like https://github.com/clash-lang/clash-compiler/pull/1778
* Printing `undefined` instead of `X`

The end result should be valid Haskell for `ShowX`.

commit 945a13c6 (2021-04-20): Account for undefined Vec spines in 'showX'

Fixes #1777
PR #1779 (2021-04-21): Account for undefined Vec spines in 'showX'
Fixes #1777
commit 19768ec2 (2021-04-21): Account for undefined Vec spines in 'showX' (#1779)

Fixes #1777
commit 1820f751 (2021-04-21): Account for undefined Vec spines in 'showX' (#1779)

Fixes #1777

(cherry picked from commit 19768ec2d27b5579e0279382db58c75177abdae8)
closed 2021-04-21 by martijnbastiaan