clash-lang/clash-compiler #1780: Feature Request: Support Library-interoperable Automatic Architecture-specific Optimizations

contrebande-labs, 2021-04-21:
This feature request is the result of an IRC conversation with @christiaanb on the #clash-lang channel yesterday.

This message is mostly a copy-paste of the IRC exchange so we can continue elaborating on the proposed ideas here and not lose track of anything.

It started with me asking for a two-part feature, the first part being a documented and standardized mechanism in Clash for library authors and users to specify the kind of FPGA architecture they are targetting, if any. If no architecture is specified, it is assumed the Clash-generated VHDL is "generic" and does not assume "hard-IPs" or any hardware feature to be present on the FPGA. This way a Clash IP library author could write an abstract core, say, a SPI  master, and possibly include one or many implementations optimized for a specific architecture (for instance, @standardsemiconductor's [iCE40 primitive/hard-IP library](https://github.com/standardsemiconductor/ice40-prim)) along with a generic/soft implementation, synthetizable on any FPGA architecture or for ASICs. And then a user of that particular Clash IP library could somehow specify the desired target architecture (or not, then it's considered "generic") and the optimal implementation would be automatically "selected" from the IP library.

Then, @christiaanb replied (with plain text to markdown conversion and slight editing to remove the IRC specifics) :

> It could be done with [`Hidden`](http://hackage.haskell.org/package/clash-prelude-1.4.1/docs/Clash-Hidden.html), which is something we used at a client. Let me elaborate I'm wring up [some code](https://gist.github.com/christiaanb/fc17f0c524a49e3d304ba6dc4b94bd00). So the idea is that we basically parametrise the `spi` function over the architure of the FPGA but instead of an explicit argument that you have to pass at the term level. We use this `Hidden` constraint to pass it at the type-level. That is, you know that they're there because they show up in the type.

But then, what happens if the number architecture-specific implementations exceeds the capacity of said target architecture? For example, an iCE40UP5K has 2 "hard" SPI cores, so a soft core implementation if the 2 hard cores are already used in the generated VHDL? I understand that this second part is more complicated, that it has to do with how the Verilog serializer can or cannot "count" features.

At that, @christiaanb replied :

> It might be possible using `unsafePerformIO`. Well, we have support for Haskell black boxes, where you can basically have any haskell function to generate the Netlist IR that Clash renders to Verilog/VHDL like https://github.com/clash-lang/clash-compiler/blob/c44ac34d728f339eff41c568b7c2eba239fcb719/clash-cores/src/Clash/Cores/LatticeSemi/ICE40/Blackboxes/IO.hs#L50-L108 which is the blackbox for: https://github.com/clash-lang/clash-compiler/blob/c44ac34d728f339eff41c568b7c2eba239fcb719/clash-cores/src/Clash/Cores/LatticeSemi/ICE40/IO.hs#L158-L244 But that Haskell function has to be of type `Backend s => BlackBoxContext -> State s Doc` which is a pure function and has no way of knowing how often it's invoked but we could create a toplevel `IORef` using unsafePerformIO and a NOINLNE pragma and that IORef than holds a counter to counts how often blackbox haskell function is called s/than/then actually, you'd use a http://hackage.haskell.org/package/clash-lib-1.4.1/docs/Clash-Netlist-BlackBox-Types.html#t:BlackBoxFunction since that operates within the NetlistMonad, which is a wrapper around the IO monad.

We are currently gathering a number of Verilog implementations we would like to translate into Clash in a library using those two features. As an exercise, in the coming weeks, we will make a list at least a couple of those Verilog IPs and look for architecture-specific optimizations that are available for each. We will do this for the Lattice iCE40 UltraPlus iCE40UP5K and for the Xilinx Artix-7 XC7A200T. And there are many more details on how all this could work that we will commit here too. If others are interested in those features, their input would be greatly appreciated.

martijnbastiaan, 2021-04-21:
I'm wondering whether this should be part of the [synthesis domain](https://hackage.haskell.org/package/clash-prelude-1.4.1/docs/Clash-Explicit-Signal.html#t:DomainConfiguration). We could expose:

- vendor: e.g. Xilinx, Intel
- family: e.g. Cyclone IV E
- device: e.g. EP4CE55F23C8, xczu2cg-sfvc784-2-e
- board: e.g. DE0-Nano

At the very least, we could expose this in Clash's manifest files - which should be enough to generate a fully specified Edalize file. For your use case, it could be used to switch between different architectures. 

> But that Haskell function has to be of type Backend s => BlackBoxContext -> State s Doc which is a pure function and has no way of knowing how often it's invoked but we could create a toplevel IORef [...]

I agree that this would be a way to achieve this right now, but I'd rather explicitly support this. We already track whether a primitive has been called before (`NetlistState._seenPrimitives`), we could simply change this into a counter.

contrebande-labs, 2021-04-21:
Putting it in the manifest files seem like a clean way to go, yes. But there is the synthesis part and the "support declaration" part (what architectures are supported by a library) that should both be configurable/declared in the manifest, no?

Is there a way that I can start writing some library/user code now so as to minimize the refactoring once/if those features are implemented? What are the current best practices?

christiaanb, 2021-04-25:
I would go with what I wrote in https://gist.github.com/christiaanb/fc17f0c524a49e3d304ba6dc4b94bd00; it at least allows you to abstract over architectures and the configuration is passed around implicitly.

I think this is something we should both solve as a library (using techniques such as https://gist.github.com/christiaanb/fc17f0c524a49e3d304ba6dc4b94bd00) and at the compiler level. Doing it at the library level allows us to have different (cycle) latencies between architectures and witness those differences during simulation. If we would do it at the compiler-level only, we would always be stuck to the cycle latencies of the worst architecture, e.g. we'd have to introduce latency in the better performing architectures just to be cycle accurate.