clash-lang/clash-compiler #1796: Certain test bench inlining causes Clash to enter infinite loop

DigitalBrains1, 2021-05-01:
I could not reproduce the following issue using "plain" Clash invocation. Rather, they occur when using a Clash starter project with cabal. I've created a starter project with this code at https://github.com/DigitalBrains1/testbenchinline.git for your convenience. Note that it expects current Clash master git repository at `../clash-compiler`. My ghc version is 8.8.4.

This code:
```haskell
module TestBenchInline where

import Clash.Prelude

import Clash.Explicit.Testbench

topEntity
  :: Signal System Bool
  -> Signal System Bool
topEntity = id
{-# NOINLINE topEntity #-}

tb
  :: Signal System Bool
tb = done
  where
    testInput = stimuliGenerator clk rst (singleton True)
    expectOutput = outputVerifier' clk rst (singleton True)
    done = expectOutput $ topEntity testInput
    clk = tbClockGen @System (not <$> done)
    rst = resetGen @System
{-# INLINE tb #-}

testBench :: Signal System Bool
testBench = tb
-- {-# NOINLINE testBench #-}
```
works. Reproduction:
```sh
$ git clone https://github.com/DigitalBrains1/testbenchinline.git
$ cd testbenchinline
$ cabal update
$ cabal run clash -- --vhdl TestBenchInline
```

However, either removing the `INLINE` on `tb` or adding the `NOINLINE` to `testBench` will cause Clash to go into an endless loop:

```
$ cabal run -- --vhdl TestBenchInline
Up to date
Loaded package environment from /home/peter/src/clash/testbenchinline/.ghc.environment.x86_64-linux-8.8.4
GHC: Parsing and optimising modules took: 1.001s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.002s
Clash: Parsing and compiling primitives took 0.163s
GHC+Clash: Loading modules cumulatively took 1.438s
Clash: Compiling TestBenchInline.topEntity
Clash: Normalization took 0.002s
Clash: Netlist generation took 0.000s
Clash: Compiling TestBenchInline.testBench
```
and there it stays looping and eating gradually more and more memory.

This also occurs with Clash 1.4.1.

DigitalBrains1, 2021-05-01:
It also happens with GHC 8.10.4.

alex-mckenna, 2021-05-06:
It also happens with GHC 9.0.1. Running with `-fclash-debug DebugApplied` tells us that this is the problem:

```
inlineSmall {0}                                                                                                                               
Changes when applying rewrite to:                                                                                                             
testBench_done[GlobalId]                                                                                                                      
Result:                                                                                                                                       
$soutputVerifier[GlobalId]                                                                                                                    
  testBench_clk[GlobalId]                                                                                                                     
  testBench_rst[GlobalId]                                                                                                                     
  testBench2[GlobalId]                                                                                                                        
  testBench1[GlobalId]                                                                                                                        
                                                                                                                                              
inlineWorkFree {1}                                                                                                                            
Changes when applying rewrite to:                                                                                                             
testBench_clk[GlobalId]                                                                                                                       
Result:                                                                                                                                       
tbClockGen @"System"                                                                                                                          
  $fKnownDomain"System"[GlobalId]                                                                                                             
  testBench5[GlobalId]                                                                                                                        
                                                                                                                                              
removeUnusedExpr {2}                                                                                                                          
Changes when applying rewrite to:                                                                                                             
tbClockGen @"System"                                                                                                                          
  $fKnownDomain"System"[GlobalId]                                                                                                             
  testBench5[GlobalId]                                                                                                                        
Result:                                                                                                                                       
tbClockGen @"System"                                                                                                                          
  (removedArg @(KnownDomain "System"))                                                                                                        
  testBench5[GlobalId]                                                                                                                        
                                                                                                                                              
Clash.Rewrite.Util(231): Error when applying rewrite inlineSmall to:                                                                          
testBench5[GlobalId]                                                                                                                          
Result:                                                                                                                                       
not testBench_done[GlobalId]                                                                                                                  
Changes type from:                                                                                                                            
Signal "System" Bool                                                                                                                          
to:                                                                                                                                           
Bool                                                                                                                                          
inlineSmall {3}                                                                                                                               
Changes when applying rewrite to:                                                                                                             
testBench5[GlobalId]                                                                                                                          
Result:                                                                                                                                       
not testBench_done[GlobalId] 
```

(repeat ad infinitum)

alex-mckenna, 2021-05-06:
Intuitively I suspect this is `inlineWorkFree`'s fault. Or more specifically, the fault of a change I made to `isWorkFree` a while ago for PE. I'll look into it later

alex-mckenna, 2021-05-06:
Slightly smaller reproducer:

```haskell
module TestBenchInline where

import Clash.Prelude

import Clash.Explicit.Testbench

tb :: Signal System Bool
tb = done
  where
    done = outputVerifier' clk rst (singleton True) (pure True)
    clk  = tbClockGen @System (not <$> done)
    rst  = resetGen @System

testBench :: Signal System Bool
testBench = tb
```

christiaanb, 2021-05-06:
Even smaller:
```haskell
import Clash.Explicit.Prelude
import Clash.Explicit.Testbench

tb
  :: Signal System Bool
tb = done
  where
    done = register clk rst enableGen False $ id (pure True)
    clk = tbClockGen @System (not <$> done)
    rst = resetGen @System

topEntity :: Signal System Bool
topEntity = tb
```

alex-mckenna, 2021-05-06:
So this was a bit of a search, good thing @christiaanb was here. Turns out this issue stems from the fact that Clash internally turns recursive groups of function such as

```haskell
f x = ... g ...
g x = ... h ...
h x = ... f ...
```

into global binders which don't depend on other global binders, i.e. something like

```haskell
f x = let g x = ... h ...; h x = ... f ...; in ... g ...
g x = let f x = ... g ...; h x = ... f ...; in ... h ...
h x = let f x = ... g ...; g x = ... h ...; in ... f ...
```

We do this in the [makeRecursiveGroups](https://github.com/clash-lang/clash-compiler/blob/8e461a910f2f37c900705a0847a9b533bce4d2ea/clash-ghc/src-ghc/Clash/GHC/LoadModules.hs#L617) function by taking our set of bindings from a module and making graph where the nodes are global bindings and the edges are free variables in those bindings.

This function uses a GHC API function, `exprFreeIds`, which according to it's documentation / implementation only gives free variables which are also locally scoped _meaning it makes no edges for our recursive global binders_.

```haskell
-- | Find all locally-defined free Ids in an expression
exprFreeIds :: CoreExpr -> IdSet        -- Find all locally-defined free Ids
exprFreeIds = exprSomeFreeVars isLocalId
```

Fortunately, the solution is simple: change `CoreFVS.exprFreeIds` to `CoreFVS.exprSomeFreeVars Var.isId`, i.e. change the predicate that determines whether to keep a free variable to be whether it is an `Id` at all, global or local.