clash-lang/clash-compiler #1799: unsafeCoerce-ing constraints away doesn't work

Jhana1, 2021-05-03:
```
import           Clash.Prelude
import           Unsafe.Coerce

at' :: ((n + 1) <= m) => SNat n -> Vec m a -> a
at' = unsafeCoerce at

f :: Vec 2 Bool -> Bool
f a = at' d0 a

topEntity :: Signal System (Vec 2 Bool) -> Signal System Bool
topEntity = fmap f
```

Fails to compile with
```
[1 of 1] Compiling Main             ( build/breaks.hs, build/breaks.o )
Ok, one module loaded.
GHC: Parsing and optimising modules took: 0.416s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.000s
Clash: Parsing and compiling primitives took 0.073s
GHC+Clash: Loading modules cumulatively took 0.589s
Clash: Compiling Main.topEntity
Clash: Normalisation took 0.007s
[1 of 1] Compiling Main             ( build/breaks.hs, interpreted )
Ok, one module loaded.
*** Exception: Clash.Netlist(279): Clash.Core.Util(322): Cannot reduce to an integer:
GHC.TypeNats.+[3674937295934325068] (GHC.Types.Any[3674937295934325098] GHC.Types.Nat[3674937295934325064]) 1
```

Using `at` without `unsafeCoerce` works fine
```
import           Clash.Prelude

f :: Vec 2 Bool -> Bool
f a = at d0 a

topEntity :: Signal System (Vec 2 Bool) -> Signal System Bool
topEntity = fmap f
```
```
[1 of 1] Compiling Main             ( build/breaks.hs, build/works.o )
Ok, one module loaded.
GHC: Parsing and optimising modules took: 0.295s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.000s
Clash: Parsing and compiling primitives took 0.071s
GHC+Clash: Loading modules cumulatively took 0.473s
Clash: Compiling Main.topEntity
Clash: Normalisation took 0.008s
Clash: Netlist generation took 0.000s
Clash: Total compilation took 0.483s
```
version
```
$ stack exec clashi -- --version                       
Clash, version 1.2.4 (using clash-lib, version: 1.2.4)
$ stack exec ghc -- --version
The Glorious Glasgow Haskell Compilation System, version 8.6.5
```
Fair enough if this isn't allowed.

christiaanb, 2021-05-03:
So `at` is defined as:
```
at :: forall m n a . SNat m -> Vec (m + (n + 1)) a -> a
```
Normally, those type arguments, `forall m n a`, are implicitly applied through type-inference whenever you apply a function like `at` to regular type arguments.

The problem is that the combination of `unsafeCoerce` and type inference causes GHC to type-instantiate `at` with `Any`. So when we run with `-ddump-ds` we see that
```haskell
at' :: forall m n a . ((n + 1) <= m) => SNat n -> Vec m a -> a
at' s v = at s (unsafeCoerce v)
```
is desugared to:
```
at'
  = \ (@ (m_aFy6 :: Nat))
      (@ (n_aFy7 :: Nat))
      (@ a_aFy8)
      _ [Occ=Dead]
      (s_aFxl :: SNat n_aFy7)
      (v_aFxm :: Vec m_aFy6 a_aFy8) ->
      break<0>(s_aFxl,v_aFxm)
      at
        @ n_aFy7
        @ GHC.Types.Any
        @ a_aFy8
        s_aFxl
        (unsafeCoerce
           @ (Vec m_aFy6 a_aFy8)
           @ (Vec (n_aFy7 + (GHC.Types.Any + 1)) a_aFy8)
           v_aFxm)
```
which basically corresponds to the following haskell:
```haskell
at' :: forall m n a . ((n + 1) <= m) => SNat n -> Vec m a -> a
at' s v = at @n @Any @a s (unsafeCoerce @(Vec m a) @(Vec (n + (Any + 1)) a) v)
```
The `Any` type is a type that unifies with any other type: something expects an `Int`, you can give it an `Any`, and GHC will be alright with that. GHC introduced `Any` because the `unsafeCoerce` wasn't linked properly to the type variables `m`, `n`, and `a`.

So you have to be explicit from what type to what other type we want to coerce unsafely:
```haskell
at' :: forall m n a . ((n + 1) <= m) => SNat n -> Vec m a -> a
at' = unsafeCoerce (at @n @m @a)
```
and we can see from the `-ddump-ds` output that GHC no longer instantiates anything at `Any`:
```
at'
  = \ (@ (m_aON4 :: Nat))
      (@ (n_aON5 :: Nat))
      (@ a_aON6)
      _ [Occ=Dead] ->
      break<0>()
      unsafeCoerce
        @ (SNat n_aON5 -> Vec (n_aON5 + (m_aON4 + 1)) a_aON6 -> a_aON6)
        @ (SNat n_aON5 -> Vec m_aON4 a_aON6 -> a_aON6)
        (at @ n_aON5 @ m_aON4 @ a_aON6)
```

Jhana1, 2021-05-03:
Nice. Thanks heaps.

closed 2021-06-02 by alex-mckenna