clash-lang/clash-compiler #1810: VHDL shift_right shift amount must be a natural:

christiaanb, 2021-05-16:
As can be seen on https://groups.google.com/g/clash-language/c/CQTJhkDrb1Q/m/6YcY0Ur-AAAJ GHDL raises an error when the shift amount to the `shift_right` call is a negative number. As we can see in:

https://github.com/ghdl/ghdl/blob/c2b3f93dfa40ee92de53b54e38e1819cfc33cb04/libraries/ieee/numeric_std.vhdl#L556-L560
```vhdl
  -- Id: S.2
  function SHIFT_RIGHT (ARG: UNSIGNED; COUNT: NATURAL) return UNSIGNED;
  -- Result subtype: UNSIGNED(ARG'LENGTH-1 downto 0)
  -- Result: Performs a shift-right on an UNSIGNED vector COUNT times.
  --         The vacated positions are filled with '0'.
  --         The COUNT rightmost elements are lost.
```

and

https://github.com/ghdl/ghdl/blob/c2b3f93dfa40ee92de53b54e38e1819cfc33cb04/libraries/ieee/numeric_std.vhdl#L570-L574
```vhdl
  function SHIFT_RIGHT (ARG: SIGNED; COUNT: NATURAL) return SIGNED;
  -- Result subtype: SIGNED(ARG'LENGTH-1 downto 0)
  -- Result: Performs a shift-right on a SIGNED vector COUNT times.
  --         The vacated positions are filled with the leftmost
  --         element, ARG'LEFT. The COUNT rightmost elements are lost.
```
the shift count must be a natural number.

However, the default implementation for the `shift` method of the `Bits` class is: http://hackage.haskell.org/package/base-4.14.1.0/docs/src/Data.Bits.html#shift
```haskell
    x `shift`   i | i<0       = x `shiftR` (-i)
                  | i>0       = x `shiftL` i
                  | otherwise = x|
```
which gets translated to concurrent assignments by clash, i.e. that `shift` method will get translated to something like the following VHDL:
```vhdl
alt1 <= shift_right(x,-i);
alt2 <= shift_left(x,i);
res <= alt1 when (i < 0) else
       alt2 when (i > 0) else
       x;
```

The solution is to create primitives for the `shift` instance methods which we can control: `Signed`, `Unsigned`, `Bit` and `BitVector`; so that we can generate the following VHDL instead:

```vhdl
process (x,i) begin
  if (i < 0) then
    res <= shift_right(x,-i);
  elsif (i >0) then
    res <= shift_left(x,i)
  else
    res <= x;
  end if;
end;
```

ensuring that the `shift_right` is not evaluated concurrently to the `shift_left` when `i > 0`.

christiaanb, 2021-05-16:
Another option is to translate `shiftR` to:
```vhdl
res <= shift_right(x,i)
-- pragma translate_off
          when (i >= 0) else shift_left(x,-i)
-- pragma translate_on
          ;
```
so that we only get a `shift_right` for synthesis, but no errors during simulation.

alex-mckenna, 2021-05-17:
Same applies to rotate:

```vhdl
function ROTATE_LEFT (ARG: SIGNED; COUNT: NATURAL) return SIGNED;
```

```haskell
x `rotate`  i | i<0       = x `rotateR` (-i)
              | i>0       = x `rotateL` i
              | otherwise = x
```

PR #1825 (2021-05-21): Don't render negative shift/rotate in VHDL
This PR stops clash rendering calls to the shift and rotate functions in numeric_std with negative amounts to shift/rotate by. This should prevent errors in simulation from unexpected negative shift amounts.

Fixes #1810 

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files

[comment]: # (Depending on the change, some of these points may not be necessary. If so, leave the boxes unchecked so it is easier for a reviewer to see what has been omitted.)
closed 2021-05-21 by alex-mckenna