clash-lang/clash-compiler #1813: Clash hangs when compiling testBench

rowanG077, 2021-05-18:
I'm not sure if this is supposed to be possible. But what I want is to be able to run a circuit under GHDL to be able to benchmark it's performance. To test this out I made a very simple circuit. But when I try to compile it clash hangs. Is it required a testBench returns a signal of bool, I could not find this in the documentation? In my case I'm not interested checking any output. I just want to run the circuit.

```
import Clash.Explicit.Prelude
import Clash.Explicit.Testbench

topEntity
  :: Signal System (Unsigned 32)
  -> Signal System (Unsigned 32)
topEntity a = a + 1

testBench :: Signal System (Unsigned 32)
testBench = topEntity a
  where
    clk = tbClockGen (fromEnable en)
    rst = systemResetGen
    en = tbEnableGen
    a = register clk rst en 0 (a + 1)
```

rowanG077, 2021-05-18:
It seems to be looping infinitely over some transformations, this is the output with `-fclash-debug DebugName`:

```
...
Irreducible constant as case subject: Clash.Explicit.Testbench.tbEnableGen @"System"
Can be reduced to: Clash.Explicit.Testbench.tbEnableGen @"System"
inlineSmall {25797}
removeUnusedExpr {25798}
inlineWorkFree {25799}
inlineWorkFree {25800}
inlineWorkFree {25801}
inlineSmall {25802}
inlineWorkFree {25803}
inlineWorkFree {25804}
inlineWorkFree {25805}
inlineSmall {25806}
inlineWorkFree {25807}
inlineWorkFree {25808}
inlineSmall {25809}
inlineSmall {25810}
inlineSmall {25811}
removeUnusedExpr {25812}
inlineSmall {25813}
inlineSmall {25814}
removeUnusedExpr {25815}
removeUnusedExpr {25816}
inlineSmall {25817}
Irreducible constant as case subject: Clash.Explicit.Testbench.tbEnableGen @"System"
Can be reduced to: Clash.Explicit.Testbench.tbEnableGen @"System"
....
```

rowanG077, 2021-05-18:
This has in fact been fixed 7 days ago :dancers: in https://github.com/clash-lang/clash-compiler/commit/94de6265f3f95f32409bd48ae39595e9a79e0bc1. I should have tested master before making this issue

closed 2021-05-18 by rowanG077

christiaanb, 2021-05-18:
Yeah, this is a pretty serious bug. We should make a 1.4.2 release really soon.