clash-lang/clash-compiler #1828: Support for WebAssembly via Asterius

contrebande-labs, 2021-05-27:
Hi!

My use case is described in #1780, but I will give more details. I'm looking to build a node-based programming environment entirely within the browser, most notably with [WebUSB](https://web.dev/usb/), like [WebFPGA is doing](https://beta.webfpga.io/). I would like the nodes to be softcore building blocks written in Clash. The trees/graphs formed by those nodes would "compile" to Verilog and then go through the open source FPGA toolchain to generate a bitmap and then be uploaded onto the FPGA via WebUSB.

For this to happen, I would need to be able to "compile" my Clash "node/core" library to WebAssembly to be able to call it at compile time. I'm submitting this here to know if other people have pointers into making this possible. Thanks!

christiaanb, 2021-05-27:
That WebFPGA looks really cool. So to understand it a bit better: what you'd like is something like what WebFPGA is currently offerening, but instead of writing Verilog, you write Haskell? And then you'd use Clash to gen Verilog and use the existing WASM-based Verilog-to-Bitstream tools to program the FPGA?

christiaanb, 2021-05-27:
I guess if the WebAssembly route using Asterius doesn't work, we could also create some REST interface to the Clash compiler, where we could run on instance on one of our servers; of course leaving open the option to run your own instance.

basile-henry, 2021-05-27:
> create some REST interface to the Clash compiler

Sandboxing this is going to be tricky. Someone will likely try to mine bitcoins with TemplateHaskell's `runIO`. Maybe such a service would disallow TemplateHaskell? You will likely want compilation time and resource limits and somehow prevent what kind of IO can be done if you allow TemplateHaskell while compiling. :sweat_smile:

christiaanb, 2021-05-27:
I mean...  we would finally have a use for [`Clash.Prelude.Safe`](http://hackage.haskell.org/package/clash-prelude-1.4.2/docs/Clash-Prelude-Safe.html)

contrebande-labs, 2021-05-27:
Hi,

A first step could be to write Clash Haskell in a web editor, if that's easier (I don't think it is, though). But what I wanted to do is to have Clash code compiled into WebAssembly to have a graph-based visual programming environment to put together the Clash modules/nodes/cores, like with [Icestudio](https://icestudio.io) but higher-level (with nodes like Spi, Uart, I2C, CAN FD, etc.). So the library of nodes would be written in Clash and then compiled to WebAssembly. The WebAssembly would be executed in the browser to generate the Verilog from the graph definition and then the Verilog would be passed on to the rest of the open source FPGA toolchain (Yosys, etc.) to generate the bitmap. And finally the bitmap would be sent to the (FTDI) programmer via WebUSB. Everything would run in the browser, no external services (REST or otherwise), no server at all. It could all be packaged nicely into an ElectronJS application. So in short, would it be possible to fork or patch Asterius and to have it behave as a Clash compiler (just like you did for GHC) so that Clash code could be compiled as WebAssembly libraries that can be called from within the browser to generate Verilog?

christiaanb, 2021-05-27:
Mmm... what you describe would work if Clash was a so-called embedded domain specific language, like [kansas-lava](https://ku-fpg.github.io/software/kansas-lava/) or [blarney](https://github.com/blarney-lang/blarney). But Clash is a more "traditional" compiler in that it works like:
```
Haskell source-code (.hs file)

          |
          |
  (GHC API: Lexer and parser)
          |
          V

  Haskell AST (HsSyn)

          |
          |
  (GHC API: Desugar)
          |
          V

   GHC Core AST (CoreSyn)

          |
          |
   (clash-ghc: Core2Core)
          |
          V

    Clash Core AST (Term)

          |
          |
   (clash-lib: Normalize/Flatten/Monomorphize/Defunctionalize/etc.)
          |
          V

    Clash Core AST (Term)

          |
          |
  (clash-lib: Core2Netlist)
          |
          V

    Netlist AST (Component)

          |
          |
  (clash-lib: genHDL (Verilog/VHDL/SystemVerilog))
          |
          |
          V

        Text

          |
          |
   (text: Data.Text.IO.writeFile)
          |
          V

Verilog source file (.v) 
```
So the thing we must compile to WASM is basically all the compiler code... not the Clash  circuit description.

contrebande-labs, 2021-05-27:
Ah I see. Oh well. Yes. That is definitely not realistic. Thanks for taking the time to explain it to me.

closed 2021-06-02 by alex-mckenna