clash-lang/clash-compiler #1861: cabal build error solveSimpleWanteds: too many iterations

dopamane, 2021-06-20:
I started seeing this error a day ago on a few different projects when compiling clash-prelude-1.4.2 with multiple GHC versions and cabal versions. For example when running `cabal build` with [ice40-prim](https://github.com/standardsemiconductor/ice40-prim):

```
...
src/Clash/Sized/Internal/Index.hs:1:1: error:
    solveSimpleWanteds: too many iterations (limit = 4)
      Set limit with -fconstraint-solver-iterations=n; n=0 for no limit
      Simples = {[WD] $dKnownNat_a3IvT {0}:: KnownNat n0 (CNonCanonical),
                 [WD] $dKnownNat_a3IvV {0}:: KnownNat n (CNonCanonical),
                 [WD] $dKnownNat_a3IvY {0}:: KnownNat m0 (CNonCanonical),
                 [WD] $dIntegral_a3Iw1 {0}:: Integral a0 (CNonCanonical),
                 [WD] hole{co_a3Iw2} {0}:: Index (((n - 1) * (n - 1)) + 1)
                                           ~ a0 (CNonCanonical),
                 [WD] hole{co_a3Iw3} {0}:: Index m0 ~ r0 (CNonCanonical)}
      WC = WC {wc_simple =
                 [WD] $dKnownNat_a3INU {0}:: KnownNat 2 (CNonCanonical)
                 [WD] $dKnownNat_a3INV {0}:: KnownNat 2 (CNonCanonical)}
  |
1 | {-|
  | ^
cabal: Failed to build clash-prelude-1.4.2 (which is required by
ice40-prim-0.3.1.0). See the build log above for details.
```

christiaanb, 2021-06-20:
Yeah, we noticed this on HEAD of `master`. I didn't fully appreciate this would affect released versions as well, so I'll make a fix now.

christiaanb, 2021-06-20:
Alright, I'm seeing multiple fallouts (not just the build error you are seeing) in https://github.com/clash-lang/clash-compiler/pull/1862 so I've deprecated the 0.7.5 release of `ghc-typelits-natnormalise`. When you do a `cabal update`, the build plan should pick the older 0.7.4 releaase.

clash-lang-builder, 2021-06-20:
@dopamane I've released https://hackage.haskell.org/package/ghc-typelits-natnormalise-0.7.6 which should no longer give any of those errors. So when you do a `cabal` update, a build plan should pick up version 0.7.6

dopamane, 2021-06-20:
Ok, resolved. Thank you!

commit ea3f3d06 (2021-06-20): Fix fallout from ghc-typelits-natnormalise-0.7.5 release

Fixes #1861
commit a43dd489 (2021-06-20): Fix fallout from ghc-typelits-natnormalise-0.7.5 release

Fixes #1861
commit e5bd83b8 (2021-06-20): Fix fallout from ghc-typelits-natnormalise-0.7.5 release

Fixes #1861
commit 1932e549 (2021-06-20): Fix fallout from ghc-typelits-natnormalise-0.7.5 release

Fixes #1861
closed 2021-06-20 by dopamane