clash-lang/clash-compiler #1870: AutoReg Convenience Functions

Jhana1, 2021-06-29:
Some additional AutoReg convenience functions would be really nice, replicating the `delay`, `delayN`, `delayI` interface would be great. I've tried to do the same, but the following blows out compile times (so I imagine some internal special casing must happen on these functions to prevent that)
```
 dAutoRegN
   :: (HiddenClockResetEnable App, AutoReg a, Default a, KnownNat delay)
   => SNat n
   -> DSignal App delay a
   -> DSignal App (delay + n) a
 dAutoRegN s = unsafeFromSignal . go (snatToNum s) . toSignal
   where
     go 0 = id
     go n = autoReg def . go (n - 1)
```

alex-mckenna, 2021-06-29:
We do have [a branch](https://github.com/clash-lang/clash-compiler/tree/auto-delay) from a while ago that @christiaanb was working on, but I'm not sure what state it's in. I'm not that familiar with the `AutoReg` stuff though, maybe it needs more adaptation to work with `DSignals`

christiaanb, 2021-06-29:
@Jhana1 Interesting, I post a rewrite of what you wanted that shouldn't blow up compile times, after I've had my breakfast.

christiaanb, 2021-06-29:
@Jhana1 The following should compile relatively quickly:
```haskell
module DAutoReg where

import Clash.Prelude

dAutoRegN ::
  forall dom a delay n .
  (HiddenClockResetEnable dom, AutoReg a, Default a, KnownNat delay) =>
  SNat n ->
  DSignal dom delay a ->
  DSignal dom (delay + n) a
dAutoRegN s@SNat = unsafeFromSignal . go (toUNat s) . toSignal
  where
    go :: UNat n -> Signal dom a -> Signal dom a
    go UZero     inp = inp
    go (USucc _) inp =
      let shIn  = init (inp `Cons` shOut)
          shOut = map (autoReg def) (lazyV @n shIn)
       in last shOut

topEntity xs = dAutoRegN @System @(Vec 10 (Maybe Int)) @0 (SNat @200) xs
```
So the 200 stage deep pipeline of type `Vec 10 (Maybe Int)` for me gives:
```
$ time cabal run clash -- --vhdl DAutoReg.hs
Up to date
Loaded package environment from /home/christiaan/clash-compiler/.ghc.environment.x86_64-linux-8.10.4
GHC: Parsing and optimising modules took: 1.413s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.001s
Clash: Parsing and compiling primitives took 0.142s
GHC+Clash: Loading modules cumulatively took 2.793s
Clash: Compiling DAutoReg.topEntity
Clash: Normalization took 0.291s
Clash: Netlist generation took 0.024s
Clash: Total compilation took 3.130s

real    0m3.743s
user    0m3.378s
sys     0m0.176s
```
This is on HEAD of `master`, but I don't expect much difference for the 1.4.2 release.

christiaanb, 2021-06-29:
`dAugoRegN` can actually be simplified to:
```haskell
dAutoRegN ::
  forall dom a delay n .
  (HiddenClockResetEnable dom, AutoReg a, Default a, KnownNat delay) =>
  SNat n ->
  DSignal dom delay a ->
  DSignal dom (delay + n) a
dAutoRegN SNat = unsafeFromSignal . go . toSignal
  where
    go :: Signal dom a -> Signal dom a
    go inp =
      let shIn  = init (inp :> shOut)
          shOut = map (autoReg def) (lazyV @n shIn)
       in last (inp :> shOut)
```

christiaanb, 2021-06-29:
The reason this doesn't blow up is because it allows the Clash compiler to take hard-coded "short-cuts" where it doesn't have to unroll the definitions for `init`, `map`, `lazyV` and `last`, but can use their "blackbox" implementations instead:
* `init`: https://github.com/clash-lang/clash-compiler/blob/1c81d8f7e4c208559de2c31cafb3114fc9aefe89/clash-lib/prims/vhdl/Clash_Sized_Vector.primitives#L25-L32
* `map`: https://github.com/clash-lang/clash-compiler/blob/1c81d8f7e4c208559de2c31cafb3114fc9aefe89/clash-lib/prims/vhdl/Clash_Sized_Vector.primitives#L102-L128
* `last`: https://github.com/clash-lang/clash-compiler/blob/1c81d8f7e4c208559de2c31cafb3114fc9aefe89/clash-lib/prims/vhdl/Clash_Sized_Vector.primitives#L17-L24
* `lazyV`: https://github.com/clash-lang/clash-compiler/blob/1c81d8f7e4c208559de2c31cafb3114fc9aefe89/clash-lib/prims/common/Clash_Sized_Vector.primitives#L2-L10

Another thing to note: [`lazyV`](https://hackage.haskell.org/package/clash-prelude-1.4.2/docs/Clash-Sized-Vector.html#v:lazyV) is there to ensure we have a "spine" of `Cons / (:>)` constructors to `map` over.

Jhana1, 2021-06-29:
Thanks, I'm trying it now. Much appreciated!