clash-lang/clash-compiler #1872: Increase default tuple size (measurements)

martijnbastiaan, 2021-07-03:
In Clash 1.4 we disabled the `-flarge-tuples` flag by default, as it caused problems on memory constrained machines (typically laptops). By default, Clash now generates instances up to 12-tuples which some users have noticed as a regression. We should probably pick a better default.

Related discussion:

* https://github.com/clash-lang/clash-compiler/issues/1770

# Graph

[image]

# Table
<details>
<summary>Click to expand</summary>

| Tuple size | Time (s) | Time v Baseline | Memory (Mbytes) |
|------------|----------|-----------------|-----------------|
| 12         | 117      | 1.00            | 2586            |
| 13         | 116      | 1.00            | 2375            |
| 14         | 119      | 1.02            | 2488            |
| 15         | 117      | 1.01            | 2481            |
| 16         | 121      | 1.04            | 2467            |
| 17         | 119      | 1.02            | 2520            |
| 18         | 121      | 1.04            | 2497            |
| 19         | 122      | 1.05            | 2491            |
| 20         | 123      | 1.05            | 2359            |
| 21         | 125      | 1.07            | 2377            |
| 22         | 127      | 1.09            | 2529            |
| 23         | 127      | 1.09            | 2483            |
| 24         | 132      | 1.13            | 2574            |
| 25         | 129      | 1.11            | 2448            |
| 26         | 133      | 1.14            | 2354            |
| 27         | 133      | 1.14            | 2475            |
| 28         | 134      | 1.15            | 2317            |
| 29         | 138      | 1.18            | 2543            |
| 30         | 138      | 1.19            | 2387            |
| 31         | 141      | 1.21            | 2403            |
| 32         | 143      | 1.23            | 2538            |
| 33         | 147      | 1.26            | 2528            |
| 34         | 148      | 1.27            | 2319            |
| 35         | 151      | 1.30            | 2426            |
| 36         | 155      | 1.33            | 2511            |
| 37         | 158      | 1.36            | 2465            |
| 38         | 160      | 1.37            | 2473            |
| 39         | 164      | 1.41            | 2488            |
| 40         | 168      | 1.44            | 2530            |
| 41         | 171      | 1.46            | 2622            |
| 42         | 175      | 1.50            | 2691            |
| 43         | 179      | 1.54            | 2802            |
| 44         | 182      | 1.56            | 2752            |
| 45         | 188      | 1.61            | 2856            |
| 46         | 193      | 1.65            | 2956            |
| 47         | 197      | 1.69            | 2948            |
| 48         | 202      | 1.73            | 3231            |
| 49         | 208      | 1.79            | 3284            |
| 50         | 213      | 1.83            | 3314            |
| 51         | 218      | 1.87            | 3451            |
| 52         | 224      | 1.92            | 3539            |
| 53         | 229      | 1.96            | 3481            |
| 54         | 238      | 2.04            | 3887            |
| 55         | 243      | 2.09            | 3807            |
| 56         | 250      | 2.15            | 4122            |
| 57         | 257      | 2.20            | 3853            |
| 58         | 265      | 2.28            | 4287            |
| 59         | 270      | 2.32            | 4070            |
| 60         | 279      | 2.39            | 4218            |
</details>

leonschoorl, 2021-07-09:
Nice graph

But what have you benchmarked here?
Is it building clash(-prelude)?
Or compiling a design with clash?

martijnbastiaan, 2021-07-09:
Just building clash-prelude

martijnbastiaan, 2021-07-09:
@leonschoorl  To expand on this a bit (which I should have done in the first post :-)):

* We encountered problems with people unable to compile Clash (specifically clash-prelude, IIRC)
  * Windows's GHC would crash due to a Template Haskell bug not being able to allocate more than 4 GB of memory
  * Memory constrained devices (laptops) couldn't allocate enough memory
* I vaguely remember us testing whether tuple sizes affected `clash-lib` / `clash-ghc` compilation time and Clash normalization time, but it didn't. It might be worth rechecking this.