clash-lang/clash-compiler #1881: GHC inlining Boolean combinational logic.

exclusive-and, 2021-07-26:
It looks like GHC automatically inlines combinational logic. The inlining leads to Clash compiling what should be logic gates as muxes.

What follows is an example of code with both the current and desired behaviours:
```
import Data.Bool

import qualified GHC.Magic

-- Custom Boolean operator suggested by Christiaan as a workaround.
(&!) = GHC.Magic.noinline (&&)

-- Sample circuit will return the normally generated result, alongside the fixed/noinlined version

topEntity :: Bool -> Bool -> (Bool, Bool)
topEntity a b = (a && b, a &! b)
```
and the resulting circuit in Verilog:
```
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.5.0. DO NOT MODIFY.
*/
`timescale 100fs/100fs
module topEntity
    ( // Inputs
      input  a
    , input  b

      // Outputs
    , output wire  result_0
    , output wire  result_1
    );
  wire  c$app_arg;
  wire [1:0] result;

  assign c$app_arg = a ? b : 1'b0;

  assign result = {c$app_arg,   a & b};

  assign result_0 = result[1:1];

  assign result_1 = result[0:0];

endmodule
```

leonschoorl, 2021-07-28:
Do you know if this actually matters in practice?
Both should be semantically equivalent.

I tried synthesizing your example in Vivado 2019.1.
Both synthesize to the same LUT2. And because the LUT's where the same Vivado even combined them in to one:
[image]

I do think the `a & b` is much easier to read.

exclusive-and, 2021-07-28:
Synthesizing Verilog in Quartus gives me a bunch of mux logic that's difficult to read and uses more transistors than I'd like. It's not exactly a huge or insurmountable problem; Christiaan's workaround of re-exporting a magically noinlined operator does work. But it would be nice for my code to compile as intended.

alex-mckenna, 2021-08-12:
The use of `noinline` is now part of the exported definition from `clash-prelude`. So now

```haskell
topEntity :: Bool -> Bool -> (Bool, Bool, Bool)
topEntity a b = (a && b, a || b, not a)
```

renders like

```verilog
module topEntity
    ( // Inputs
      input  a
    , input  b

      // Outputs
    , output wire  result_0
    , output wire  result_1
    , output wire  result_2
    );
  wire [2:0] result;

  assign result = {a & b, a | b, ~ a};

  assign result_0 = result[2:2];
  assign result_1 = result[1:1];
  assign result_2 = result[0:0];
endmodule
```

instead of

```verilog
module topEntity
    ( // Inputs
      input  a
    , input  b

      // Outputs
    , output wire  result_0
    , output wire  result_1
    , output wire  result_2
    );
  wire  c$app_arg;
  wire  c$app_arg_0;
  wire  c$app_arg_1;
  wire [2:0] result;

  assign c$app_arg = a ? 1'b0 : 1'b1;
  assign c$app_arg_0 = a ? 1'b1 : b;
  assign c$app_arg_1 = a ? b : 1'b0;
  assign result = {c$app_arg_1, c$app_arg_0, c$app_arg};

  assign result_0 = result[2:2];
  assign result_1 = result[1:1];
  assign result_2 = result[0:0];
endmodule
```

DigitalBrains1, 2021-08-13:
> Synthesizing Verilog in Quartus gives me a bunch of mux logic that's difficult to read and uses more transistors than I'd like.

Of course the mux in the RTL view is user unfriendly, and it would be better if Clash would just emit an AND in HDL. But if you're targetting an FPGA with Quartus, I can put your mind at ease that it does not increase resource usage. I tried it in Quartus with plain Verilog, and I got the following results:

[image]
Not so nice RTL.

[image]
Luckily, in the Post-Map Viewer we see that Quartus also recognises the two are equivalent, and merges them.

[image]
And when we dive into the Resource Property Editor, we see it agrees this is an AND. There is something funny going on, as the Post-Map Viewer indicated usage of `DATAA` and `DATAB`, but here it's `DATAC` and `DATAD`. I'll put that down to Quartus being weird.

commit 3667491d (2021-08-13): Preserve boolean combinatorial logic in HDL

When rendering HDL, clash now better preserves boolean functions
from GHC which may be targeted by the simplifier. This helps ensure
they are rendered in HDL as the expected operator instead of being
rendered as a multiplexer.

Fixes #1881.
commit 84ac1ac2 (2021-08-13): Preserve boolean combinatorial logic in HDL

When rendering HDL, clash now better preserves boolean functions
from GHC which may be targeted by the simplifier. This helps ensure
they are rendered in HDL as the expected operator instead of being
rendered as a multiplexer.

Fixes #1881.
PR #1900 (2021-08-16): Preserve boolean combinatorial logic in HDL
When rendering HDL, clash now better preserves boolean functions from GHC which may be targeted by the simplifier. This helps ensure they are rendered in HDL as the expected operator instead of being rendered as a multiplexer.

Fixes #1881.

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files

[comment]: # (Depending on the change, some of these points may not be necessary. If so, leave the boxes unchecked so it is easier for a reviewer to see what has been omitted.)
commit 1afe0084 (2021-08-16): Preserve boolean combinatorial logic in HDL

When rendering HDL, clash now better preserves boolean functions
from GHC which may be targeted by the simplifier. This helps ensure
they are rendered in HDL as the expected operator instead of being
rendered as a multiplexer.

Fixes #1881.
closed 2021-08-16 by alex-mckenna