clash-lang/clash-compiler #1901: `DomainConfigurationPeriod` type family apparently breaks inequality solver

DigitalBrains1, 2021-08-12:
This code does not typecheck:
```haskell
proxyInEq :: ( KnownDomain dom
             , KnownNat (DomainPeriod dom)
             , 1 <= DomainPeriod dom)
          => Proxy (dom :: Domain) -> ()
proxyInEq _ = ()

proxyInEq' :: ( KnownDomain dom
              , KnownNat (DomainPeriod dom)
              , 2 <= DomainPeriod dom)
           => Proxy (dom :: Domain) -> ()
proxyInEq' dom = proxyInEq dom
```
result:
```
Infer.hs:20:18: error:
    * Could not deduce: (1
                         <=? DomainConfigurationPeriod (KnownConf dom))
                        ~ 'True
        arising from a use of `proxyInEq'
      from the context: (KnownDomain dom, KnownNat (DomainPeriod dom),
                         2 <= DomainPeriod dom)
        bound by the type signature for:
                   proxyInEq' :: forall (dom :: Domain).
                                 (KnownDomain dom, KnownNat (DomainPeriod dom),
                                  2 <= DomainPeriod dom) =>
                                 Proxy dom -> ()
        at Infer.hs:(16,1)-(19,41)
    * In the expression: proxyInEq dom
      In an equation for proxyInEq': proxyInEq' dom = proxyInEq dom
    * Relevant bindings include
        dom :: Proxy dom (bound at Infer.hs:20:12)
        proxyInEq' :: Proxy dom -> () (bound at Infer.hs:20:1)
   |
20 | proxyInEq' dom = proxyInEq dom
   |                  ^^^^^^^^^^^^^
```

Type families in general are not a problem:
```haskell
type family F (dom :: Domain) :: Nat

proxyInEq :: (1 <= F dom) => Proxy (dom :: Domain) -> ()
proxyInEq _ = ()

proxyInEq' :: (2 <= F dom) => Proxy (dom :: Domain) -> ()
proxyInEq' dom = proxyInEq dom
```
works just fine (it compiles and functions as expected, I tested with a type instance).

Some more examples of code that does not work, all with periods of a domain:
```haskell
proxyInEq :: ( KnownDomain dom
             , KnownNat (DomainPeriod dom)
             , 1 <= DomainPeriod dom)
          => Proxy (dom :: Domain) -> ()
proxyInEq _ = ()

proxyInEq' :: ( KnownDomain dom
              , KnownNat (DomainPeriod dom)
              , KnownNat (DomainConfigurationPeriod (KnownConf dom))
              , 2 <= DomainConfigurationPeriod (KnownConf dom))
           => Proxy (dom :: Domain) -> ()
proxyInEq' dom = proxyInEq dom
```
```haskell
proxyInEq :: ( KnownDomain dom
             , KnownNat (DomainConfigurationPeriod (KnownConf dom))
             , 1 <= DomainConfigurationPeriod (KnownConf dom))
          => Proxy (dom :: Domain) -> ()
proxyInEq _ = ()

proxyInEq' :: ( KnownDomain dom
              , KnownNat (DomainConfigurationPeriod (KnownConf dom))
              , 2 <= DomainConfigurationPeriod (KnownConf dom))
           => Proxy (dom :: Domain) -> ()
proxyInEq' dom = proxyInEq dom
```

DigitalBrains1, 2021-08-12:
I just realised I did all this with Clash 1.4.2, but initial tests together with @christiaanb did not show any appreciable differences with `master`.

rowanG077, 2021-08-12:
I checked this a few days back and still have the log, this is the relevant part of the when tracing the type checker plugin: 

```
tcPluginSolve start ghc-typelits-natnormalise
  given   = [[G] $d~_a1JES {0}:: 'True ~ 'True (CDictCan),
             [G] co_a1JEM {1}:: KnownConf dom_a1JEr[sk:1]
                                ghc-prim-0.6.1:GHC.Prim.~# fsk_a1JEL[fsk:1] (CFunEqCan),
             [G] co_a1JEO {1}:: Clash.Signal.Internal.DomainConfigurationPeriod
                                  fsk_a1JEL[fsk:1]
                                ghc-prim-0.6.1:GHC.Prim.~# fsk_a1JEN[fsk:1] (CFunEqCan),
             [G] co_a1JEQ {1}:: (1 <=? fsk_a1JEN[fsk:1])
                                ghc-prim-0.6.1:GHC.Prim.~# fsk_a1JEP[fsk:1] (CFunEqCan),
             [G] co_a1JER {1}:: fsk_a1JEP[fsk:1]
                                ghc-prim-0.6.1:GHC.Prim.~# 'True (CTyEqCan)]
  derived = []
  wanted  = [[WD] hole{co_a1JEY} {1}:: (1
                                        <=? (16 * fsk_a1JEN[fsk:1]))
                                       ghc-prim-0.6.1:GHC.Prim.~# 'True (CNonCanonical)]
simplifyNats
  [(Left ([G] co_a1JEO {1}:: Clash.Signal.Internal.DomainConfigurationPeriod
                               fsk_a1JEL[fsk:1]
                             ghc-prim-0.6.1:GHC.Prim.~# fsk_a1JEN[fsk:1] (CFunEqCan),
          Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
          fsk_a1JEN[fsk:1]),
    []),
   (Right ([G] co_a1JEQ {1}:: (1
                               <=? Clash.Signal.Internal.DomainConfigurationPeriod
                                     (KnownConf dom_a1JEr[sk:1]))
                              ghc-prim-0.6.1:GHC.Prim.~# 'True (CNonCanonical),
           (1,
            Clash.Signal.Internal.DomainConfigurationPeriod
              (KnownConf dom_a1JEr[sk:1]),
            True)),
    []),
   (Right ([WD] hole{co_a1JEY} {1}:: (1 <=? (16 * fsk_a1JEN[fsk:1]))
                                     ghc-prim-0.6.1:GHC.Prim.~# 'True (CNonCanonical),
           (1, 16 * fsk_a1JEN[fsk:1], True)),
    [])]
unifyNats
  [G] co_a1JEO {1}:: Clash.Signal.Internal.DomainConfigurationPeriod
                       fsk_a1JEL[fsk:1]
                     ghc-prim-0.6.1:GHC.Prim.~# fsk_a1JEN[fsk:1] (CFunEqCan)
  Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1]
  fsk_a1JEN[fsk:1]
unifyNats result
  Draw [fsk_a1JEN[fsk:1]  :=  Clash.Signal.Internal.DomainConfigurationPeriod
                                fsk_a1JEL[fsk:1]]
New coercion hole: co_a1JEZ
unifyNats(ineq) results
  ([G] co_a1JEQ {1}:: (1
                       <=? Clash.Signal.Internal.DomainConfigurationPeriod
                             (KnownConf dom_a1JEr[sk:1]))
                      ghc-prim-0.6.1:GHC.Prim.~# 'True (CNonCanonical),
   (1,
    Clash.Signal.Internal.DomainConfigurationPeriod
      (KnownConf dom_a1JEr[sk:1]),
    True),
   -1 + Clash.Signal.Internal.DomainConfigurationPeriod
          (KnownConf dom_a1JEr[sk:1]),
   [(Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     fsk_a1JEN[fsk:1],
     True),
    (fsk_a1JEN[fsk:1],
     Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     True),
    (Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     True),
    (Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     True),
    (1,
     Clash.Signal.Internal.DomainConfigurationPeriod
       (KnownConf dom_a1JEr[sk:1]),
     True)])
unifyNats(ineq) results
  ([WD] hole{co_a1JEY} {1}:: (1 <=? (16 * fsk_a1JEN[fsk:1]))
                             ghc-prim-0.6.1:GHC.Prim.~# 'True (CNonCanonical),
   (1, 16 * fsk_a1JEN[fsk:1], True),
   -1 + 16 * Clash.Signal.Internal.DomainConfigurationPeriod
               fsk_a1JEL[fsk:1],
   [(1,
     Clash.Signal.Internal.DomainConfigurationPeriod
       (KnownConf dom_a1JEr[sk:1]),
     True),
    (Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     fsk_a1JEN[fsk:1],
     True),
    (fsk_a1JEN[fsk:1],
     Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     True),
    (1,
     Clash.Signal.Internal.DomainConfigurationPeriod
       (KnownConf dom_a1JEr[sk:1]),
     True),
    (Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     True),
    (Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     Clash.Signal.Internal.DomainConfigurationPeriod fsk_a1JEL[fsk:1],
     True),
    (1,
     Clash.Signal.Internal.DomainConfigurationPeriod
       (KnownConf dom_a1JEr[sk:1]),
     True)])
normalised
  Simplified
  [((CO: Univ(nominal plugin "ghc-typelits-natnormalise"
              :: 1
                 <=? Clash.Signal.Internal.DomainConfigurationPeriod
                       (KnownConf dom_a1JEr[sk:1]), 'True),
     [G] co_a1JEQ {1}:: (1
                         <=? Clash.Signal.Internal.DomainConfigurationPeriod
                               (KnownConf dom_a1JEr[sk:1]))
                        ghc-prim-0.6.1:GHC.Prim.~# 'True (CNonCanonical)),
    []),
   ((CO: Univ(nominal plugin "ghc-typelits-natnormalise"
              :: Clash.Signal.Internal.DomainConfigurationPeriod
                   fsk_a1JEL[fsk:1], fsk_a1JEN[fsk:1]),
     [G] co_a1JEO {1}:: Clash.Signal.Internal.DomainConfigurationPeriod
                          fsk_a1JEL[fsk:1]
                        ghc-prim-0.6.1:GHC.Prim.~# fsk_a1JEN[fsk:1] (CFunEqCan)),
    [[WD] hole{co_a1JEZ} {1}:: fsk_a1JEN[fsk:1]
                               ghc-prim-0.6.1:GHC.Prim.~# Clash.Signal.Internal.DomainConfigurationPeriod
                                                            fsk_a1JEL[fsk:1] (CNonCanonical)])]
tcPluginSolve ok ghc-typelits-natnormalise
  solved = []
  new    = []
```

rowanG077, 2022-02-11:
Fixed in https://github.com/clash-lang/ghc-typelits-natnormalise/pull/63

closed 2023-09-01 by rowanG077