clash-lang/clash-compiler #1904: Program using GADTs recursive after normalization

alex-mckenna, 2021-08-16:
From the FP slack:

> i wrote this minimizer component to use in part of my project. it illustrates the example of having two different registers that need to be enabled separately; one should be updated whenever the circuit is enabled, and the other should only update when it finds a new minimum value

```haskell
{-# LANGUAGE GADTs #-}

import Data.Maybe
import Clash.Prelude

-- minLoop minimizes over an input domain examined one item at a time.
minLoop
    :: forall n a dom.
       ( KnownNat n, Ord a, Num a
       , NFDataX a, HiddenClockResetEnable dom )
    => SNat n
    -> Signal dom (Maybe a)
    -> Signal dom (MinLoop n a)
minLoop SNat item = x where
    y = go <$> item <*> x
    x = save y

    go item (MinLoop idx low min) =
        case (qinc, qupd) of
            (True, True ) -> NewMin idx' idx (fromJust item)
            (True, False) -> KeepMin idx'
            _             -> SkipMin
      where
        idx' = idx + 1

        qinc = isJust item
        qupd = qinc && (qzro || qmin)
        qzro = 0 == idx
        qmin = (fromJust item) < min
    -- ^ decide whether to update the current minimum

    save u = MinLoop <$> idx' <*> low' <*> min' where
        idx' = regMaybe 0 idx
        low' = regMaybe 0 low
        min' = regMaybe 0 min

        (idx, low, min) = (unbundle . fmap go2) u
        go2 v = case v of
            NewMin i l m -> (Just i , Just l , Just m )
            KeepMin i    -> (Just i , Nothing, Nothing)
            _            -> (Nothing, Nothing, Nothing)
    -- ^ save all values in the registers
{-# NOINLINE minLoop #-}

data MinLoop (n :: Nat) a where
    MinLoop
        :: (KnownNat n, Ord a)
        => Unsigned n -> Unsigned n -> a -> MinLoop n a
    -- ^ Complete minloop data.
    NewMin
        :: (KnownNat n, Ord a)
        => Unsigned n -> Unsigned n -> a -> MinLoop n a
    -- ^ Update minloop state with new minimum.
    KeepMin
        :: (KnownNat n, Ord a)
        => Unsigned n -> MinLoop n a
    -- ^ Don't update minimum, just increment the loop.
    SkipMin
        :: (KnownNat n, Ord a) => MinLoop n a
    -- ^ Don't do anything; the module wasn't enabled.

topEntity
    :: Clock System
    -> Reset System
    -> Enable System
    -> Signal System (Maybe Int)
    -> Signal System (Unsigned 3)
topEntity = exposeClockResetEnable go where
    go :: HiddenClockResetEnable dom
       => Signal dom (Maybe Int) -> Signal dom (Unsigned 3)
    go = fmap (\k -> case k of { MinLoop idx _ _ -> idx }) . minLoop d3
```

> my hope is basically that the compiler realizes the MinLoop datatype can be "transparent". by that i mean i'd like it if the inliner could notice that qinc enables the idx register and qupd enables the low and min registers. otherwise, it'll probably compile pretty inefficiently, since alternative constructors get compiled to muxes and decoders in verilog; lots more logic than necessary for something like this.

> i don't actually want this datatype to appear in the resulting verilog - it's just an abstraction over the real controls that i hope to have

It would be nice if there's some change we can make to normalization and/or this input program such that it can render to HDL more in line with that the user is after. If this is not possible, it would be nice to have an explanation documented somewhere

exclusive-and, 2021-08-18:
Thanks for opening this issue for me Alex.

I've been thinking about this a bit. I think that fundamentally this whole question goes back to how Clash handles pattern matching in the outputted HDL. Function alternatives and case expressions currently get compiled to series of multiplexers and comparators leading either into or out of a circuit. I'll provide some examples below.

This (admittedly somewhat hasty) code uses pattern matching in the definitions of `go` and `go2`:
```

{-# LANGUAGE GADTs #-}

import qualified GHC.Magic

import Data.Bool
import Data.Maybe

import Clash.Prelude

(&$) = GHC.Magic.noinline (&&)
(|$) = GHC.Magic.noinline (||)

-- minLoop minimizes over an input domain examined one item at a time.

minLoop
    :: forall n a dom.
       ( KnownNat n, Ord a, Num a
       , NFDataX a, HiddenClockResetEnable dom )
    => SNat n
    -> Signal dom (Maybe a)
    -> Signal dom (MinLoop n a)

minLoop SNat item = x where
    y = go <$> item <*> x
    x = save y

    go (Just item) (MinLoop idx low min)
        | qupd      = NewMin idx' idx item
        | otherwise = KeepMin idx
      where
        idx' = idx + 1

        qupd = qzro |$ qmin
        qzro = 0 == idx
        qmin = item < min

    go _ _ = SkipMin
    -- ^ decide whether to update the current minimum

    save u = MinLoop <$> idx' <*> low' <*> min' where
        idx' = regMaybe 0 idx
        low' = regMaybe 0 low
        min' = regMaybe 0 min

        (idx, low, min) = (unbundle . fmap go2) u
        go2 (NewMin i l m)  = (Just i , Just l , Just m )
        go2 (KeepMin i)     = (Just i , Nothing, Nothing)
        go2 _               = (Nothing, Nothing, Nothing)
    -- ^ save all values in the registers

data MinLoop (n :: Nat) a where
    MinLoop :: Unsigned n -> Unsigned n -> a -> MinLoop n a
    -- ^ Complete minloop data.
    NewMin  :: Unsigned n -> Unsigned n -> a -> MinLoop n a
    -- ^ Update minloop state with new minimum.
    KeepMin :: Unsigned n -> MinLoop n a
    -- ^ Don't update minimum, just increment the loop.
    SkipMin :: MinLoop n a
    -- ^ Don't do anything; the module wasn't enabled.
```
The code produces RTL that looks like this:
[image]
Note the unnecessary logic I mentioned.

I've also gone and quickly written a version which _does_ generate control lines as intended:
```

(&$) = GHC.Magic.noinline (&&)
(|$) = GHC.Magic.noinline (||)

minLoop
    :: forall n a dom.
       ( KnownNat n, Ord a, Num a
       , NFDataX a, HiddenClockResetEnable dom )
    => SNat n
    -> Signal dom (Maybe a)
    -> Signal dom (MinLoop2 n a)

minLoop SNat item = x where
    y = go <$> item <*> x
    x = save y

    go item (MinLoop2 idx low min) =
        MinLoop1 qinc qupd idx' idx (fromJust item)
      where
        idx' = idx + 1

        qinc = isJust item
        qupd = qinc &$ (qzro |$ qmin)
        qzro = 0 == idx
        qmin = (fromJust item) < min

    save u = MinLoop2 <$> idx' <*> low' <*> min' where
        idx' = regMaybe 0 idx
        low' = regMaybe 0 low
        min' = regMaybe 0 min

        (idx, low, min) = (unbundle . fmap go2) u
        go2 (MinLoop1 qinc qupd i l m) =
            ( bool Nothing (Just i) qinc
            , bool Nothing (Just l) qupd
            , bool Nothing (Just m) qupd )

data MinLoop1 (n :: Nat) a =
    MinLoop1 Bool Bool (Unsigned n) (Unsigned n) a

data MinLoop2 (n :: Nat) a =
    MinLoop2 (Unsigned n) (Unsigned n) a
```
which generates this:
[image]
Much better!

I think the thing to consider is that many important circuits have the shape
```
logic :: s1 -> a -> s2
save :: Signal dom s2 -> Signal dom s1

circ :: Signal dom a -> Signal dom s2
circ a = x where
  y = logic <$> x <*> a
  x = save y
```
In this kind of circuit, the behaviours are mostly determined by the enable lines of the registers in `save`. Note that in my second example of `minLoop`: the state of the circuit only changes when the appropriate register is enabled. But Clash doesn't exactly have a very natural representation of that shape. If we pack constructors into binary representations, then use comparators and multiplexers to select between wires, we inevitably end up with more hardware than necessary in a lot of cases.

So my thought is that perhaps Clash should try to preferentially generate control lines like in the second example, instead of muxes like in the first. In my `minLoop` examples, it's not a huge issue either way. But when you've got much more involved types, like in the project I'm working on where there are many queues that need to be written and read from independently, it becomes more of a hassle. Writing the behaviour exclusively as records with a bunch of Bools for control is definitely un-Haskell-y.

alex-mckenna, 2021-08-18:
Hmm, perhaps you could avoid passing around the booleans and make it more idiomatic Haskell by using `Clash.Explicit.Prelude` and [this function to combine enables](https://hackage.haskell.org/package/clash-prelude-1.4.3/docs/Clash-Explicit-Signal.html#v:enable). Then when you declare the registers it would look more like:

```haskell
idx' = regMaybe clk rst (enable en qinc) 0 idx
```

There isn't an implicit version of `enable` currently, but @DigitalBrains1 has a PR which addresses this [here](https://github.com/clash-lang/clash-compiler/pull/1849)

exclusive-and, 2021-08-18:
I'd considered trying to use `enable`, but I can't really see how it avoids the problem of passing Bools around.

The reason for the Bool passing is that most of the actual work is done in a non-signal function (`go :: Maybe a -> MinLoop2 n a -> MinLoop1 n a`), which is how I'd like to keep it for the most part. If I wrote all the logic as a `Signal` function, I'd quickly go down a spiral of hundreds of `fmap`s. I actually wrote some code this way a few months ago, and it was awful.

exclusive-and, 2021-08-19:
I've found an almost-workaround. In short, instead of doing
```hs
data MyType = MyType1 Arg | MyType2 Arg
```
I've tried
```hs
data MyType c = MyType c Arg
data MyTypeCtrl = MyType1 | MyType2

-- ...
go :: MyType () -> MyType MyTypeCtrl
```

This approach makes circuits that are somewhat less messy.

The same `MinLoop` example reimplemented this way:
```hs
minLoop
    :: forall n a dom.
       ( KnownNat n, Ord a, Num a
       , NFDataX a, HiddenClockResetEnable dom )
    => SNat n
    -> Signal dom (Maybe a)
    -> Signal dom (MinLoop () n a)

minLoop SNat item = x where
    y = go <$> item <*> x
    x = save y

    go item (MinLoop () idx low min) = MinLoop c idx' idx val where
        c = case (qinc, qupd) of
            (True , True ) -> NewMin
            (True , False) -> KeepMin
            (_    , _    ) -> SkipMin

        idx' = idx + 1

        qupd = qinc &$ (qzro |$ qmin)
        qzro = 0 == idx
        qmin = val < min

        qinc = isJust item
        val = fromJust item
    -- ^ decide whether to update the current minimum

    save u = MinLoop () <$> idx' <*> low' <*> min' where
        idx' = (regMaybe 0 . fmap idx) u
        low' = (regMaybe 0 . fmap low) u
        min' = (regMaybe 0 . fmap min) u

        idx (MinLoop NewMin  i _ _) = Just i
        idx (MinLoop KeepMin i _ _) = Just i
        idx (MinLoop _       _ _ _) = Nothing

        low (MinLoop NewMin _ l _) = Just l
        low (MinLoop _      _ _ _) = Nothing

        min (MinLoop NewMin _ _ m) = Just m
        min (MinLoop _      _ _ _) = Nothing
    -- ^ save all values in the registers

data MinLoop c (n :: Nat) a = MinLoop c (Unsigned n) (Unsigned n) a

data MinLoopCtrl = NewMin | KeepMin | SkipMin
```
Which generates the closer-to-what's-desired circuit:
[image]

An example with two queues:
```hs
queues
    :: forall dom. HiddenClockResetEnable dom
    => Signal dom Int
    -> Signal dom (Maybe Int)
    -> Signal dom (Maybe Int)
    -> Signal dom (Queues ())

queues addr a b = x where
    y = go <$> addr <*> a <*> b <*> x
    x = save y

    go addr a b (Queues () queue1 queue2) = Queues c queue1' queue2' where
        c = case (qset1, qset2) of
            (True , True ) -> QWrite3 addr
            (True , False) -> QWrite2 addr
            (False, True ) -> QWrite1 addr
            (False, False) -> QDoNothing

        qset1 = isJust a
        qset2 = isJust b

        queue1' = fromJust a
        queue2' = fromJust b

    save :: HiddenClockResetEnable dom => Signal dom (Queues QueuesCtrl) -> Signal dom (Queues ())

    save u = Queues () <$> queue1' <*> queue2' where
        queue1' = (asyncRam d4 cnt . fmap queue1) u
        queue2' = (asyncRam d4 cnt . fmap queue2) u

        queue1 :: Queues QueuesCtrl -> Maybe (Int, Int)

        queue1 (Queues (QWrite3 addr) v _) = Just (addr, v)
        queue1 (Queues (QWrite2 addr) v _) = Just (addr, v)
        queue1 (Queues _              _ _) = Nothing

        queue2 (Queues (QWrite3 addr) _ v) = Just (addr, v)
        queue2 (Queues (QWrite1 addr) _ v) = Just (addr, v)
        queue2 (Queues _              _ _) = Nothing

        cnt = register 0 $ cnt + 1

data Queues c = Queues c Int Int

data QueuesCtrl
    = QWrite3 Int
    | QWrite2 Int
    | QWrite1 Int
    | QDoNothing

topEntity
    :: Clock System
    -> Reset System
    -> Enable System
    -> Signal System Int
    -> Signal System (Maybe Int)
    -> Signal System (Maybe Int)
    -> Signal System (Queues ())

topEntity = exposeClockResetEnable queues
```
Which results in the not-so-bad circuit:
[image]

Unfortunately both of these examples still retain a few vestiges of the pattern-matching issue. In the `minLoop` example, there's still a stray multiplexer that should probably be an AND gate. And in the queues example, there are the three weird comparators that should get optimized out.

There's also the slight disadvantage that this approach is incompatible with `autoReg`. We don't have `MyType () ~ MyType MyTypeCtrl`, which is necessary for `autoReg :: Signal dom a -> Signal dom a`.