clash-lang/clash-compiler #1905: Allow BiSignalIn to be "unbundled" and "bundled"

rowanG077, 2021-08-23:
Currently there is no way to split a  `BiSignalIn` of `n` bits into a vector of `BiSignalIn` with 1 bit and vice versa. This is something I need since I have a tristate buffer primitive which I can only instantiate with a width of a single bit. Currently there is no to instantiate this primitive because I cannot split my 32 bit width `BiSignalIn` into 1 bit `BiSignalIn` to apply this primitive to.

I can make a PR for this but I was thinking of a good name for this. `bundle` and `unbundle` are already taken so I'd rather avoid those, so are `pack` and `unpack`. Maybe `splitBiSigIn` and `unsplitBiSigIn` could be used.

christiaanb, 2021-08-23:
Some languages only have qualified imports, so perhaps shadowing `bundle` and `unbundle` inside `Clash.Signal.BiSignal` isn't so bad. 

It would also be cool if you could check whether you can make `Bundle` instances for `BiSignalIn` and `BiSignalOut` on the https://github.com/clash-lang/clash-compiler/tree/alt_bundle branch; currently we're somewhat undecided on merging that work, leaning towards not merging it. But perhaps if it allows bundling for `BiSignalIn` and `BiSignalOut` that might speak in its favor. That being said.... it's probably just a fun exercise, the loss of type inference in that `alt_bundle` branch is just too high of a price.

rowanG077, 2021-08-23:
It's possible using that type class, it's quite ugly because of a lot of conversions between `Vec` and `BitVector`: 

```
instance (KnownNat n, n ~ (m + 1)) => Bundle (BiSignalIn ds dom) n (Vec n (BiSignalIn ds dom 1)) where
  type Unbundled (BiSignalIn ds dom) n = Vec n (BiSignalIn ds dom 1)
  bundle vs = BiSignalIn (getDs (head vs)) bun
    where
      bun = fmap (fmap $ v2bv . fmap lsb)
              $ fmap sequenceA
              $ bundle $ fmap getSig vs
      getDs (BiSignalIn ds _) = ds
      getSig (BiSignalIn _ s) = s
  unbundle (BiSignalIn ds s) = fmap (BiSignalIn ds) unb
    where
      f Nothing = repeat Nothing
      f (Just bv) = fmap (Just . v2bv . singleton) (bv2v bv)
      unb = unbundle $ fmap f s
```

alex-mckenna, 2021-08-23:
I'm not a huge fan of this instance, it says it's a `Bundle` but it feels like it wants to be a `BitPack` too. Although I'm also somewhat against

  * `Bundle` instances like the one on `BitVector n` where `Unbundled a ~ Signal dom a` and `bundle = unbundle = id`
  * the proposed new `Bundle` class which feels a lot less ergonomic / unfriendly to beginners

so maybe I'm just somewhat negative about these kinds of thing anyway :sweat_smile:

rowanG077, 2021-08-23:
What are you supposed to do with a `BiSignalOut` exactly? I just discard it in my circuit. If I look at the docs and the code the only thing I can see that you can actually do with it is call `veryUnsafeToBiSignalIn`, which beside checking that you are not writing multiple times at a single clock cycles doesn't have any use?

christiaanb, 2021-08-23:
You're supposed to combine them with [`mergeBiSignalOuts`](https://hackage.haskell.org/package/clash-prelude-1.4.3/docs/Clash-Signal-BiSignal.html#v:mergeBiSignalOuts)

rowanG077, 2021-08-23:
@christiaanb And then? Won't you just discard the result in the end anyway?

christiaanb, 2021-08-23:
I think the example in https://hackage.haskell.org/package/clash-prelude-1.4.3/docs/Clash-Signal-BiSignal.html shows how you use both `mergeBiSignalOuts` and `veryUnsafeToBiSignalIn` to simulate multiple writers on a shared bus.

rowanG077, 2021-08-23:
Ah yes, indeed!

rowanG077, 2021-08-23:
@alex-mckenna What do you propose for this use case?

alex-mckenna, 2021-08-23:
I have no immediate suggestions. It would be more convenient if `BiSignal` took an actual type instead of it's `BitSize`, but I assume there's a good reason it isn't like that already...

For your specific use case my preference would be that you define standalone functions instead of a class instance, or if you really need it to be `Bundle` then make an orphan instance

TL;DR :man_shrugging:

christiaanb, 2021-08-23:
Yeah, there's a reason for it being basically a bitvector: the `PullUp` makes it all `1`s and the `PullDown` makes it all `0`s when nobody is writing to the bus. What his means for an arbitrary `a`, I don't know.

So my suggestion would simply to have:
```
module Clash.Signal.BiSignal 
  ( ...
  , bundle
  , unbundle
  )
where

bundle :: Vec n (BiSignalOut ds dom 1) -> BiSignalOut ds dom n
unbundle :: BiSignalIn ds dom n -> Vec n (BiSignalIn ds dom 1)
```
that should work for your 1-bit primitives I believe

alex-mckenna, 2021-08-23:
Hmm, `readFromBiSignalIn` and `writeToBiSignal` have `BitPack` constraints, so I guess it would be acceptable to use `unpack (0 :: BitVector (BitSize a))` and `unpack (-1 :: BitVector (BitSize a))` to get the values of `a` for `PullDown` and `PullUp` respectively. Although maybe that's more prone to simulation exceptions when the bit patterns don't correspond to valid values for `a`.

Not that it particularly matters, changing the definition now would break API anyway

rowanG077, 2022-02-03:
Closed as the issue can be worked around.

closed 2022-02-03 by rowanG077