clash-lang/clash-compiler #1918: Spurious warning for `Index`

DigitalBrains1, 2021-09-03:
It would seem that commit 788526756 unearthed a somewhat hidden use of `integerToInt`.

The following code:
```haskell
topEntity
  :: Vec 5 Int
  -> Index 5
  -> Int
topEntity = (!!)
```
has a spurious warning on synthesis:
```
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.integerToInt: GHC.Integer.Type.integerToInt: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
```

It seems to be caused by the [definition of `fromEnum`](https://github.com/clash-lang/clash-compiler/blob/bdc6e7576612c77c1384e68b36d4bead9a0a7fa3/clash-prelude/src/Clash/Sized/Internal/Index.hs#L226) for `Index.`

The full output of `DebugFinal` is:
```
GHC: Parsing and optimising modules took: 0.959s
GHC: Loading external modules from interface files took: 0.000s
GHC: Parsing annotations took: 0.000s
Clash: Parsing and compiling primitives took 0.216s
GHC+Clash: Loading modules cumulatively took 1.344s
Clash: Compiling IndexInteger.topEntity
Clash: Ignoring previously made caches
IndexInteger.topEntity[8214565720323816207] before normalization:

IndexInteger.$s!![8214565720323820773][GlobalId]
  @GHC.Types.Int[3674937295934324766]

IndexInteger.topEntity[8214565720323816207] after normalization:

λ(c$arg[0] :: Clash.Sized.Vector.Vec[8214565720323788096]
                5
                GHC.Types.Int[3674937295934324766]) ->
λ(c$arg[1] :: Clash.Sized.Internal.Index.Index[8214565720323788089]
                5) ->
letrec
  result[5] :: GHC.Types.Int[3674937295934324766]
  = IndexInteger.$s!![2][GlobalId]
      c$arg[0][LocalId]
      c$arg[1][LocalId]
in result[5][LocalId]

IndexInteger.$s!![2] before normalization:

(Λa[6989586621679042489] ->
  λ(xs[6989586621679042492] :: Clash.Sized.Vector.Vec[8214565720323788096]
                                 5
                                 a[6989586621679042489]) ->
  λ(i1[6989586621679042493] :: Clash.Sized.Internal.Index.Index[8214565720323788089]
                                 5) ->
  letrec
    wild[6989586621679042829] :: GHC.Prim.Int#[3674937295934324764]
    = GHC.Integer.Type.integerToInt
        (Clash.Sized.Internal.Index.toInteger# @5
           i1[6989586621679042493][LocalId])
  in case wild[6989586621679042829][LocalId] of
       _ ->
         Clash.Sized.Vector.index_int @5
           @a[6989586621679042489]
           IndexInteger.$s!!1[8214565720323820828][GlobalId]
           xs[6989586621679042492][LocalId]
           (GHC.Types.I#
              wild[6989586621679042829][LocalId]))
  @GHC.Types.Int[3674937295934324766]

IndexInteger.$s!![2] after normalization:

λ(xs[6989586621679042492] :: Clash.Sized.Vector.Vec[8214565720323788096]
                               5
                               GHC.Types.Int[3674937295934324766]) ->
λ(i1[6989586621679042493] :: Clash.Sized.Internal.Index.Index[8214565720323788089]
                               5) ->
letrec
  wild[6989586621679042829] :: GHC.Prim.Int#[3674937295934324764]
  = GHC.Integer.Type.integerToInt
      c$wild_app_arg[6][LocalId]
  result[8] :: GHC.Types.Int[3674937295934324766]
  = Clash.Sized.Vector.index_int @5
      @GHC.Types.Int[3674937295934324766]
      5
      xs[6989586621679042492][LocalId]
      (GHC.Types.I#
         wild[6989586621679042829][LocalId])
  c$wild_app_arg[6] :: GHC.Integer.Type.Integer[3674937295934324784]
  = Clash.Sized.Internal.Index.toInteger# @5
      i1[6989586621679042493][LocalId]
in result[8][LocalId]

IndexInteger.topEntity[8214565720323816207] :: Clash.Sized.Vector.Vec[8214565720323788096]
                                                 5
                                                 GHC.Types.Int[3674937295934324766]
                                           -> Clash.Sized.Internal.Index.Index[8214565720323788089]
                                                5
                                           -> GHC.Types.Int[3674937295934324766] before flattenExpr:

λ(c$arg[0] :: Clash.Sized.Vector.Vec[8214565720323788096]
                5
                GHC.Types.Int[3674937295934324766]) ->
λ(c$arg[1] :: Clash.Sized.Internal.Index.Index[8214565720323788089]
                5) ->
letrec
  result[5] :: GHC.Types.Int[3674937295934324766]
  = (λ(xs[6989586621679042492] :: Clash.Sized.Vector.Vec[8214565720323788096]
                                    5
                                    GHC.Types.Int[3674937295934324766]) ->
      λ(i1[6989586621679042493] :: Clash.Sized.Internal.Index.Index[8214565720323788089]
                                     5) ->
      letrec
        wild[6989586621679042829] :: GHC.Prim.Int#[3674937295934324764]
        = GHC.Integer.Type.integerToInt
            c$wild_app_arg[6][LocalId]
        result[8] :: GHC.Types.Int[3674937295934324766]
        = Clash.Sized.Vector.index_int @5
            @GHC.Types.Int[3674937295934324766]
            5
            xs[6989586621679042492][LocalId]
            (GHC.Types.I#
               wild[6989586621679042829][LocalId])
        c$wild_app_arg[6] :: GHC.Integer.Type.Integer[3674937295934324784]
        = Clash.Sized.Internal.Index.toInteger# @5
            i1[6989586621679042493][LocalId]
      in result[8][LocalId])
      c$arg[0][LocalId]
      c$arg[1][LocalId]
in result[5][LocalId]

IndexInteger.topEntity[8214565720323816207] :: Clash.Sized.Vector.Vec[8214565720323788096]
                                                 5
                                                 GHC.Types.Int[3674937295934324766]
                                           -> Clash.Sized.Internal.Index.Index[8214565720323788089]
                                                5
                                           -> GHC.Types.Int[3674937295934324766] after flattenExpr:

λ(c$arg[0] :: Clash.Sized.Vector.Vec[8214565720323788096]
                5
                GHC.Types.Int[3674937295934324766]) ->
λ(c$arg[1] :: Clash.Sized.Internal.Index.Index[8214565720323788089]
                5) ->
letrec
  wild[6989586621679042829] :: GHC.Prim.Int#[3674937295934324764]
  = GHC.Integer.Type.integerToInt
      c$wild_app_arg[6][LocalId]
  result[8] :: GHC.Types.Int[3674937295934324766]
  = Clash.Sized.Vector.index_int @5
      @GHC.Types.Int[3674937295934324766]
      5
      c$arg[0][LocalId]
      (GHC.Types.I#
         wild[6989586621679042829][LocalId])
  c$wild_app_arg[6] :: GHC.Integer.Type.Integer[3674937295934324784]
  = Clash.Sized.Internal.Index.toInteger# @5
      c$arg[1][LocalId]
in result[8][LocalId]

Clash: Applied 14 transformations
Clash: Normalization took 0.006s
[WARNING] Dubious primitive instantiation for GHC.Integer.Type.integerToInt: GHC.Integer.Type.integerToInt: Integers are dynamically sized in simulation, but fixed-length after synthesis. Use carefully. (disable with -fclash-no-prim-warn)
Clash: Netlist generation took 0.001s
Clash: Total compilation took 1.371s
```

alex-mckenna, 2021-09-03:
Maybe a good idea would be to have a black box for `fromEnum#` so the warning is at least more relevant. Then we also have the choice of keeping the truncating behaviour that `integerToInt` has in the Clash simulation or replacing it with an `errorX` if the `Index` doesn't fit into an `Int`

DigitalBrains1, 2021-09-03:
We already warn when you try to do `Index.toInteger` on an `Index` that is more than 64 bits wide. So the `toInteger` step of a `fromEnum` is handled gracefully by Clash, but then it's the `integerToInt` that follows that gives a warning. I think we can just make the latter go away.

alex-mckenna, 2022-01-21:
Closing in favour of the more general https://github.com/clash-lang/clash-compiler/issues/2046

closed 2022-01-21 by alex-mckenna