clash-lang/clash-compiler #1927: Use of satSucc triggers "Not in normal form: application of a Lambda-expression"

danielsmw, 2021-09-14:
Consider the following simple example (sorry for not being able to get it more minimal, I tried but failed):

```haskell
topEntity
  :: Signal System (Vec 1 (Signed 1))
  -> Signal System (Maybe (Vec 1 (Signed 1, Signed 1)))
  -> Clock System -> Reset System -> Enable System
  -> Signal System (Maybe (Vec 1 (Signed 1, Signed 1)))
topEntity angles xs c r e = exposeClockResetEnable (machine @8 angles xs) c r e 

machine
  :: forall numStages pairs coord angle dom counter
   . HiddenClockResetEnable dom
  => (1 <= numStages)
  => (KnownNat numStages, KnownNat pairs)
  => (Default coord, NFDataX coord)
  => (Default angle, NFDataX angle)
  => counter ~ (Index numStages)
  => Signal dom (Vec pairs angle)
  -> Signal dom (Maybe (Vec pairs (coord, coord)))
  -> Signal dom (Maybe (Vec pairs (coord, coord)))
machine thetas coords = mealy transition def $ liftA2 (liftM2 zip) coords (return <$> thetas)
  where
    transition :: (counter, Vec pairs ((coord, coord), angle))
               -> Maybe (Vec pairs ((coord, coord), angle))
               -> ( (counter, Vec pairs ((coord, coord), angle))
                  , Maybe (Vec pairs (coord, coord))
                  )
    transition (k, xyzs) maybeInputs = case (k == 0, maybeInputs)
                                         of (True, Nothing) -> ((0, xyzs), Just $ map fst xyzs)
                                            (True, Just xyzs0) -> ((0, xyzs0), Just $ map fst xyzs)
                                            (False, _) -> ((satSucc SatWrap k, xyzs), Nothing)
```

This triggers the following error:
```
    Clash.Netlist(914): Not in normal form: application of a Lambda-expression

    k[6989586621679046945][LocalId] ▷
      GHC.Integer.Type.Integer[3674937295934324784]
      ~
      Clash.Sized.Internal.Index.Index[8214565720323789603] 8
```

Now, remarkably, the following modifications *do* allow this design to compile:

- Fixing `numStages` as a static value in the type signature of `machine`, rather than passing it in via TypeApplication.
- Replacing `satSucc SatWrap k` with `k`, `k+1`, or even -- to my great surprise -- `satAdd SatWrap k 1` or `satAdd SatWrap 1 k`.

The fact that I can use `satAdd SatWrap 1 k` but not `satAdd SatSucc k` feels like a bug, so I'm reporting it here. Apologies again for not being able to reduce it further... simpler examples didn't seem to give the same error, which makes me worry that there's something more complicated going on with the `Vec`s or `Maybe`s that's helping induce Clash's confusion.

christiaanb, 2021-09-14:
This is due to Clash' improper handling of casts. We currently only keep casts of the form:
```
Integer ~ Index n
Integer ~ Signed n
Natural ~ Unsigned n
```
and their symmetric variant. But your example introduces:
```
counter ~ Integer
```
which is thrown away. This then results in:
```
k ▷ (Index 8 ~ Integer) ▷ (Integer ~ Index 8)
```
not being eliminated, because the `▷ (Index 8 ~ Integer)` was thrown away at a very early stage.

I'm investigating a fix.

christiaanb, 2021-09-14:
So adding:
```diff
diff --git a/clash-ghc/src-ghc/Clash/GHC/GHC2Core.hs b/clash-ghc/src-ghc/Clash/GHC/GHC2Core.hs
index a9b8deeb..f2978efc 100644
--- a/clash-ghc/src-ghc/Clash/GHC/GHC2Core.hs
+++ b/clash-ghc/src-ghc/Clash/GHC/GHC2Core.hs
@@ -659,6 +659,7 @@ isSizedCast (TyConApp tc1 _) (TyConApp tc2 _) = do
         ,tc2 `hasKey` naturalTyConKey &&
           tc1Nm == "Clash.Sized.Internal.Unsigned.Unsigned"
         ])
+isSizedCast (TyVarTy {}) (TyConApp tc2 _) = return (tc2 `hasKey` integerTyConKey)
 isSizedCast _ _ = return False
 
 hasPrimCo :: Coercion -> C2C (Maybe Type)
```
makes the bug disappear.

christiaanb, 2021-09-14:
Thanks for the report @danielsmw 

Also thanks for trying to minimize! For future reference: Already having access to the code is a big help. So don’t feel too bad about not being able to minimize further.

danielsmw, 2021-09-14:
Sounds good -- thanks!

commit 2923f046 (2021-09-14): Keep casts of the form `x ~ Integer`

Really a work-around until we properly handle all casts.

Fixes #1927